RobTillaart/Arduino #26: Running test .ino files

scheung38, 2015-10-28:
Not sure why after cloning project and for example, within Ardiuno IDE 1.6.5:

dht_test1.ino:

verify:
```
dht_test1.ino:11:17: fatal error: dht.h: No such file or directory
compilation terminated.
Error compiling.
```

RobTillaart, 2015-10-28:
I do not know for IDE1.6.5 (I'm still at 1.5.8) 

In the IDE -> File menu-> preferences you can find your sketchbook path.
In that path you need to make a folder libraries (or it exists already)

create a folder dht under libraries 
so for you the path should become:  /Users/seb/Documents/Arduino/Libraries/dht

copy dht.h dht.cpp and the example folder there

restart all instances of the IDE 

and then it should work.

RobTillaart, 2015-11-21:
As there are no further remarks I close the issue

closed 2015-11-21 by RobTillaart

andrew-raphael-lukasik, 2016-05-21:
(Arduino IDE 1.6.9 and .h .cpp and .ino in the same folder)
Changing 
#include <dht.h>
to:
#include "dht.h"
fixed this issue in my case