RobTillaart/Arduino #30: read22 always returns DHTLIB_ERROR_TIMEOUT on ATtiny85

SashkaCosmonaut, 2015-12-03:
I use your glorious library to work with DHT22 module on attiny85 @8MHz (I also used 1MHz, 8MHz, 16 MHz). But the read22 function allways returns -2 (DHTLIB_ERROR_TIMEOUT). 

A more detailed description of my problem I posted on [Arduino Stackexchange](http://arduino.stackexchange.com/questions/18163/using-attiny85-nrf24l01-dht2-cant-get-data-from-dht22). 

Now I just want to get data from DHT22 with your library, but the answer is allways DHTLIB_ERROR_TIMEOUT (but if connection is broken purposely, it returns DHTLIB_ERROR_CONNECT, it shows that library is working well).

Here is my code:

    #include <dht.h>
    #include <RF24.h>

    #define DHT22_PIN 3

    byte currentNodeAddr[6] = "1Node";  // Address of this node
    RF24 radio(2,2);                                // Create instance of RF24 radio object
    dht DHT;                                            // Create instance of DHT22 module object

    // Struct for sending measured data
    struct Data {
        uint16_t temp;  // Temperature value
        uint16_t hum;   // Humidity value
        int status;     // DHT22 module status
    } data;

    void setup() {
        // Set up the radio
        radio.begin();
        radio.openWritingPipe(currentNodeAddr);
        radio.stopListening();
    }

    void loop() {
        delay(2000);
  
        // Get values from DHT22
        data.status = DHT.read22(DHT22_PIN);
        data.temp = DHT.temperature;
        data.hum = DHT.humidity;
  
        radio.write(&data, sizeof(Data));  // Send data to the Raspberry
    }

Maybe I have done something wrong with connection to DHT22, or is it really a problem in the radio module?

RobTillaart, 2015-12-03:
Q1: Which version of the library?
Q2: If you leave out the radio code, does it work then?

SashkaCosmonaut, 2015-12-03:
1. Version: 0.1.21.
2. I used the I2C connection instead of the radio module, but nothing changed.

I have added dome editions to my [question](http://arduino.stackexchange.com/questions/18163/using-attiny85-nrf24l01-dht2-cant-get-data-from-dht22). Can you please look at it?

RobTillaart, 2015-12-03:
please try experiment 1:
patch the dht.h file by uncommenting some lines

// #ifndef F_CPU
#define DHTLIB_TIMEOUT 1000  // ahould be approx. clock/40000
// #else
// #define DHTLIB_TIMEOUT (F_CPU/40000)
// #endif

it might be that the timeout counter is critical.
optional increase the TIMEOUT to 10000

RobTillaart, 2015-12-03:
experiment 2:
replace the 4K7 resistor with a 3K3 to get sharper edges.

Q: do you use 5V or 3.3V ?

yesyesuk, 2015-12-03:
Interesting...
I have used many (over 20) DHT22s with Arduinos but I never put a resistor between Vcc and the DATA pin and it always worked. Is that resistor required?

While I'm commenting, thanks sooo much for this library!!

RobTillaart, 2015-12-03:
@yesyesuk You're welcome
The resistor makes the data line going faster to 5V so the pulses are more square. That said, I use them only when wires are long.

RobTillaart, 2015-12-03:
Experiment 3:
@SashkaCosmonaut 
In this thread 8Mhz on a tiny84 is discussed. http://forum.arduino.cc/index.php?topic=291695.0.
It refers to the 0.1.18 version, you could give it a try too

RobTillaart, 2015-12-03:
Q: does the RF library use interrupts?

SashkaCosmonaut, 2015-12-06:
Yes! It works! Thank you very, very much! I struggled with this problem for a week.

I only made the first experiment (only uncommented the line “#define DHTLIB_TIMEOUT 1000”). ATtiny is programmed on 8 MHz. Also the DHT22 is working at a voltage of 3.3V with a 3K3 resistor (as well as the module is working without it).

But after that there is probability of 2/10 to get the DHTLIB_ERROR_CHECKSUM error and wrong data (during the checking every 2-3 second). But I hope it is much easier to fix this problem. I will do a few more tests to determine optimum operating conditions of the DHT22.

Answering to your question, I use this [repository]( https://github.com/TMRh20/RF24) for RF24. As I see, this library does not use interrupts.

Will not you mind, if I mention about this issue and your solution in the answer on Arduino Stackexchange?

Thank you very much for your help and for your library!

RobTillaart, 2015-12-06:
>> Will not you mind, if I mention about this issue and your solution in the answer on Arduino Stackexchange?
On contrary Alexander, please mention it as it can help other people too.

For the CHECKSUM error I would first try another smaller resistor to get sharper edges.

I'll leave the issue open for some days so you can add your new insights for future reference.

SashkaCosmonaut, 2015-12-07:
After a few experiments I have added a 200 Ohm resistor and programmed the ATtiny on 1 MHz and now the DHT returns data pretty stable. Anyway, in accordance to [manual page of the library](http://playground.arduino.cc//Main/DHTLib) I just reread the sensor after a one second if it returns the CHECKSUM error.

By the way, I have added the [answer to the Arduino Stackexchange](http://arduino.stackexchange.com/a/18500/14918). I think that there may be an incorrect way of programming the ATtiny85 with old version of Arduino IDE, thats why there was the problem with timeout counter. I will work on this problem a little bit later.

RobTillaart, 2015-12-07:
1 Mhz is pretty amazing and beyond what I expected as possible!

closed 2017-02-27 by RobTillaart