RobTillaart/Arduino #32: DHT11 lib on UDOO NEO

ckuehnel, 2016-02-19:
I got some compiling errors in UDOO NEO. I had the hope that it would work due to the DUE compatibility.
Do you have any hints to solve this issue?
Regards,
Claus

[image]

RobTillaart, 2016-02-20:
Hi Claus,

You should use the DHTstable library as that has no specific optimizations for AVR that I see in the error messages. The DHTstable can be found here  https://github.com/RobTillaart/Arduino/tree/master/libraries/DHTstable

Please let me know if it works.

ckuehnel, 2016-02-20:
Hi Rob
The first step was ok. But, there is an issue with F_CPU. See below.
Another hint?
Claus

[image]

RobTillaart, 2016-02-20:
For a 16 MHz Arduino UNO the value of F_CPU  is 16000000. So the DHTLIB_TIMEOUT, which is actually a max loop count, becomes 400. 

The UDOO NEO is a 1 GHZ processor if I recall correctly so the max loop count should be about 1e9/4e4 = 25000. 

So, you should replace the line to
 
#define DHTLIB_TIMEOUT 25000

You could make it e.g. 30000 or 35000 or larger as I do not know how efficient the code will be that is generated by the compiler.

ckuehnel, 2016-02-20:
The Cortex-M4 inside UDOO NEO is clocked with 200 MHz.
I included the line
`#define F_CPU 200000000UL`
before the line
`#define DHTLIB_TIMEOUT (F_CPU/40000)`
in dht.h and it works.
Thanks @RobTillaart

RobTillaart, 2016-02-20:
Good to hear it is solved,

closed 2016-02-20 by RobTillaart