RobTillaart/Arduino #33: float vs double

PaulStoffregen, 2016-02-25:
Any reason to use double rather than float?

On AVR, they are the same 32 bit single precision floats.

On ARM, double really is a (much slower) 64 bit double precision float.  On ARM chips with FPU, single precision is implemented in fast hardware, but double precision still requires slow library functions.

RobTillaart, 2016-02-25:
The most important reason is that double has more significant bits. The larger range is not that important e.g. sensor measurements are typical in the same range. 
That said, the class should be rewritten using templates so one can select float / double / triple (oops  that is Belgian beer ;) 
I'll keep the issue open until I have time to dive into my libs. They are also not compatible with the new lib mgr etc.

RobTillaart, 2017-07-26:
@Paul,
is this still an issue as I might have some time pretty soon to revert double to float.

PaulStoffregen, 2017-07-26:
Yes, still an issue.  I'd recommend using float everywhere, since it's much faster than double on all ARM and ESP boards.  Numerically, float gives consistency with AVR which implements double as 32 bit float.

Do you have any boards with a FPU?  I could send you a Teensy 3.6 for testing....

RobTillaart, 2017-07-26:
I have a few Teensy3.5's here for a project, that has an FPU too IIRC.
I'll update the RunningAverage tnight (it's 21:00 overhere)

RobTillaart, 2017-07-26:
RunningAverage is updated, refactored the code to make it cleaner again. 
RunningMedian will be next.

RobTillaart, 2017-07-31:
@PaulStoffregen,
Fixed most libs now, Complex and Fraction still could be done.

RobTillaart, 2018-04-02:
Fixed Fraction for the core code, only a constructor Fraction(double d) remained a.o. to solve constants like PI and EULER which are defined as double.

RobTillaart, 2018-04-02:
Fixed souble -> float in Complex too (now all are done. If problems arise, please make a new issue for the specific library. Sorry it took so int32_t  :)

closed 2018-04-02 by RobTillaart