RobTillaart/Arduino #34: wrong readings from time to time

bluuu, 2016-03-02:
From time to time (about 1 or 2 readings in 20 reading "pack") i getting wrong values like:

DHT22, 	OK,	33.3,	24.3,	5160
DHT22, 	OK,	33.8,	24.3,	5160
DHT22, 	OK,	34.4,	24.3,	5164
DHT22, 	OK,	46.9,	-24.3,	5208
DHT22, 	OK,	34.4,	24.3,	5160
DHT22, 	OK,	35.7,	24.3,	5256
DHT22, 	OK,	36.0,	24.3,	5204

RobTillaart, 2016-03-02:
Check the return value of the read function. Probably it gives a CRC/checksum error.
Do you use a pull up resistor?
Cable length?
Soldered or breadboard?
...

bluuu, 2016-03-02:
W dniu 2016-03-02 o 15:12, Rob Tillaart pisze:
> Check the return value of the read function. Probably it gives a
> CRC/checksum error.
> Do you use a pull up resistor?
> Cable length?
> Soldered or breadboard?
> ...

thanks for reply

These values are from a test sketch. There are no CRC error display 
while getting "minus" readings.

PullUP resistor engaged, cable lenght about 10cm.

Setup works well on other lib.

-- 
Adam Paluch
Ex.T Sp. z o.o. , Al. Mireckiego 22, Sosnowiec
tel. (32) 750 61 55 , tel. 602 68 35 73
www.ext.com.pl - radiolinie, teleinformatyka

RobTillaart, 2016-03-02:
Which version of the library are you using?  DHTstable is for most people best version to use.
Which processor/board?
Which test sketch used? Please post a link, as the lib has multiple.
You can add a check for the return value of the read function in a similar way as other test sketches.

bluuu, 2016-03-02:
W dniu 2016-03-02 o 17:30, Rob Tillaart pisze:

> Which version of the library are you using? DHTstable is for most people
> best version to use.

you are right. Stable marked version works well. Spotted issue is in 
0.1.21 version.
> Which processor/board?

328p, custom board

> Which test sketch used? Please post a link, as the lib has multiple.

test sketch from 0.1.21 lib

> You can add a check for the return value of the read function in a
> similar way as other test sketches.

thanks for help

closed 2016-03-02 by RobTillaart