RobTillaart/Arduino #35: Opening tag with multiple fields

TravisH84, 2016-03-14:
Hi there,

Just wondering how an opening tag can be created that contains multiple fields. I understand when it is A=B style tag, but not when it is A=B C=D like below;

```
<shoot>
          <pict id="[int]" bracket="[int]">
              [dateTime]
              <position pitch="[float]" yaw="[float]" roll="[float]"/>
          </pict>
          ...
      </shoot>
```

RobTillaart, 2016-03-16:
This code shows a workaround (with the 0.1.05 lib version)

```
//
//    FILE: XMLWriterTest.ino
//  AUTHOR: Rob Tillaart
// VERSION: 0.1.00
// PURPOSE: XML demo
//    DATE: 2016-03-16
//     URL:
//
// Released to the public domain
//

#include <XMLWriter.h>

XMLWriter XML(&Serial);

char buffer[24];

void setup()
{
  Serial.begin(115200);

  float p = random(100) * 1.0;
  float y = random(100) * 1.0;
  float r = random(100) * 1.0;

  shoot(p, y, r);
}

void shoot(float p, float y, float r)
{
  XML.header();
  XML.comment("Demo opening tag with multiple fields");
  XML.tagOpen("shoot");

  XML.tagStart("pict");
  XML.tagField("id", "12345");
  XML.tagField("bracket", "67");
  XML.tagEnd(NEWLINE, NOSLASH);

  XML.writeNode("Date", "20160316");

  XML.tagStart("position");
  // add float to string here
  XML.tagField("pitch", "4.59");
  XML.tagField("yaw", "2.78");
  XML.tagField("roll", "3.14");
  XML.tagEnd();

  XML.tagStart("/pict");
  XML.tagEnd(NEWLINE, NOSLASH);

  XML.tagClose();  // PICT

  XML.tagClose();  // SHOOT
}

void loop()
{
}
```

To better control the indentation the class needs these public functions (to be added in .h)
incrIndent() { _indent += _indentStep; };
decrIndent(); { _indent -= _indentStep; };
and probably a few more.

Time to think this over.

RobTillaart, 2016-03-16:
and the [DateTime] need another fix then  XML.writeNode("Date", "20160316");

TravisH84, 2016-03-16:
Thanks Rob, I will give it a go tonight. All coming together which is great; I intend to use the library to write XML to an SD card for a pan tilt camera robot and this library is invaluable. At the moment my testing is just to serial until the SD card module comes :).

Thanks again for the help, will be in touch.

TravisH84, 2016-03-23:
worked a treat on the xml file :)

RobTillaart, 2016-03-26:
I close this issue as there are no questions. I'll think how to improve the XML lib to support this scenario.

closed 2016-03-26 by RobTillaart