RobTillaart/Arduino #37: Library will print Data 1-3 times, then prints gibberish

Tibbiticus, 2016-04-06:
Simply using the example code from the arduino playground results in this. I'm not too fluent in code, maybe it's an error on my end. I'm using the Arduino uno and a DHT-22 sensor. Here's the example code I used and the outputs

`#include <dht.h>`
`dht DHT;`
`#define DHT11_PIN 4`
`//#define DHT21_PIN 5`
`#define DHT22_PIN 5`
`void setup()`
`{`
 `Serial.begin(115200);`
 `Serial.println("DHT TEST PROGRAM ");`
 `Serial.print("LIBRARY VERSION: ");`
 `Serial.println(DHT_LIB_VERSION);`
 `Serial.println();`
 `Serial.println("Type,\tstatus,\tHumidity (%),\tTemperature (C)");`
`}`

`void loop()`
`{`
  `// READ DATA`
  `Serial.print("DHT22, \t");`
  `int chk = DHT.read22(DHT22_PIN);`
  `switch (chk)`
  `{`
    `case DHTLIB_OK:`  
    `Serial.print("OK,\t");`
    `break;`
    `case DHTLIB_ERROR_CHECKSUM:` 
    `Serial.print("Checksum error,\t");` 
    `break;`
    `case DHTLIB_ERROR_TIMEOUT:` 
    `Serial.print("Time out error,\t");` 
    `break;`
    `default:` 
    `Serial.print("Unknown error,\t");` 
    `break;`
 ` }`
  `// DISPLAY DATA`
  `Serial.print(DHT.humidity, 1);`
  `Serial.print(",\t");`
  `Serial.println(DHT.temperature, 1);`

  `delay(1000);`
`}`

[http://imgur.com/bPj6xHx](url)

RobTillaart, 2016-04-06:
Please try the DHTstable library as the DHT 0.1.14 is not the most stable one.
The DHT 0.1.14 .. 21 are experimental versions for lower clock speeds by using AVR specific code.

Tibbiticus, 2016-04-06:
I appreciate it Rob, all of a sudden the code is working fine and I cannot for the life of me figure out what's different other than re-installing the arduino IDE. Even then things didnt function correctly for a while, I couldn't even get a simple `Serial.println("test");` to work properly.

RobTillaart, 2016-04-06:
could be bad wiring?

Tibbiticus, 2016-04-06:
I don't think so. I tried with an out-of-the-box Uno and i was having the same issues. I reinstalled the IDE, same issues, then 20 minutes later everything worked properly with 0 issues. Must have been cyberghosts (:p)

RobTillaart, 2016-04-06:
So it could be a connection that is only good when the Arduino is slightly warmed up?
As it is apparently not the code, I close the issue.

closed 2016-04-06 by RobTillaart