RobTillaart/Arduino #38: PCF8574 - Write to another pin while input is pulled low sets it low

sseptillion, 2016-04-28:
Hi,
I was just messing around with the PCF8574 and came across your library. The chip isn't that hard but if someone made a decent library already that always saves time. 

But I came across a bug if you use the device as input and output device on the same time (not necessarily the same pin) and you use the write() function. This function always read8() the data, set/clears the desired bit (call it bit1) and then transmits it again. But is another pin (call it bit2) is used as input (so it should be set high) but is pulled low externally (switch etc) when you read it, then you also read that state. And after the set/clear of bit1 you also write bit2 to be low. But after you then release the button the line will never go high aka, now you can't use that pin as input anymore.

Fix would be to have another variable which holds only the data you set. Then you use that byte when you do a write() instead of reading it.

RobTillaart, 2016-04-30:
Thanks for this feedback, 

Never used the chip in "mixed" mode but the scenario would be a bug indeed. The solution you propose would solve it. I have to verify if there are more side effects as there are more methods that have the "read modify write" pattern [shiftLeft, shiftRight, Toggle, Invert].

This would imply that the class would keep the _dataOut for the output lines as THE reference and never read it from the chip, only write it. The status of the input lines should always be read and should not be cached (except as previousValue).

As I have a (non published) version 0.1.05 under developement I need to refactor your proposed solution in. I do not know yet if the added button functions are needed when the bug is solved in the existing functions. It might take some time to fix.

Thanks again,

sseptillion, 2016-04-30:
Hi,

Yeah, I also only used them for basic stuff until now (mostly LCD's) but I started to mimic the PCF8574 on a Arduino (because damn, they're cheater then the PCF8574 itself!). And came across that neat solution in the datasheet. In my edit I also edited the toggle() and shiftLeft/Right().

Indeed, _dataOut is now THE reference and _data only as a previous read. 

I added the buttonRead() and buttonRead8() for when you want to use the same pin as output and input at the same time. If you connect a led (or other output) between Vcc and the pin and at the same time a button between the pin and GND you can use them both. Only downside is the led will also light up the led. But to read the button you have to make the output high for a short time to read the button. After that restoring the state to turn on/off (whatever the state was) the led again. Tested this and it works a charm :)

Good luck with it!

RobTillaart, 2016-05-02:
Hi,
Merged most of your code manually except the begin() as it is essentially a write()
Can you please check if it works to your needs?

Thanks,
Rob

sseptillion, 2016-05-03:
Looks pretty fine, I'll test when I'm home. But I spot 1 error. In buttonRead8() I used the begin() so that will give an error.

Yeah, begin() is just a write but I added it because I run into some errors when changing the Arduino code while not resetting the PCF8574. If you reset the Arduino it thinks all outputs are high but this might not be the case if you only reset the Arduino. So in style of other Arduino libraries I made a begin() function that sets all outputs to there default value so you never have an error when resetting the Arduino. Yes, you could also do that with write(0xFF) but I think a begin() is a more clear way of doing it. Most people are used to have a begin() in many classes they use (wire, serial etc) so it's a clear start.

RobTillaart, 2016-05-04:
OK I understand the begin() better now. The argument makes sense, however.
Question: Does the readButton8() really need the begin() ?  It sets all lines HIGH which may cause unwanted side effects depending on what is connected (optionally inverted) to the output lines.E.g. suppose a security valve is connected... ?
In short you never can be sure that 0xFF is the correct initialization...

sseptillion, 2016-05-04:
readButton8() does need the begin() aka all lines being pulled high when you want to read all pins as a button. If not all are buttons but eg a valve, then you indeed have a error. Then it would be better to read all the pins individual with buttonRead(). A compromise would be to give buttonRead8() a optional mask in which you can select the buttons to read.

RobTillaart, 2016-05-05:
I think the mask approach is the most powerful and flexible. The mask can have a default value of 0xFF.  I'll update to version 0.1.08 with the buttonRead8(uint8_t mask = 0xFF).

RobTillaart, 2016-05-05:
Can you code below a try?
Can you provide a simple test / demo sketch for the readButton functions? 
(one call with mas one without)

**PCF8574.h**
```
uint8_t readButton8(const uint8_t mask=0xFF);
```

**PCF8574.cpp**
```
//added 0.1.07 Septillion
uint8_t PCF8574::readButton8(const uint8_t mask)
{
    uint8_t temp = _dataOut;
    PCF8574::write8(mask); 
    PCF8574::read8();
    PCF8574::write8(temp);
    return _dataIn;
}
```

sseptillion, 2016-05-20:
Sorry for the delay. Had a bank holiday weekend and kind of forgot after that...

I tested the code and found an error. You should not write the mask but (mask | _dataOut). Otherwise a pin you don't want to read and that's set to high will be low during the read so is unwanted effected. 

Fixed the error and also added a variable to hold the buttonRead8() bitmask. This way you can set which pins are read by buttonRead8() just once and be done with it.

Also fixed the error in the note. Now we have a begin() function we can add Wire.begin() to that to fix all object creation order errors.

RobTillaart, 2016-05-21:
Thanks
updated version to 0.1.08 + some small additions.

RobTillaart, 2021-11-22:
Hi @septillion-git 

I was going through my libraries and came across PCF8574 and have a question..

You added the function below and I saw it still needs documentation - https://github.com/RobTillaart/PCF8574/issues/9

The function takes care that input lines are set high (work as pull up) to get appropriate values when reading.
(to be used in mixed read/write within one IC)

current code
```cpp
uint8_t PCF8574::readButton8(const uint8_t mask)
{
  uint8_t temp = _dataOut;
  PCF8574::write8(mask | _dataOut);  // read only selected lines
  PCF8574::read8();
  PCF8574::write8(temp);             // restore
  return _dataIn; 
}
```

Question is should the return value be masked too?

proposed code
```cpp
uint8_t PCF8574::readButton8(const uint8_t mask)
{
  uint8_t temp = _dataOut;
  PCF8574::write8(mask | _dataOut);  // read only selected lines
  PCF8574::read8();
  PCF8574::write8(temp);             // restore
  return _dataIn & mask;    // <<<<<<<<<
}
```

Then only the input lines asked for (defined by mask)  are returned.
Otherwise the return value may also include output lines that are HIGH and that makes processing possibly error prone.

Assuming you test on ``` if (readButton8(0x0F) > 0)``` to see if one of 4 input lines are pressed that could fail.
Proper usage would be ```if ((readButton8(0x0F) & 0x0F) > 0)** to mask out the output lines. 
Better do that in the library.

Opinion?
Rob

septillion-git, 2021-11-29:
Hi @RobTillaart 

Sorry, for the delay bit busy. Almost getting the key to my new house.

But damn, been a while. But yeah, I think you could mask the return as well. Don;t think that adding a pure input pin to the mask is an issue.

But doesn't the PCF8574 return 0 for a pin with a pressed button? Then I think returning it with a 1 for the masked bits makes more sense I think. Then there is a button pressed when it returns !255. Otherwise the return value in rest (no button press) is still based on the mask.

So
```C++
uint8_t PCF8574::readButton8(const uint8_t mask)
{
  uint8_t temp = _dataOut;
  PCF8574::write8(mask | _dataOut);  // read only selected lines
  PCF8574::read8();
  PCF8574::write8(temp);             // restore
  return _dataIn | ~mask;    // <<<<<<<<<
}
```

RobTillaart, 2021-11-30:
The bits set HIGH in the mask are the INPUT lines  

> But doesn't the PCF8574 return 0 for a pin with a pressed button? 

that is the right question. It depends on how the user connects the PCF as both ways are possible.
And because both are possible, there is no good way to mask it.  I will leave it out.

Thanks, and success with your new home!

PR #41 (2016-05-21): buttonRead8() with mask function implementation
Fixes #38 in more extend.

Adds a masking function to buttonRead8().

Adds examples for buttonRead() and buttonRead8().

Fixes wire.begin() object creation mis-order by moving it to begin().
closed 2021-11-30 by RobTillaart