RobTillaart/Arduino #42: I2C__EEPROM: Constructor - page size

owenduffy, 2016-06-02:
Hi, it seems your code doesn't actually implement the addressing for <32kbit, so that it seems it really only works with two byte addresses.

Wouldn't it be better to feed the constructor the page size rather than trying to guess it from the devicesize which seems to serve no other purpose?

eg:

I2C_eeprom::I2C_eeprom(const uint8_t deviceAddress)
{
    I2C_eeprom(deviceAddress, I2C_EEPROM_PAGESIZE);
}

I2C_eeprom::I2C_eeprom(const uint8_t deviceAddress, const unsigned int pageSize)
{
    _deviceAddress = deviceAddress;
    this->_pageSize = pageSize;
    this->_isAddressSizeTwoWords = pageSize >= 32;
}

Keep up the good work.

Cheers
Owen

PS: putting all this stuff in one repo might be convenient for you, but I think a repo per lib would be better for users.

RobTillaart, 2016-06-02:
Thanks for your comments. 
Your proposal would reduce the footprint of the lib (assumption) but people should know the page size, meaning they should read the data sheet. The current auto detect version keeps it simpler for beginners. 

The current solution is also able to work with any (larger) EEPROM without modification of the source code. So replacing one EEPROM for another would still result in a working sketch (OK to certain extend).

Still your proposal tickles the following idea:
```
I2C_eeprom::I2C_eeprom(const uint8_t deviceAddress, char * type)
{
_deviceAddress = deviceAddress;
this->_pageSize = lookup(type);     // function to be written
this->_isAddressSizeTwoWords = (_pageSize >= 32);
}
```
e.g.   I2C_eeprom EE(88, "LC24");

owenduffy, 2016-06-03:
Hi,
You didn't confirm that the code does not handle single byte addresses fully (ie the 11 bit address where 3bits are a block number in the I2C device address.

It seems to me that if you write code to use page operations to enhance endurance, the pagesize is something you must get right... even at the expense of code that is specific to a pagesize rather than generic across 24LCxxx devices.

Yes, but perhaps better on memory, you could add a set of defines:
#define 24LC512_PAGESIZE 128

So I could write:
I2C_eeprom ee(0x50,24LC512_PAGESIZE);

Owen

owenduffy, 2016-06-03:
Try:

Yes, but perhaps better on memory, you could add a set of defines:
#define PAGESIZE_24LC512 128

So I could write:
I2C_eeprom ee(0x50,PAGESIZE_24LC512);

Owen

MarkJeronimus, 2019-10-26:
Hi Rob,

I also struggled with the 2k<N<32kbit EEPROMS. I just started out with serial EEPROMS and picked the smallest from my bin, ST24C16 (2048 bytes) and it always wrapped around at 256 bytes. I found in the data sheet that the lower 3 bits of the I2C address are the upper memory address bits.

As this isn't supported, I consider this a bug for a feature request.

As this is the only library I can find that works with serial EEPROMS (search results generally get swamped with results concerning the AVR's built-in EEPROM) this is of even greater concern.

RobTillaart, 2019-10-27:
Hi ,
as I do not have all different EEPROM's it is hard for me to test them.
Do you have a link to the datasheet of the ST24C16?
If time permits this week I can have a look at it.

MarkJeronimus, 2019-10-29:
https://cdn-reichelt.de/documents/datenblatt/A300/24C08_16.PDF

RobTillaart, 2021-01-27:
@MarkJeronimus 
Apologies, way too late but just to let you know I am working on the fix for this.

I close the issue so it can be followed up here - https://github.com/RobTillaart/I2C_EEPROM/issues/17
The EEPROM class has got a repo of its own, like all libraries here.

closed 2021-01-27 by RobTillaart