RobTillaart/Arduino #43: read11 always return DHT_ERROR_TIMEOUT on Arduino2

GrichH, 2016-07-01:
i'm using your library for a school project and i'm actually test the sensor(DHT11) with your example project but i always get the dht_error_timeout and i can't figure out why. Could anybody help me pls?
Then only change i've made on the test Project is the pin in which the sensor is attached which is now set at 7.

Here it is the module which i'm using:
http://www.danilolarizza.com/wp-content/uploads/2012/12/dht11-300x300.jpg

RobTillaart, 2016-07-01:
Can you post the code you are using?

RobTillaart, 2016-07-01:
Which version of the library are you using? 
Best to use the DHT_stable library which is also in my repo as the latest versions are quite experimental and AVR specific.

GrichH, 2016-07-01:
i'm using the DHT_stable library version 0.1.13. The program is the one in the example folder btw the code is this`
#include <dht.h>

dht DHT;

#define DHT11_PIN 5

void setup()
{
  Serial.begin(115200);
  Serial.println("DHT TEST PROGRAM ");
  Serial.print("LIBRARY VERSION: ");
  Serial.println(DHT_LIB_VERSION);
  Serial.println();
  Serial.println("Type,\tstatus,\tHumidity (%),\tTemperature (C)");
}

void loop()
{
  // READ DATA
  Serial.print("DHT11, \t");
  int chk = DHT.read11(DHT11_PIN);
  switch (chk)
  {
    case DHTLIB_OK:  
		Serial.print("OK,\t"); 
		break;
    case DHTLIB_ERROR_CHECKSUM: 
		Serial.print("Checksum error,\t"); 
		break;
    case DHTLIB_ERROR_TIMEOUT: 
		Serial.print("Time out error,\t"); 
		break;
    default: 
		Serial.print("Unknown error,\t"); 
		break;
  }
  // DISPLAY DATA
  Serial.print(DHT.humidity, 1);
  Serial.print(",\t");
  Serial.println(DHT.temperature, 1);

  delay(2000);
}`

RobTillaart, 2016-07-01:
Do you have a logic analyzer to check if there is any signal?
A dht_error_timeout  error can occur if the sensor is broken.

What board is connected to the DHT11?
if it is a 3.3Volt board it might not have enough power.

RobTillaart, 2017-02-27:
Please reopen issue if not solved.

closed 2017-02-27 by RobTillaart