RobTillaart/Arduino #45: Attempted compile on nodemcu using playground sample.

cjwolff, 2016-09-09:
Dear Mr. Tillaart,

After many problems with the DHT11 I thought I'd try your example sketch from the playground using your library:

`//
//    FILE: dht_test.ino
//  AUTHOR: Rob Tillaart
// VERSION: 0.1.07
// PURPOSE: DHT Temperature & Humidity Sensor library for Arduino
//     URL: http://arduino.cc/playground/Main/DHTLib
//
// Released to the public domain
//

#include <dht.h>

dht DHT;

#define DHT11_PIN 14
#define DHT21_PIN 5
#define DHT22_PIN 6

void setup()
{
  Serial.begin(115200);
  Serial.println("DHT TEST PROGRAM ");
  Serial.print("LIBRARY VERSION: ");
  Serial.println(DHT_LIB_VERSION);
  Serial.println();
  Serial.println("Type,\tstatus,\tHumidity (%),\tTemperature (C)");
}

void loop()
{
  // READ DATA
  Serial.print("DHT22, \t");
  int chk = DHT.read22(DHT22_PIN);
  switch (chk)
  {
    case DHTLIB_OK:  
    Serial.print("OK,\t"); 
    break;
    case DHTLIB_ERROR_CHECKSUM: 
    Serial.print("Checksum error,\t"); 
    break;
    case DHTLIB_ERROR_TIMEOUT: 
    Serial.print("Time out error,\t"); 
    break;
    default: 
    Serial.print("Unknown error,\t"); 
    break;
  }
  // DISPLAY DATA
  Serial.print(DHT.humidity, 1);
  Serial.print(",\t");
  Serial.println(DHT.temperature, 1);

  delay(1000);

    // READ DATA
  Serial.print("DHT21, \t");
  chk = DHT.read21(DHT21_PIN);
  switch (chk)
  {
    case DHTLIB_OK:  
    Serial.print("OK,\t"); 
    break;
    case DHTLIB_ERROR_CHECKSUM: 
    Serial.print("Checksum error,\t"); 
    break;
    case DHTLIB_ERROR_TIMEOUT: 
    Serial.print("Time out error,\t"); 
    break;
    default: 
    Serial.print("Unknown error,\t"); 
    break;
  }
  // DISPLAY DATA
  Serial.print(DHT.humidity, 1);
  Serial.print(",\t");
  Serial.println(DHT.temperature, 1);

  delay(1000);

  // READ DATA
  Serial.print("DHT11, \t");
  chk = DHT.read11(DHT11_PIN);
  switch (chk)
  {
    case DHTLIB_OK:  
    Serial.print("OK,\t"); 
    break;
    case DHTLIB_ERROR_CHECKSUM: 
    Serial.print("Checksum error,\t"); 
    break;
    case DHTLIB_ERROR_TIMEOUT: 
    Serial.print("Time out error,\t"); 
    break;
    default: 
    Serial.print("Unknown error,\t"); 
    break;
  }
 // DISPLAY DATA
  Serial.print(DHT.humidity,1);
  Serial.print(",\t");
  Serial.println(DHT.temperature,1);

  delay(1000);
}
//
// END OF FILE
//`

The compile errored out with the following:  Do you have any suggestions?  Thank you.

`In file included from /Users/christopherwolff/Documents/Arduino/libraries/DHT2/dht.h:18:0,
                 from /Users/christopherwolff/Documents/Arduino/libraries/DHT2/dht.cpp:41:
/Users/christopherwolff/Documents/Arduino/libraries/DHT2/dht.cpp: In member function 'int8_t dht::_readSensor(uint8_t, uint8_t, uint8_t)':
/Users/christopherwolff/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/Arduino.h:226:63: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in initialization
 #define portInputRegister(port)     ((volatile uint32_t*) &GPI)
                                                               ^
/Users/christopherwolff/Documents/Arduino/libraries/DHT2/dht.cpp:121:29: note: in expansion of macro 'portInputRegister'
     volatile uint8_t *PIR = portInputRegister(port);
                             ^
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).`

RobTillaart, 2016-09-09:
Hi Christopher,

The 1.0.7 version is rather old (obsolete in fact).
Please use the stable version found here https://github.com/RobTillaart/Arduino/tree/master/libraries/DHTstable
and see if the problems still exist.

It is easier for me to support. That said I never used the lib on a esp8266 myself.

Rob

cjwolff, 2016-09-09:
Dear Mr. Tillaart, thank you for your prompt and kind reply.  

It appears i'm using 1.21 as described in the comments:
//
//    FILE: dht.cpp
//  AUTHOR: Rob Tillaart
// VERSION: 0.1.21
// PURPOSE: DHT Temperature & Humidity Sensor library for Arduino
//     URL: http://arduino.cc/playground/Main/DHTLib
//

I've tried probably every combination of Google/github/nodemcu variances and it seems like my placement of the Delay() statement truly has an impact on whether the entire sketch works or not.  I'll be happy to coordinate with you as much as possible to try to resolve this issue and broaden the support for the NodeMCU community (if it truly is a NodeMCU issue.

Sincerely,
Christopher Wolff

RobTillaart, 2016-09-09:
The delay() should not affect the "compilability" of the source code. It can have an effect on the runtime behavior of the program.

The most stable version of my lib is the **DHTstable** version. (0.1.13)  Versions after that including the 0.1.21 are optimisations specific for AVR and low clock (8 & 4 Mhz).  

You should download this version 
https://github.com/RobTillaart/Arduino/tree/master/libraries/DHTstable
NOT the DHTlib version.

please give it a try.

Do you have a link to a noceMCU toolchain?

RobTillaart, 2016-09-09:
BTW just Rob is ok ;)

cjwolff, 2016-09-12:
Dear Rob,

I found a link to a nodeMCU toolchain at http://www.esp8266.com/wiki/doku.php?id=toolchain#using_the_arduino_ide under https://github.com/pfalcon/esp-open-sdk.

I'm going to try the DHTstable version now and will post results momentarily.

Sincerely,
Christopher

cjwolff, 2016-09-12:
Rob,

I was able to get the sketch to compile with the DHTstable version; however, I still get only one reading from the DHT11.  Thank you for your input on this issue:

`WiFi connected
IP address: 
192.168.0.8
Netmask: 
255.255.255.0
Gateway: 
192.168.0.1
Humidity: 37.00 %	Temperature: 23.00 *C 73.40 *F	Heat index: 22.32 *C 72.18 *F
Object: 149cm
Humidity: nan %	Temperature: nan *C nan *F	Heat index: nan *C nan *F
Object: 153cm
Humidity: nan %	Temperature: nan *C nan *F	Heat index: nan *C nan *F
Object: 154cm
Humidity: nan %	Temperature: nan *C nan *F	Heat index: nan *C nan *F
Object: 148cm
`

RobTillaart, 2016-09-12:
Please add a delay of minimal 2 seconds between the readings. 
The DHT11's internal processor needs time to "recalculate the sensor" after a read.

RobTillaart, 2017-02-27:
PLease reopen if issue is not solved.

closed 2017-02-27 by RobTillaart

areman95, 2018-05-23:
may i know how to download this https://github.com/RobTillaart/Arduino/tree/master/libraries/DHTstable  ?

RobTillaart, 2018-05-23:
All my libs are in one git archive - so you need to download them all OR download the individual files of a library individually.

soheilpaper, 2020-09-15:
use this post instead of that library:
https://desire.giesecke.tk/index.php/2018/01/30/esp32-dht11/