RobTillaart/Arduino #46: DHT21 without delay

SayedBaqer, 2016-09-09:
is there any way to use this library as non blocking, where i need to avoid delay in the arduino sketch where i am beginner in programing  ?
Best regards,

RobTillaart, 2016-09-09:
No, the library is blocking  (between 5-6 milliseconds) for one read.

Please check this thread on the Arduino forum - http://forum.arduino.cc/index.php?topic=351576.0 -
for links to a non-blocking version.

closed 2016-09-09 by RobTillaart

SayedBaqer, 2017-09-28:
Hi,
I need your help if you can,
I need to do PCB design, i have some component need to fit with this pcb board if you can i will be happy, and let me know how much you can do the design and if need to manufacture low quantities 2-10 pcs per time price as well 
10x10cm expected with two layer

Best regards,
Sayed Baqer Mohammed Alseba

RobTillaart, 2017-09-28:
Please send all information about this project to my gmail address. I have too little information to say something that makes sense. What are the requirements, what designtool / file format you need, how do you intent to test, delivery address (I do not know where you live but sending from Netherlands can be costly), etc etc whatever might be helpful.

SayedBaqer, 2017-09-28:
May i get it?

Best regards,
Sayed Baqer Mohammed Alseba

RobTillaart, 2017-09-28:
rob.tillaart@gmail.com