RobTillaart/Arduino #49: Error during conversion

mulasmarco, 2016-12-05:
In file included from /Users/marco/Documents/Arduino/libraries/DHTlib/dht.h:19:0,
                 from /Users/marco/Documents/Arduino/libraries/DHTlib/dht.cpp:41:
/Users/marco/Documents/Arduino/libraries/DHTlib/dht.cpp: In member function 'int8_t dht::_readSensor(uint8_t, uint8_t, uint8_t)':
/Users/marco/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/Arduino.h:226:63: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in initialization
 #define portInputRegister(port)     ((volatile uint32_t*) &GPI)
                                                               ^
/Users/marco/Documents/Arduino/libraries/DHTlib/dht.cpp:121:29: note: in expansion of macro 'portInputRegister'
     volatile uint8_t *PIR = portInputRegister(port);
                             ^
exit status 1

using Mac OS 10.12.1 - Arduino IDE 1.6.13

RobTillaart, 2016-12-06:
Which version of the dhtlib are you using?

mulasmarco, 2016-12-08:
I used the last one downloaded here

mulasmarco, 2016-12-08:
Oh great! I'll try the new one asap! :)

closed 2016-12-17 by RobTillaart