RobTillaart/Arduino #55: Alghoritm for the DHT22 could be wrong

samm-git, 2017-01-24:
In the DHT21 code i see such lines:

```
 Buf[0] &= 0x03;
 Buf[1] &= 0x83;
```

It seems that this assumption is not correct, and affect results on my dht21. Any comments why it is there?

RobTillaart, 2017-01-24:
The humidity can be max 1000(dec) which equals 11.1110.1000 (bin), 
which means that for buf[0] the highest bits should always be zero.
by masking them flipped bits are ignored.

The same is true for the temperature which uses one bit for sign
and can range from -40 .. 80 (according to datasheets I have) so some bits are never set.

That said, these lines are not mandatory.
Do you have information that the DHT21 has temperature-range above 100C ?
Then there is indeed an error and the mask should become 
```
buf[2] &= 0x87;
```

note: in fact to be consequent, I could improve the masking in these lines too
```
    humidity = (bits[0]*256 + bits[1]) * 0.1;
    temperature = ((bits[2] & 0x7F)*256 + bits[3]) * 0.1;
```

thanks

samm-git, 2017-01-24:
buf[0] and buf[1] are both about humidity, so line "Buf[1] &= 0x83;" affects humidity and not temperature, as far as i could see. I found this bug when was writing code for LUA and used your lib as an initial reference.

RobTillaart, 2017-01-24:
but the code refers to buf[0] and buf[2] or do I miss anything?
https://github.com/RobTillaart/Arduino/blob/master/libraries/DHTlib/dht.cpp

can you give a link?

samm-git, 2017-01-24:
Hm, sorry, it is probably my fault :( Old version of the code, or bad copypaste. Now i checked that it works fine and in my LUA version it was wrongly set to [1]. Sorry for that noise and thanks again for the good code.

RobTillaart, 2017-01-24:
OK, do you have a datasheet of the DHT21?

samm-git, 2017-01-24:
I was using this one: http://mikroshop.ch/pdf/DHT21.pdf

closed 2017-04-20 by RobTillaart