RobTillaart/Arduino #58: Impossible but valid speed

kb-, 2017-02-22:
Hi,

I added higher frequencies to the list when I saw my 24LC256 EEPROM was detected up to 800 kHz. The code returned valid for any speed (except 1 MHz), even those higher than the clock of my Mega 2560 ...

TIME	DEC	HEX		50	100	200	250	400	500	800	1000	1600	3200	4000	8000	16000	32000	64000	[KHz]
----------------------------------------------------------------------------------------------------------------------------------------------------------------
2373	0	0x00		.	.	.	.	.	.	.	.	.	.	.	.	.	.	.
2378	1	0x01		.	.	.	.	.	.	.	V	.	.	.	.	.	.	.
2383	2	0x02		.	.	.	.	.	.	.	.	.	.	.	.	.	.	.

2745	78	0x4E		.	.	.	.	.	.	.	.	.	.	.	.	.	.	.
2750	79	0x4F		.	.	.	.	.	.	.	.	.	.	.	.	.	.	.
2754	80	0x50		V	V	V	V	V	V	V	.	V	V	V	V	V	V	V

RobTillaart, 2017-02-24:
Never tried this but I assume the hardware cannot handle the higher speeds for all those higer speeds.
I expect values are truncated someway so although you specify e.g. 64000, the register will hold another typically smaller value which the hardware is capable to handle. This assumption can be verified by reading back the value from the registers and check.

kb-, 2017-02-25:
I tried the same with a Nano Zero (SAMD21G @48MHz). Any value is returned valid up to the last one I tested (64 MHz...) for another Nano Zero's address. Everything is valid, except 16 MHz for the 24LC256 EEPROM. The speed test seems buggy.
I haven't figured out how to read back the register.

RobTillaart, 2017-02-27:
The speeds of the multispeed scanner are based upon those supported by the original 16 Mhz Arduino UNO/MEGA. 
To read back the registers you need to dive into the low level library code and or datasheet to see if it could be done. I am not familiar with the Nano Zero, and have no time to dig into this.

kb-, 2017-03-03:
I checked what's going on with a logic analyzer.

800 kHz address 60: fine
[image]

1 MHz address 60: NAK and frequency is unstable
[image]

1 MHz  address 59: read as 29 and NAK in logic analyzer, seen as valid by the Mega
[image]

1.6 MHz (and up): The Mega falls back to the lowest speed (30 kHz)
[image]

Results in logic analyzer for 50, 100, 200, 250, 400, 500, 800, 1000, 1600, 3200, 3400, 4000 kHz;  address 60 being a device 
[image]

RobTillaart, 2017-03-06:
The low level wire core can be found here (or similar path; note AVR hardware specific)
C:\Program Files (x86)\Arduino.1.8.1\hardware\arduino\avr\libraries\Wire\src

look into utility\twi.c - critical function is 
```C++
void twi_setFrequency(uint32_t frequency)
{
  TWBR = ((F_CPU / frequency) - 16) / 2;
  
  /* twi bit rate formula from atmega128 manual pg 204
  SCL Frequency = CPU Clock Frequency / (16 + (2 * TWBR))
  note: TWBR should be 10 or higher for master mode
  It is 72 for a 16mhz Wiring board with 100kHz TWI */
}
```

So this is the reverse formula: SCL Frequency = CPU Clock Frequency / (16 + (2 * TWBR))

```C++
uint32_t getFrequency()
{
  return ((uint32_t)F_CPU) / (16 + (2 * TWBR));
}
```
F_CPU = 16.000.000, 
so a frequency of 100.000 => TWBR = (160 -16)/2 = 72
SCL Frequency = 16.000.000 / (16 + 2 * 72) = 100.000 seems ok

frequency of 1.000.000 => TWBR == 0; 
SCL Frequency = 16.000.000 / (16 + 2 * 0) = 1.000.000 seems ok 
but datasheet states TWBR should be 10 or higher.

frequency of 1.600.000 => TWBR = (10 - 16)/2 = -3   ==> 252  (!! 1 unsigned byte register !! )
calculate back
SCL Frequency = 16.000.000 / (16 + 2 * 252) = 30769 is pretty much the 30Khz you see 

Epilogue
Datasheet states TWBR should be 10 or higher. so frequency should be max:
SCL Frequency = 16.000.000 / (16 + 2 * 10) = 444444 so infact the 800KHz is already over the edge.

RobTillaart, 2017-03-06:
Note: 800KHz gives TWBR = 4, which is the highest value > 0 that gives no rounding errors in the math

closed 2017-04-20 by RobTillaart