RobTillaart/Arduino #60: DHT11 start signal to short, delayMicroseconds waits max 16ms @16Mhz

fesselk, 2017-03-02:
delayMicroseconds on AVR has a maximum waittime which depends on F_CPU @16MHz it's (2^16)/4 µs ~= 16000 µs.

16000 µs is shorter than 18ms and due to the calculation the actual delay is even shorter.

see (avr) wiring.c:

delayMicroseconds uses an unsigned int (16Bit) to count down and calculates how often it needs to do this 4 cycles.

RobTillaart, 2017-03-02:
Good point, which library are you referring too? probably all of them.

fesselk, 2017-03-02:
sorry since i only looked at DHTlib  (not stable) i did not check the other.

its an issue with the long "wakeupdelay" for dht11.

I glanced shortly over the other DHT-libraries  they all seem to use delay for the wakeupdelay.

If there are other uses of delayMicroseconds which wait longer than 16ms they are hit by the same thing.

for safety consider as border 10ms (F_CPU@ 24Mhz)   formula: max_delay = (2^16)/(F_CPU/4) µs   

delayMicroseconds is defined for  1, 8, 12, 16, 20 or 24 MHz (other F_CPU  map to the next lower one)

RobTillaart, 2017-03-06:
Fixed it the following way
```C++
    if (wakeupDelay > 8) delay(wakeupDelay);
    else delayMicroseconds(wakeupDelay * 1000UL);
```
Thanks again!

closed 2017-03-06 by RobTillaart

fesselk, 2017-03-07:
i am not sure but i do not think the optimizer is able to solve the `if` while it is compiled ( since it depends on function input)

i am not sure why u want to keep delayMicroseconds. -> investigate

//--- 
i guessed the reason why u changed the `delay `for `delayMicroseconds` in the first place
seems like u want to circumvent the low F_CPU problem of `millis()`, which is used for delay internally and does steps lager than 1 ms at low F_CPU.
 <- while the fact about `millis()` is true `delay()` uses `micro()` so no problem there -> investigate further
---//

So i did a review of the issues. Turns out u changed delay to delayMicroseconds on 
11 Oct 2015 to fix: "Fix sensor reading for use within Realtime OS environments"
(millis() and mircos() (and hence delay()) need interrupts to work which where turned off by the issue author to deal with the preemptiv nature of the used RTOS;
basically disabling the Arduino clock; --- waiting multiple milliseconds blocking in an RTOS seems intuitive to me;)  

-If you want to do real busy waiting u might consider using `_delay_ms` (part of avr_libc).
-Arduinos `delay` also calls `yield` which might be used by other people to do some parallelism on the Arduino.
-The NilRTOS has its own sleep function to keep the system running while waiting that might be used for NilRTOS.
-The NiRTOS bug is atm unnfixed for F_CPU = 16 MHz (previously it had the bug described above).
-For ISR free systems  `_delay_ms` is would work since avr_libc does not use interrupts itself
-alternative `for( int i = wakeupDelay; i ;i--) delayMicroseconds(1000)`
-For Arduino in its default configuration delay works best.