RobTillaart/Arduino #63: No timeout needed after Wire.requestFrom

Koepel, 2017-04-18:
In the files I2C_eeprom.cpp and MCP4725.cpp there is a timeout after the Wire.requestFrom. That is not needed. There is no such thing as waiting for data with Wire.available. When the Wire.requestFrom returns, the I2C transaction has completely finished and the received data is in a buffer in the Wire library. The Wire.available returns the number of bytes in that buffer, and that number is never increased with new data because the I2C was already completely finished.

RobTillaart, 2017-04-19:
That's 4 years ago, let me think why I added that 

The reason for the timeout is not the Wire library code but because the EEPROM could be blocking sometimes. And that would be exactly the case when less than the requested length bytes are available. Although the code is functionally not wrong, it is less efficient as possible so I will rewrite code a.s.a.p.

Koepel, 2017-04-19:
I don't understand what the EEPROM will do when it is blocking.

When a sensor or eeprom needs some time and does not acknowledge to its address, then the Wire.requestFrom could be repeated until it returns the same number of bytes that is requested or until something that is not zero is returned. A number of chips act this way (being not visible on the I2C for a short time).
When a slave uses clock pulse stretching, then the Arduino code only needs a single Wire.requestFrom, and of course without any delay after the Wire.requestFrom.

It was not really "wrong", just like this example is not "wrong":
```
void myFunction()
{
  char buffer[10];
  while( sizeof( buffer) < 10);  // wait until buffer is created on the stack.
  ...
}
```
[(http://www.arduinoforum.nl/viewtopic.php?f=31&t=2247#p16887)](http://www.arduinoforum.nl/viewtopic.php?f=31&t=2247#p16887)

RobTillaart, 2017-04-20:
OK, new code could have been shorter (wanted to fix too quickly)
```
Wire.requestFrom(_deviceAddress, length);
uint8_t cnt = 0;
uint8_t n = Wire.available();
while ( (cnt < n) && (cnt < length))
```
could have been
```
uint8_t returnedBytes = Wire.requestFrom(_deviceAddress, length);
uint8_t cnt = 0;
while (cnt < returnedBytes)
```
I'll update asap.

closed 2017-07-24 by RobTillaart