RobTillaart/Arduino #66: Not working on NodeMCU when pin >= 8

flurl, 2017-05-12:
I use your library on a esp8266 based NodeMCU. I noticed that it doesn't work on any GPIO pin numbered > 7.
I found out, that this is due to the data type of the [state](https://github.com/RobTillaart/Arduino/blob/master/libraries/DHTlib/dht.cpp#L110), [pstate](https://github.com/RobTillaart/Arduino/blob/master/libraries/DHTlib/dht.cpp#L111), [bit](https://github.com/RobTillaart/Arduino/blob/master/libraries/DHTlib/dht.cpp#L120) and [PIR](https://github.com/RobTillaart/Arduino/blob/master/libraries/DHTlib/dht.cpp#L122) variables in _readSensor(), which need to be uint32_t, since the portInputRegister is `#define portInputRegister(port) ((volatile uint32_t*) &GPI)` in Arduino.h

I guess, changing those variables to uint32_t wouldn't do any harm on boards with smaller input registers, will it?

RobTillaart, 2017-05-12:
which library are you referring to?

tofrnr, 2017-05-27:
hello,
I run into the same issue, both for Arduino Due and ESP8266.
But I am just a beginner to Arduino and I do not understand what to do now or what DHTstale means.

RobTillaart, 2017-05-27:
DHTStable is the stable version of the library, to be found at the following URL
- https://github.com/RobTillaart/Arduino/tree/master/libraries/DHTstable

closed 2017-05-27 by RobTillaart