RobTillaart/Arduino #73: choose between I2C lines?

pwspania, 2017-07-13:
Is there a way to specify which I2C to use? wire1 for example on a due?

RobTillaart, 2017-07-13:
Some libraries do others don't.  Most of mine don't.
Which library are you referring to?

waspinator, 2017-07-13:
I'm trying to use your MCP4725 dac library.

waspinator, 2017-07-14:
Hmm I tried that, there must be something more to it. Works fine on the first I2C, but not on the second. I had to remove references to `TWBR`, but I'm not sure if that is related, as it still works on the first port.

RobTillaart, 2017-07-14:
TWBR is AVR specific I guess, there is a call something like Wire.setClock(bitrate)
Have no due installed, maybe it is called differently, (I picked Wire1 from Teensy which is also ARM based)

RobTillaart, 2017-07-17:
Updated  https://github.com/RobTillaart/Arduino/tree/master/sketches/MultiSpeedI2CScanner  to support Wire1 and Wire2 for the Teensy3.5 as I could test that one. Might give some insights.

I did not find time yet to install the due board to see how wire1 is implemented (conditionally or not)

waspinator, 2017-07-18:
looks like it should be `Wire1` for the Due, but it wasn't working for me. I might be over looking something though.

https://github.com/arduino/ArduinoCore-sam/blob/master/libraries/Wire/src/Wire.h

RobTillaart, 2017-07-18:
Wire1 for the Due is only defined if  WIRE_INTERFACES_COUNT > 1, 
but it is set in variants/arduino_due_x/variant.h to 2 so Wire1 should be known.
What error message do you get if you compile the library with Wire1  ?

RobTillaart, 2017-07-31:
@waspinator 
Is this issue solved now? if so how?

waspinator, 2017-08-02:
It doesn't show any errors compiling, just doesn't seem to communicate with the device. The logic in your scanner doesn't detect any additional wire channels implemented in a Due. My modified version of your scanner does pick the DAC up though, but I haven't had a chance to try out your MCP4725 library again. I submitted a pull request to your scanner sketch to add Arduino Due support.

RobTillaart, 2017-08-03:
Pull request #75 is merged

closed 2017-08-03 by RobTillaart