RobTillaart/Arduino #76: Program Menu not respond

helmiwm, 2017-08-11:
[image]

Hi I have a problem 
when I ran the program in the Arduino IDE, 
then I open the serial monitor, the menu does pop up (see pic)

 but i when type 's' or type '?'  on serial monitor, it does not respond.  
any idea why?

I use Arduino Mega trying to configure my 16x2 LCD i2c address 
thx

RobTillaart, 2017-08-12:
Sorry for late response.

Serial speed is ok otherwise no menu would be on screen.

Two questions
.. do other conmands work or do they give no response?
.. did you press enter or return after the command? That is needed with the
build in serial monitor.

Regards
Rob

helmiwm, 2017-08-17:
Thanks Rob for replying

I did press enter after typing the command in serial monitor. but  then
there is no response

I have tried to type other command and it did not respond as well

RobTillaart, 2017-08-17:
Have you tried a different cable? 

Do other sketches that use serial work? (e.g. simple echo sketch below)

```void setup()
{
  Serial.begin(115200);
  Serial.println("start echo");
}

void loop()
{
  if (Serial.available())
  {
    Serial.write(Serial.read());
  }
}
```

closed 2017-11-03 by RobTillaart