RobTillaart/Arduino #77: AM2305/DHT44 does not work with Nodemcu and DHTstable

sajjadesz, 2017-09-01:
Hi. 
I already tested this library for arduino uno + AM2305/DHT44. It works perfectly but whenever I use this sketch for nodemcu with or with not pull-up resistor on pin 2, it does not work and get time out error. I,m using DHTstable library 0.2.0 version which siad it,s the latest stable library for avr and arm.
Thanks!

RobTillaart, 2017-09-05:
I do not have a nodemcu to test with. Is this a 3.3V device?

mgulsoy, 2017-09-07:
Yes. That is a 3.3v device.

RobTillaart, 2017-09-07:
You might need a bidirectional 3.3 - 5V convertor between the datalines?

sajjadesz, 2017-09-15:
It works!!!!!!!!!Thanks a lot for your attention.
It works perfectly without any pull up resistor on pin 5 (D1 on NodeMcu development board)

RobTillaart, 2017-09-15:
How did you make it work?

closed 2017-09-22 by RobTillaart