RobTillaart/Arduino #80: checksum error for DTH11

Nikolo, 2017-09-18:
https://github.com/RobTillaart/Arduino/blob/83ef3dc6367064d3ec38130524989ba81ef3e86c/libraries/DHTlib/dht.cpp#L66

first and third not always 0. Please fix for uint8_t sum = bits[0] + bits[1]+ bits[2] + bits[3];

RobTillaart, 2017-09-18:
Hi Nikolo,
For the DHT11 these values should always be zero as the humidity and temperature have no decimal part in the DHT11 according to the datasheet. OK the datasheet indicates that these bytes should taken into account for the checksum 
Can you give a temperature or humidity for which this is not the case? 
Or does the DHT11 suddenly support a decimal part? 

This is the first time in 6 years that this is reported as a problem. 
(fix is not difficult and would hardly impact size, but if not needed I would prefer not to do it)

Nikolo, 2017-09-20:
Hi!
Yes i have problem with DHT11. But when I added bits 1 and 3 into the check sum, all was normalized.
I did not find in datasheet where would be said that 1 and 3 bytes always 0, only in an example.
But in the same example it is indicated that the check sum must be counted on all bytes.
If it is interesting that I can send the photo of sensor and load any your example for the checking.

RobTillaart, 2017-09-20:
I will update it soon

RobTillaart, 2017-09-20:
Fixed in DHTlib and DHTStable

closed 2017-09-20 by RobTillaart