RobTillaart/Arduino #83: XMLWriter to String?

beicnet, 2017-12-07:
Hi there,

Please tell me, is it possible to output generated XML to String rather than Serial?

Is so, how can I achieve that?

Regards,
Viktor

RobTillaart, 2017-12-07:
Hi Viktor,

Not directly, you can however create a class e.g. PrintString that derives from Print and implement its interface to add all to an internal char buffer. That internal buffer could be exposed and used to create the String object. The XML writer could use an instance of that PrintString class as parameter. 

The PrintString class could look like (Quick and Dirty implementation)

``` C++
//
//    FILE: PrintString.h
//  AUTHOR: Rob Tillaart
// VERSION: 0.1.00
// PURPOSE: Class that captures prints into a string 
//    DATE: 2017-12-07
//     URL: 
//
// Released to the public domain
//

#ifndef PrintString_h
#define PrintString_h

#include "Arduino.h"
#include "Print.h"

#define PRINTSTRING_VERSION "0.1.0"
#define BUFFERSIZE      256

class PrintString: public Print
{
  public:
    PrintString() {};
    
    size_t write(uint8_t c)
    {
      if (index < BUFFERSIZE-1)
      {
        buffer[index] = c;
        index++;
        buffer[index] = '\0';
        return 1;
      }
      return 0;
    }
    
    void clear()
    {
      index = 0;
      buffer[index] = '\0';
    }

    int free() { return (BUFFERSIZE - index); }
    
    char * getBuffer() { return buffer; }

  private:
    char buffer[BUFFERSIZE];
    int index = 0;
};
#endif
// -- END OF FILE --

```

Usage in sletch below shows reset() and free() functions, 

``` C++
//
//    FILE: printString.ino
//  AUTHOR: Rob Tillaart
// VERSION: 0.0.1
// PURPOSE: demo
//
// HISTORY:

#include "PrintString.h"

#include "XMLWriter.h"

PrintString ps;
XMLWriter XML(&ps);
char buffer[24];

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);

  Serial.println(ps.free());
  ps.println("Hello World");
  Serial.println(ps.free());
  ps.println(3.14159265, 4);
  Serial.println(ps.free());
  Serial.println(ps.getBuffer());

  ps.clear();
  ps.println(3.14159265, 4);
  ps.println("Hello World");
  Serial.println(ps.getBuffer());

  ps.clear();

  XML.header();

  XML.comment("Weather in Nebraska");
  XML.tagOpen("Data");
  XML.writeNode("Date", "20131106");
  XML.writeNode("Time", "11:42");
  XML.writeNode("Temp", "23.4");
  XML.writeNode("Humi", "50%");
  XML.writeNode("Rain", "10mm");
  XML.writeNode("Sun", "40");
  XML.tagClose();

  XML.tagClose();

  Serial.println(ps.getBuffer());
  Serial.println(ps.free());

}

void loop()
{

}
``` 
Note instead of the fixed char array you could consider using a String object internally. However the dynamic allocation could cause problems. Not tested.

beicnet, 2017-12-07:
Thank you for your quick response, but I got some errors while trying to compile:

```
Arduino: 1.8.2 (Windows 7), Board: "WeMos D1 Mini Pro, 80 MHz, 921600, 16M (15M SPIFFS)"

C:\Users\beic\Documents\Arduino\libraries\XMLWriter\XMLWriter.cpp: In member function 'void XMLWriter::tagField(const char*, int8_t, uint8_t)':

C:\Users\beic\Documents\Arduino\libraries\XMLWriter\XMLWriter.cpp:140:39: error: call of overloaded 'tagField(const char*&, long int, const uint8_t&)' is ambiguous

     tagField(field, (long) value, base);

                                       ^

C:\Users\beic\Documents\Arduino\libraries\XMLWriter\XMLWriter.cpp:140:39: note: candidates are:

C:\Users\beic\Documents\Arduino\libraries\XMLWriter\XMLWriter.cpp:119:6: note: void XMLWriter::tagField(const char*, uint8_t, uint8_t)

 void XMLWriter::tagField(const char *field, const uint8_t value, const uint8_t base)

      ^

C:\Users\beic\Documents\Arduino\libraries\XMLWriter\XMLWriter.cpp:124:6: note: void XMLWriter::tagField(const char*, uint16_t, uint8_t)

 void XMLWriter::tagField(const char *field, const uint16_t value, const uint8_t base)

      ^

C:\Users\beic\Documents\Arduino\libraries\XMLWriter\XMLWriter.cpp:129:6: note: void XMLWriter::tagField(const char*, uint32_t, uint8_t)

 void XMLWriter::tagField(const char *field, const uint32_t value, const uint8_t base)

      ^

C:\Users\beic\Documents\Arduino\libraries\XMLWriter\XMLWriter.cpp:138:6: note: void XMLWriter::tagField(const char*, int8_t, uint8_t)

 void XMLWriter::tagField(const char *field, const int8_t value, const uint8_t base)

      ^

In file included from C:\Users\beic\Documents\Arduino\libraries\XMLWriter\XMLWriter.cpp:21:0:

C:\Users\beic\Documents\Arduino\libraries\XMLWriter/XMLWriter.h:83:10: note: void XMLWriter::tagField(const char*, int16_t, uint8_t)

     void tagField(const char* field, const int16_t  value, const uint8_t base=DEC);

          ^

C:\Users\beic\Documents\Arduino\libraries\XMLWriter/XMLWriter.h:84:10: note: void XMLWriter::tagField(const char*, int32_t, uint8_t)

     void tagField(const char* field, const int32_t  value, const uint8_t base=DEC);

          ^

C:\Users\beic\Documents\Arduino\libraries\XMLWriter/XMLWriter.h:86:10: note: void XMLWriter::tagField(const char*, double, uint8_t)

     void tagField(const char* field, const double   value, const uint8_t decimals=2);

          ^

C:\Users\beic\Documents\Arduino\libraries\XMLWriter\XMLWriter.cpp: In member function 'void XMLWriter::tagField(const char*, int16_t, uint8_t)':

C:\Users\beic\Documents\Arduino\libraries\XMLWriter\XMLWriter.cpp:145:39: error: call of overloaded 'tagField(const char*&, long int, const uint8_t&)' is ambiguous

     tagField(field, (long) value, base);

                                       ^

C:\Users\beic\Documents\Arduino\libraries\XMLWriter\XMLWriter.cpp:145:39: note: candidates are:

C:\Users\beic\Documents\Arduino\libraries\XMLWriter\XMLWriter.cpp:119:6: note: void XMLWriter::tagField(const char*, uint8_t, uint8_t)

 void XMLWriter::tagField(const char *field, const uint8_t value, const uint8_t base)

      ^

C:\Users\beic\Documents\Arduino\libraries\XMLWriter\XMLWriter.cpp:124:6: note: void XMLWriter::tagField(const char*, uint16_t, uint8_t)

 void XMLWriter::tagField(const char *field, const uint16_t value, const uint8_t base)

      ^

C:\Users\beic\Documents\Arduino\libraries\XMLWriter\XMLWriter.cpp:129:6: note: void XMLWriter::tagField(const char*, uint32_t, uint8_t)

 void XMLWriter::tagField(const char *field, const uint32_t value, const uint8_t base)

      ^

C:\Users\beic\Documents\Arduino\libraries\XMLWriter\XMLWriter.cpp:138:6: note: void XMLWriter::tagField(const char*, int8_t, uint8_t)

 void XMLWriter::tagField(const char *field, const int8_t value, const uint8_t base)

      ^

C:\Users\beic\Documents\Arduino\libraries\XMLWriter\XMLWriter.cpp:143:6: note: void XMLWriter::tagField(const char*, int16_t, uint8_t)

 void XMLWriter::tagField(const char *field, const int16_t value, const uint8_t base)

      ^

In file included from C:\Users\beic\Documents\Arduino\libraries\XMLWriter\XMLWriter.cpp:21:0:

C:\Users\beic\Documents\Arduino\libraries\XMLWriter/XMLWriter.h:84:10: note: void XMLWriter::tagField(const char*, int32_t, uint8_t)

     void tagField(const char* field, const int32_t  value, const uint8_t base=DEC);

          ^

C:\Users\beic\Documents\Arduino\libraries\XMLWriter/XMLWriter.h:86:10: note: void XMLWriter::tagField(const char*, double, uint8_t)

     void tagField(const char* field, const double   value, const uint8_t decimals=2);

          ^

exit status 1
Error compiling for board WeMos D1 Mini Pro.
```

RobTillaart, 2017-12-08:
The problem appears in the casting, please try the following

Replace **(long)**  by **(int32_t)** in line 140 and line 145 of XMLWriter.cpp

RobTillaart, 2017-12-09:
Updated the XMLWriter lib to 1.0.8 which contains the above fix.

beicnet, 2017-12-09:
Thank you, downloading again and I will check it today! ;)

RobTillaart, 2017-12-09:
Hi Victor,

A String variation of the "capture" class

``` C++
//
//    FILE: PrintString.h
//  AUTHOR: Rob Tillaart
// VERSION: 0.1.00
// PURPOSE: Class that captures prints into a String 
//    DATE: 2017-12-07
//     URL: 
//
// Released to the public domain
//

#ifndef PrintString_h
#define PrintString_h

#include "Arduino.h"
#include "Print.h"

#define PRINTSTRING_VERSION "0.1.0"

class PrintString: public Print
{
public:
  PrintString() {};
  
  size_t write(uint8_t c)
  {
    buffer += (char)c;
    return 1;
  }
  
  void clear()
  {
    buffer = "";
  }
 
  String getString() { return buffer; }

private:
  String buffer;
};
#endif
// -- END OF FILE --
```

beicnet, 2017-12-09:
Dear Rob,

Can I get some updated example, because the above mentioned does not work.

The library is okay now.

Kind regards,
Viktor

RobTillaart, 2017-12-09:
What does not work, what is the error message?

RobTillaart, 2017-12-09:
This should work with the String variation from above (not tested as I'm quite busy on my own projects)
```C++
//
//    FILE: printString.ino
//  AUTHOR: Rob Tillaart
// VERSION: 0.0.1
// PURPOSE: demo
//
// HISTORY:

#include "PrintString.h"

#include "XMLWriter.h"

PrintString ps;
XMLWriter XML(&ps);

void setup()
{
  Serial.begin(115200);
  Serial.println(__FILE__);

  ps.clear();

  XML.header();
  XML.comment("Weather in Nebraska");
  XML.tagOpen("Data");
  XML.writeNode("Date", "20131106");
  XML.writeNode("Time", "11:42");
  XML.writeNode("Temp", "23.4");
  XML.writeNode("Humi", "50%");
  XML.writeNode("Rain", "10mm");
  XML.writeNode("Sun", "40");
  XML.tagClose();

  Serial.println(ps.getString());
}

void loop()
{
}
```

beicnet, 2017-12-09:
Dear Rob,

The errors where on these lines:

```
Serial.println(ps.getBuffer());
Serial.println(ps.free());
```

Now with your new example it's working like a charm! ;)

Thank you for your support!

Kind regards,
Viktor

p.s. and if you will have a spare time please include your last post as example to the XMLWriter library example folder.

RobTillaart, 2017-12-10:
Example posted - https://github.com/RobTillaart/Arduino/tree/master/libraries/PrintString

closed 2017-12-10 by RobTillaart