RobTillaart/Arduino #84: License

lukasz6822, 2017-12-09:
Hi,

could You add a license file. Your DHT's library is the best, but i can't use it without license.
Thank You so much!

Regards,
Lukas

RobTillaart, 2017-12-09:
It has no license, only // Released to the public domain
It is free to use but no warranties. 
What license would cover that?

lukasz6822, 2017-12-09:
[https://help.github.com/articles/licensing-a-repository/](url)

"You're under no obligation to choose a license. However, without a license, the default copyright laws apply, meaning that you retain all rights to your source code and no one may reproduce, distribute, or create derivative works from your work. If you're creating an open source project, we strongly encourage you to include an open source license. The Open Source Guide provides additional guidance on choosing the correct license for your project."

so i cant use Your library. 
e.g MIT or BSD license

RobTillaart, 2017-12-09:
I will check tomorrow if time permits as this will be needed for all my libs. 
(and yes, I know there are copies from my libs already moving around)

lukasz6822, 2017-12-09:
Ok, Thanks.

RobTillaart, 2017-12-10:
You gave me a good read this morning, how to fill up a persons mental stack. 
I added the MIT license on top level, so that closes this issue, OK?

beicnet, 2017-12-10:
@RobTillaart That cooment was really :rofl:, yes, and the MIT licence would be fine by me too.

closed 2017-12-10 by RobTillaart