RobTillaart/Arduino #85: Feature request, if I may! ;)

beicnet, 2017-12-10:
Hi @RobTillaart,

Can you please add to your great library collection these three sensors too?
1. [DHT12](http://www.robototehnika.ru/file/DHT12.pdf) (single or i2c) - datasheet attached.
2. [AM2320](https://akizukidenshi.com/download/ds/aosong/AM2320.pdf) (single or i2c)  - datasheet attached.
3. [AM2301](https://kropochev.com/downloads/humidity/AM2301.pdf) (single)  - datasheet attached.

I would really apriciate it! :wink:

Kind regards,
Viktor

RobTillaart, 2017-12-10:
These seem to be fully supported, just use DHT22 or DHT11 object in your code. 
Or is the dataformatting different? didn't read the datasheets yet

The I2C communication is interesting but a different handshake over a different bus. 

I am willing to write a dedicated library for those sensors if I can test with real sensors. 
Can you provide them?

beicnet, 2017-12-10:
Yes, the DHT12 is working with DHT11 library as "1-wire", also AM2301 is working with DHT22 library, but AM2320 has a different library.

I have them all also have a few spare too.

btw...Where are you located?

Kind regards,
Viktor

RobTillaart, 2017-12-11:
I'm located near Eindhoven in the Netherlands. I propose I write the library and you do the testing and if it works I'll publish it here. Work name will be DHT_I2C.
Can you run my I2C scanner to see if these sensors support I2C @ 400 Khz?
Thanks

Please mail me firstname.lastname@gmail.com so I can send you the lib directly.

beicnet, 2017-12-11:
AM2320 ouput from your I2C scanner:

```
TIME	DEC	HEX		50	100	200	300	400	500	600	700	800	[KHz]
----------------------------------------------------------------------------------------------------------------
15167	0	0x00		.	.	.	.	.	.	.	.	.
15202	1	0x01		.	.	.	.	.	.	.	.	.
15236	2	0x02		.	.	.	.	.	.	.	.	.
15270	3	0x03		.	.	.	.	.	.	.	.	.
15304	4	0x04		.	.	.	.	.	.	.	.	.
15339	5	0x05		.	.	.	.	.	.	.	.	.
15373	6	0x06		.	.	.	.	.	.	.	.	.
15408	7	0x07		.	.	.	.	.	.	.	.	.
15441	8	0x08		.	.	.	.	.	.	.	.	.
15476	9	0x09		.	.	.	.	.	.	.	.	.
15510	10	0x0A		.	.	.	.	.	.	.	.	.
15546	11	0x0B		.	.	.	.	.	.	.	.	.
15581	12	0x0C		.	.	.	.	.	.	.	.	.
15617	13	0x0D		.	.	.	.	.	.	.	.	.
15651	14	0x0E		.	.	.	.	.	.	.	.	.
15687	15	0x0F		.	.	.	.	.	.	.	.	.
15722	16	0x10		.	.	.	.	.	.	.	.	.
15758	17	0x11		.	.	.	.	.	.	.	.	.
15793	18	0x12		.	.	.	.	.	.	.	.	.
15828	19	0x13		.	.	.	.	.	.	.	.	.
15863	20	0x14		.	.	.	.	.	.	.	.	.
15899	21	0x15		.	.	.	.	.	.	.	.	.
15935	22	0x16		.	.	.	.	.	.	.	.	.
15970	23	0x17		.	.	.	.	.	.	.	.	.
16006	24	0x18		.	.	.	.	.	.	.	.	.
16040	25	0x19		.	.	.	.	.	.	.	.	.
16076	26	0x1A		.	.	.	.	.	.	.	.	.
16111	27	0x1B		.	.	.	.	.	.	.	.	.
16147	28	0x1C		.	.	.	.	.	.	.	.	.
16182	29	0x1D		.	.	.	.	.	.	.	.	.
16218	30	0x1E		.	.	.	.	.	.	.	.	.
16252	31	0x1F		.	.	.	.	.	.	.	.	.
16288	32	0x20		.	.	.	.	.	.	.	.	.
16323	33	0x21		.	.	.	.	.	.	.	.	.
16359	34	0x22		.	.	.	.	.	.	.	.	.
16394	35	0x23		.	.	.	.	.	.	.	.	.
16430	36	0x24		.	.	.	.	.	.	.	.	.
16464	37	0x25		.	.	.	.	.	.	.	.	.
16500	38	0x26		.	.	.	.	.	.	.	.	.
16536	39	0x27		.	.	.	.	.	.	.	.	.
16571	40	0x28		.	.	.	.	.	.	.	.	.
16607	41	0x29		.	.	.	.	.	.	.	.	.
16642	42	0x2A		.	.	.	.	.	.	.	.	.
16677	43	0x2B		.	.	.	.	.	.	.	.	.
16712	44	0x2C		.	.	.	.	.	.	.	.	.
16748	45	0x2D		.	.	.	.	.	.	.	.	.
16783	46	0x2E		.	.	.	.	.	.	.	.	.
16819	47	0x2F		.	.	.	.	.	.	.	.	.
16854	48	0x30		.	.	.	.	.	.	.	.	.
16889	49	0x31		.	.	.	.	.	.	.	.	.
16924	50	0x32		.	.	.	.	.	.	.	.	.
16960	51	0x33		.	.	.	.	.	.	.	.	.
16995	52	0x34		.	.	.	.	.	.	.	.	.
17031	53	0x35		.	.	.	.	.	.	.	.	.
17067	54	0x36		.	.	.	.	.	.	.	.	.
17101	55	0x37		.	.	.	.	.	.	.	.	.
17137	56	0x38		.	.	.	.	.	.	.	.	.
17172	57	0x39		.	.	.	.	.	.	.	.	.
17208	58	0x3A		.	.	.	.	.	.	.	.	.
17243	59	0x3B		.	.	.	.	.	.	.	.	.
17278	60	0x3C		.	.	.	.	.	.	.	.	.
17313	61	0x3D		.	.	.	.	.	.	.	.	.
17349	62	0x3E		.	.	.	.	.	.	.	.	.
17384	63	0x3F		.	.	.	.	.	.	.	.	.
17420	64	0x40		.	.	.	.	.	.	.	.	.
[... 181 lines cut ...]
11224	110	0x6E		.	.	.	.	.	.	.	.	.
11259	111	0x6F		.	.	.	.	.	.	.	.	.
11296	112	0x70		.	.	.	.	.	.	.	.	.
11332	113	0x71		.	.	.	.	.	.	.	.	.
11369	114	0x72		.	.	.	.	.	.	.	.	.
11405	115	0x73		.	.	.	.	.	.	.	.	.
11442	116	0x74		.	.	.	.	.	.	.	.	.
11478	117	0x75		.	.	.	.	.	.	.	.	.
11514	118	0x76		.	.	.	.	.	.	.	.	.
11551	119	0x77		.	.	.	.	.	.	.	.	.
11587	120	0x78		.	.	.	.	.	.	.	.	.
11624	121	0x79		.	.	.	.	.	.	.	.	.
11660	122	0x7A		.	.	.	.	.	.	.	.	.
11697	123	0x7B		.	.	.	.	.	.	.	.	.
11732	124	0x7C		.	.	.	.	.	.	.	.	.
11769	125	0x7D		.	.	.	.	.	.	.	.	.
11805	126	0x7E		.	.	.	.	.	.	.	.	.
11842	127	0x7F		.	.	.	.	.	.	.	.	.

1 devices found in 4574 milliseconds.
```
Do you need other information too?

Kind regards,
Viktor

beicnet, 2017-12-11:
btw... for the AM2320 sensor, I'm seeking for both type of library, single (1-wire) and I2C

Thank you! :wink:

RobTillaart, 2017-12-11:
So they both use the same addres 0x5C, and it seem they are able to run on 400Khz (which makes them about 2.5x faster than 100Khz in practice). But the performance is a secondary issue.

I did a quick read of the datasheets this morning and teh sensors are quite different in behavior so they cannot be in one class I think. So I'll propose to start with a class per sensor 
- DHT_I2C for DHT12 and look alikes and 
- AM2320 for itself.

If you mail me your address I can send you a draft of the DHT_I2C class I created this morning so you can test it. The DHT12 is quite like the DHT11 and DHT22 in dataformat so it might work allready.

Warning The AM2320 Class will take far more time, as its specification is far bigger and I need to see when I find time to work on it.

RobTillaart, 2017-12-11:
>>btw... for the AM2320 sensor, I'm seeking for both type of library, single (1-wire) and I2C

That means 2 libs.... 

You have tested that the AM2320 is **not** compatible with my DHT library (single mode)?

beicnet, 2017-12-11:
and I found one more sensors here, it's [AM2322](http://www.electrodragon.com/w/AM2322) (single or i2c) - manufacturer documentation.

```
TIME	DEC	HEX		50	100	200	300	400	500	600	700	800	[KHz]
----------------------------------------------------------------------------------------------------------------
3363	0	0x00		.	.	.	.	.	.	.	.	.
3396	1	0x01		.	.	.	.	.	.	.	.	.
3430	2	0x02		.	.	.	.	.	.	.	.	.
3463	3	0x03		.	.	.	.	.	.	.	.	.
3496	4	0x04		.	.	.	.	.	.	.	.	.
3529	5	0x05		.	.	.	.	.	.	.	.	.
3563	6	0x06		.	.	.	.	.	.	.	.	.
3596	7	0x07		.	.	.	.	.	.	.	.	.
3630	8	0x08		.	.	.	.	.	.	.	.	.
3662	9	0x09		.	.	.	.	.	.	.	.	.
3696	10	0x0A		.	.	.	.	.	.	.	.	.
3730	11	0x0B		.	.	.	.	.	.	.	.	.
3765	12	0x0C		.	.	.	.	.	.	.	.	.
3799	13	0x0D		.	.	.	.	.	.	.	.	.
3833	14	0x0E		.	.	.	.	.	.	.	.	.
3867	15	0x0F		.	.	.	.	.	.	.	.	.
3902	16	0x10		.	.	.	.	.	.	.	.	.
3937	17	0x11		.	.	.	.	.	.	.	.	.
3971	18	0x12		.	.	.	.	.	.	.	.	.
4005	19	0x13		.	.	.	.	.	.	.	.	.
4039	20	0x14		.	.	.	.	.	.	.	.	.
4074	21	0x15		.	.	.	.	.	.	.	.	.
4108	22	0x16		.	.	.	.	.	.	.	.	.
4143	23	0x17		.	.	.	.	.	.	.	.	.
4176	24	0x18		.	.	.	.	.	.	.	.	.
4211	25	0x19		.	.	.	.	.	.	.	.	.
4245	26	0x1A		.	.	.	.	.	.	.	.	.
4280	27	0x1B		.	.	.	.	.	.	.	.	.
4314	28	0x1C		.	.	.	.	.	.	.	.	.
4348	29	0x1D		.	.	.	.	.	.	.	.	.
4382	30	0x1E		.	.	.	.	.	.	.	.	.
4417	31	0x1F		.	.	.	.	.	.	.	.	.
4451	32	0x20		.	.	.	.	.	.	.	.	.
4486	33	0x21		.	.	.	.	.	.	.	.	.
4519	34	0x22		.	.	.	.	.	.	.	.	.
4554	35	0x23		.	.	.	.	.	.	.	.	.
4588	36	0x24		.	.	.	.	.	.	.	.	.
4623	37	0x25		.	.	.	.	.	.	.	.	.
4657	38	0x26		.	.	.	.	.	.	.	.	.
4691	39	0x27		.	.	.	.	.	.	.	.	.
4725	40	0x28		.	.	.	.	.	.	.	.	.
4760	41	0x29		.	.	.	.	.	.	.	.	.
4794	42	0x2A		.	.	.	.	.	.	.	.	.
4829	43	0x2B		.	.	.	.	.	.	.	.	.
4862	44	0x2C		.	.	.	.	.	.	.	.	.
4897	45	0x2D		.	.	.	.	.	.	.	.	.
4931	46	0x2E		.	.	.	.	.	.	.	.	.
4966	47	0x2F		.	.	.	.	.	.	.	.	.
5000	48	0x30		.	.	.	.	.	.	.	.	.
5035	49	0x31		.	.	.	.	.	.	.	.	.
5069	50	0x32		.	.	.	.	.	.	.	.	.
5103	51	0x33		.	.	.	.	.	.	.	.	.
5138	52	0x34		.	.	.	.	.	.	.	.	.
5172	53	0x35		.	.	.	.	.	.	.	.	.
5207	54	0x36		.	.	.	.	.	.	.	.	.
5240	55	0x37		.	.	.	.	.	.	.	.	.
5275	56	0x38		.	.	.	.	.	.	.	.	.
5309	57	0x39		.	.	.	.	.	.	.	.	.
5344	58	0x3A		.	.	.	.	.	.	.	.	.
5378	59	0x3B		.	.	.	.	.	.	.	.	.
5412	60	0x3C		.	.	.	.	.	.	.	.	.
5446	61	0x3D		.	.	.	.	.	.	.	.	.
5481	62	0x3E		.	.	.	.	.	.	.	.	.
5515	63	0x3F		.	.	.	.	.	.	.	.	.
5550	64	0x40		.	.	.	.	.	.	.	.	.
5583	65	0x41		.	.	.	.	.	.	.	.	.
5618	66	0x42		.	.	.	.	.	.	.	.	.
5652	67	0x43		.	.	.	.	.	.	.	.	.
5687	68	0x44		.	.	.	.	.	.	.	.	.
5721	69	0x45		.	.	.	.	.	.	.	.	.
5755	70	0x46		.	.	.	.	.	.	.	.	.
5789	71	0x47		.	.	.	.	.	.	.	.	.
5824	72	0x48		.	.	.	.	.	.	.	.	.
5858	73	0x49		.	.	.	.	.	.	.	.	.
5893	74	0x4A		.	.	.	.	.	.	.	.	.
5926	75	0x4B		.	.	.	.	.	.	.	.	.
5961	76	0x4C		.	.	.	.	.	.	.	.	.
5995	77	0x4D		.	.	.	.	.	.	.	.	.
6030	78	0x4E		.	.	.	.	.	.	.	.	.
6064	79	0x4F		.	.	.	.	.	.	.	.	.
6098	80	0x50		.	.	.	.	.	.	.	.	.
6133	81	0x51		.	.	.	.	.	.	.	.	.
6167	82	0x52		.	.	.	.	.	.	.	.	.
6202	83	0x53		.	.	.	.	.	.	.	.	.
6236	84	0x54		.	.	.	.	.	.	.	.	.
6270	85	0x55		.	.	.	.	.	.	.	.	.
6304	86	0x56		.	.	.	.	.	.	.	.	.
6339	87	0x57		.	.	.	.	.	.	.	.	.
6373	88	0x58		.	.	.	.	.	.	.	.	.
6408	89	0x59		.	.	.	.	.	.	.	.	.
6441	90	0x5A		.	.	.	.	.	.	.	.	.
6476	91	0x5B		.	.	.	.	.	.	.	.	.
6511	92	0x5C		.	V	V	V	V	V	V	.	.
6545	93	0x5D		.	.	.	.	.	.	.	.	.
6579	94	0x5E		.	.	.	.	.	.	.	.	.
6614	95	0x5F		.	.	.	.	.	.	.	.	.
6647	96	0x60		.	.	.	.	.	.	.	.	.
6682	97	0x61		.	.	.	.	.	.	.	.	.
6716	98	0x62		.	.	.	.	.	.	.	.	.
6751	99	0x63		.	.	.	.	.	.	.	.	.
6785	100	0x64		.	.	.	.	.	.	.	.	.
6820	101	0x65		.	.	.	.	.	.	.	.	.
6855	102	0x66		.	.	.	.	.	.	.	.	.
6891	103	0x67		.	.	.	.	.	.	.	.	.
6926	104	0x68		.	.	.	.	.	.	.	.	.
6962	105	0x69		.	.	.	.	.	.	.	.	.
6996	106	0x6A		.	.	.	.	.	.	.	.	.
7032	107	0x6B		.	.	.	.	.	.	.	.	.
7068	108	0x6C		.	.	.	.	.	.	.	.	.
7103	109	0x6D		.	.	.	.	.	.	.	.	.
7139	110	0x6E		.	.	.	.	.	.	.	.	.
7174	111	0x6F		.	.	.	.	.	.	.	.	.
7209	112	0x70		.	.	.	.	.	.	.	.	.
7244	113	0x71		.	.	.	.	.	.	.	.	.
7280	114	0x72		.	.	.	.	.	.	.	.	.
7315	115	0x73		.	.	.	.	.	.	.	.	.
7351	116	0x74		.	.	.	.	.	.	.	.	.
7386	117	0x75		.	.	.	.	.	.	.	.	.
7421	118	0x76		.	.	.	.	.	.	.	.	.
7456	119	0x77		.	.	.	.	.	.	.	.	.
7492	120	0x78		.	.	.	.	.	.	.	.	.
7527	121	0x79		.	.	.	.	.	.	.	.	.
7563	122	0x7A		.	.	.	.	.	.	.	.	.
7599	123	0x7B		.	.	.	.	.	.	.	.	.
7633	124	0x7C		.	.	.	.	.	.	.	.	.
7669	125	0x7D		.	.	.	.	.	.	.	.	.
7704	126	0x7E		.	.	.	.	.	.	.	.	.
7740	127	0x7F		.	.	.	.	.	.	.	.	.

1 devices found in 4521 milliseconds.
```

beicnet, 2017-12-11:
> That means 2 libs....

No, no, just separate libs for the I2C...the (single - 1-wire) need to be integrated/nested into your existing DHT library, no?!

Kind regards,
Viktor

RobTillaart, 2017-12-11:
OK, I understand now, 

so
1) the exisiting DHTlib needs to be extended to support the AM2320 
e.g. by means of a readAM2320() call similar to read22 read33 etc

2) an I2C library for AM2320 sensor

3) an I2C library for AM2322 sensor - if it can't be merged with (2)

beicnet, 2017-12-11:
Thats correct and what you just sent me:

4. an I2C library for DHT12 sensor

beicnet, 2017-12-11:
Output from testing DHT_I2C lib:

DHT12 sensor:
```
LIBRARY VERSION: 0.1.0
Type,	Status,	Humidity (%),	Temperature (C)
DHT_I2C, 	Checksum error,	0.1,	0.0
DHT_I2C, 	Checksum error,	44.2,	-78.0
DHT_I2C, 	OK,	0.0,	0.0
DHT_I2C, 	Checksum error,	0.8,	0.0
DHT_I2C, 	Checksum error,	38.3,	76.9
DHT_I2C, 	Checksum error,	0.0,	0.1
DHT_I2C, 	Checksum error,	0.0,	8.5
DHT_I2C, 	Checksum error,	26.1,	25.8
DHT_I2C, 	Checksum error,	25.6,	5.4
DHT_I2C, 	OK,	0.0,	0.0
DHT_I2C, 	Checksum error,	0.6,	-69.7
DHT_I2C, 	Checksum error,	1.6,	25.5
DHT_I2C, 	Checksum error,	102.3,	-102.3
DHT_I2C, 	Checksum error,	102.3,	-102.3
DHT_I2C, 	Checksum error,	102.3,	-102.3
DHT_I2C, 	Checksum error,	102.3,	-102.3
DHT_I2C, 	Checksum error,	102.3,	-102.3
DHT_I2C, 	Checksum error,	102.3,	-102.3
DHT_I2C, 	Checksum error,	102.3,	-102.3
DHT_I2C, 	Checksum error,	102.3,	-102.3
DHT_I2C, 	Checksum error,	102.3,	-102.3
DHT_I2C, 	Checksum error,	102.3,	-102.3
DHT_I2C, 	Checksum error,	102.3,	-102.3
DHT_I2C, 	Checksum error,	102.3,	-102.3
DHT_I2C, 	Checksum error,	102.3,	-102.3
DHT_I2C, 	Checksum error,	102.3,	-102.3
DHT_I2C, 	Checksum error,	102.3,	-102.3
DHT_I2C, 	Checksum error,	102.3,	-102.3
DHT_I2C, 	Checksum error,	102.3,	-102.3
DHT_I2C, 	Checksum error,	102.3,	-102.3
```
AM2320 and AM2322 sensor:
```
LIBRARY VERSION: 0.1.0
Type,	Status,	Humidity (%),	Temperature (C)
DHT_I2C, 	OK,	0.0,	0.0
DHT_I2C, 	OK,	0.0,	0.0
DHT_I2C, 	OK,	0.0,	0.0
DHT_I2C, 	OK,	0.0,	0.0
DHT_I2C, 	OK,	0.0,	0.0
DHT_I2C, 	OK,	0.0,	0.0
DHT_I2C, 	OK,	0.0,	0.0
DHT_I2C, 	OK,	0.0,	0.0
DHT_I2C, 	OK,	0.0,	0.0
DHT_I2C, 	OK,	0.0,	0.0
DHT_I2C, 	OK,	0.0,	0.0
DHT_I2C, 	OK,	0.0,	0.0
DHT_I2C, 	OK,	0.0,	0.0
DHT_I2C, 	OK,	0.0,	0.0
DHT_I2C, 	OK,	0.0,	0.0
DHT_I2C, 	OK,	0.0,	0.0
DHT_I2C, 	OK,	0.0,	0.0
DHT_I2C, 	OK,	0.0,	0.0
DHT_I2C, 	OK,	0.0,	0.0
DHT_I2C, 	OK,	0.0,	0.0
DHT_I2C, 	OK,	0.0,	0.0
DHT_I2C, 	OK,	0.0,	0.0
```

RobTillaart, 2017-12-11:
OK this output shows that all sensors can be communicated with and that only the data part needs to be fixed. The DHT_I2C will only work for the DHT12 as the AM232X series are more complex.

The AM232X series apparently read first 5 bytes from first 5 registers. These are all zero which causes the Checksum be zero so it is labeled as OK. 

I will check the DHT_I2C code if I can explain what is seen.

RobTillaart, 2017-12-11:
Please check this version of DHT_I2C.cpp;

regards,
Rob

beicnet, 2017-12-11:
Testing as we speak!

beicnet, 2017-12-11:
Test results for DHT12 I2C:

```
LIBRARY VERSION: 0.1.0
Type,	Status,	Humidity (%),	Temperature (C)
DHT_I2C, 	Checksum error,	28.5,	-28.5
DHT_I2C, 	Checksum error,	28.5,	-28.5
DHT_I2C, 	Checksum error,	28.5,	-28.5
DHT_I2C, 	Checksum error,	28.5,	-28.5
DHT_I2C, 	Checksum error,	28.5,	-28.5
DHT_I2C, 	Checksum error,	28.5,	-28.5
DHT_I2C, 	Checksum error,	28.5,	-28.5
DHT_I2C, 	Checksum error,	28.5,	-28.5
DHT_I2C, 	Checksum error,	28.5,	-28.5
DHT_I2C, 	Checksum error,	28.5,	-28.5
DHT_I2C, 	Checksum error,	28.5,	-28.5
DHT_I2C, 	Checksum error,	28.5,	-28.5
DHT_I2C, 	Checksum error,	28.5,	-28.5
DHT_I2C, 	Checksum error,	28.5,	-28.5
DHT_I2C, 	Checksum error,	28.5,	-28.5
DHT_I2C, 	Checksum error,	28.5,	-28.5
DHT_I2C, 	Checksum error,	28.5,	-28.5
DHT_I2C, 	Checksum error,	28.5,	-28.5
DHT_I2C, 	Checksum error,	28.5,	-28.5
DHT_I2C, 	Checksum error,	28.5,	-28.5
DHT_I2C, 	Checksum error,	28.5,	-28.5
```

RobTillaart, 2017-12-11:
Do you have a DHT22 nearby to check what the real values are?

Can you remove these 2 lines 
  bits[0] &= 0x03;
  bits[2] &= 0x83;

beicnet, 2017-12-11:
Yes, I have it, but how to test it with the i2c library?

RobTillaart, 2017-12-11:
Yes,
(I clean up some long posts sometimes to strip them to the essence)

beicnet, 2017-12-11:
Removed those two lines, and the result are:

```
LIBRARY VERSION: 0.1.0
Type,	Status,	Humidity (%),	Temperature (C)
DHT_I2C, 	Checksum error,	280.5,	-152.5
DHT_I2C, 	Checksum error,	280.5,	-152.5
DHT_I2C, 	Checksum error,	280.5,	-152.5
```

RobTillaart, 2017-12-11:
humidity of 280, that is really wrong .... Let me recheck code

RobTillaart, 2017-12-11:
Can you add this line in the DHT_I2C::read() function after the sensor is read?

  for (int i=0; i<5; i++) Serial.println(bits[i], HEX);

And test again?

beicnet, 2017-12-11:
Yes, I can!

RobTillaart, 2017-12-11:
OK, no data is read apparently, need to check datasheet (can take half an hour)

beicnet, 2017-12-11:
Not a problem, I have time!

RobTillaart, 2017-12-11:
**8.1DHT12 Sensor I2C Communication protocol**
I2C Clock bus,
**customers need to both pin a 1KΩ~10KΩ Pull-up resistor**

Do you have pull ups connected?

beicnet, 2017-12-11:
Yes, shoure, have 2x 4,7k resistors, otherwise your i2c scanner wont worked too!

RobTillaart, 2017-12-11:
true

beicnet, 2017-12-11:
btw...can your DHT library work on ESP8266 and ATtiny85?

RobTillaart, 2017-12-11:
Be sure to use the DHTstable version as the "normal" DHT has several AVR only optimizations.

Never tried myself,  if I recall correctly ATTiny should work as some people did that, ESP8266 I just don't know.

beicnet, 2017-12-11:
Ok, then I will test DHTstable on the ESP8266 device

RobTillaart, 2017-12-11:
DHT12 After power up (start **wait 2 Seconds** to cross the unstable condition of the
sensor)Tested environment temperature and humidity data, and record data, sensor automatic
hibernation.

Can you add that delay in the test sketch?

beicnet, 2017-12-11:
Yes, I can, you didn't know that all DHT sensors must have 2s delay between readings?

RobTillaart, 2017-12-11:
In practice all sensors I used could do with 1 second between readings.
This is a power up delay, so for the first read.

RobTillaart, 2017-12-11:
In the meantime I started also with the AM232X library....

beicnet, 2017-12-11:
Wow, nice! ;)

RobTillaart, 2017-12-12:
AM232X library published 0.1.1 version, 
TODO: write functions; optimization

RobTillaart, 2017-12-12:
libs are now operational on github.

closed 2017-12-12 by RobTillaart