RobTillaart/Arduino #90: Complex Library Arduino Due

alcazarjuanc, 2018-01-25:
**The Complex Library doesn't compile in Arduino Due.

I have the following errors:**

complex:16: error: 'COMPLEX_LIB_VERSION' was not declared in this scope

   Serial.println(COMPLEX_LIB_VERSION);

                  ^

complex:19: error: 'Complex' was not declared in this scope

   Complex c1(10.0, -2.0);

   ^

complex:19: error: expected ';' before 'c1'

   Complex c1(10.0, -2.0);

           ^

complex:20: error: expected ';' before 'c2'

   Complex c2(3, 0);

           ^

complex:21: error: expected ';' before 'c3'

   Complex c3(-10, 4);

           ^

complex:22: error: expected ';' before 'c4'

   Complex c4(-5,-5);

           ^

complex:23: error: expected ';' before 'c5'

   Complex c5(0, 0);

           ^

complex:25: error: 'one' was not declared in this scope

   Serial.println(one);

                  ^

complex:26: error: 'c1' was not declared in this scope

   Serial.println(c1);

                  ^

complex:27: error: 'c2' was not declared in this scope

   Serial.println(c2);

                  ^

complex:28: error: 'c3' was not declared in this scope

   Serial.println(c3);

                  ^

complex:29: error: 'c4' was not declared in this scope

   Serial.println(c4);

                  ^

complex:36: error: 'c5' was not declared in this scope

   c5 = c1;

   ^

exit status 1
'COMPLEX_LIB_VERSION' was not declared in this scope

**Does someone have any idea how to fix it?**

RobTillaart, 2018-01-25:
Not yet, I have to try reproduce it. Although I have no DUe nearby I can at least compile the example sketch. Which version of the IDE are you using? (I am on 1.8.5)

alcazarjuanc, 2018-01-25:
I am using the same version of the IDE.

I was trying to upload one of the examples of this library called "complex"

If there is anything I can do to help you with this problem, let me know it.

RobTillaart, 2018-01-25:
1.8.5 has no Due in it. 
mmmmm lets think
The fact that it does not see COMPLEX_LIB_VERSION means the compiler cannot find or read the complex.h file.. Lets try Teensy that is also ARM .... <some time passes> .....  =>same error as you have. OK now I can investigate.

alcazarjuanc, 2018-01-25:
When I installed the IDE on my computer, I also installed the package "Arduino SAM Boards (32 - bits ARM Cortex - M3) version 1.6.11" that was in the Boards Manager.

Maybe this information could help.

RobTillaart, 2018-01-25:
PLease add these three lines just after #include "complex.h" and try to compile

#ifndef Complex_h
#error include failed
#endif

If that fails the compiler was not able to find the file.

RobTillaart, 2018-01-25:
at the start of the compilation there is this line

Detecting libraries used...
...
And apparently it does not find the library. I tried another home brew lib and it could be found without problem. So why one does and one doesn't ?

TBC

alcazarjuanc, 2018-01-25:
Right now I'm not in front of my computer but I wrote this problem in an Arduino forum as well. 

Apparently, someone found the solution of this problem. Check this link:

http://forum.arduino.cc/index.php?topic=144446.new#new

alcazarjuanc, 2018-01-25:
I had a similar compilation problem with another library (MatrixMath) and the solution was very simple.

Check this link too: https://forum.arduino.cc/index.php?topic=391425.0

It could help to find the answer of this problem.

RobTillaart, 2018-01-25:
Found something similar. I copied the library to CC and replaced all Complex with CC and then it does work. So there is indeed some name conflict,  I'll have to think how to solve this as the compiler refuses to read the include complex.h file, so I cannot put a warning in that. For now I'll add a releasenote.txt file to make people aware. 
For now a local copy in the sketch folder seems the way to go.

Thank you for reporting this "feature"

closed 2018-01-25 by RobTillaart

RobTillaart, 2018-01-25:
ReleaseNote pushed

alcazarjuanc, 2018-01-25:
"Found something similar. I copied the library to CC and replaced all Complex with CC and then it does work."

I didn't get that solution. What is CC? 

Where can I find this releasenote?

Sorry, I'm new in this website and also in this kind of topics.

RobTillaart, 2018-01-25:
No need to apologize, here it is
https://github.com/RobTillaart/Arduino/blob/master/libraries/Complex/ReleaseNote.txt

alcazarjuanc, 2018-01-25:
Ok, now I get it.

Thank you for your time. Now I will be able to finish my school project.

alcazarjuanc, 2018-01-25:
"Solution:
- Make a copy of the Complex Library and name the foldername to CComplex 
- Rename Complex.h to CComplex.https
- Rename Complex.cpp to CComplec.cpp
- change one line in CComplex.cpp to include CComplex.h"

There are some little errors in the ReleaseNote or am I wrong?

You meant to say:

"Solution:
- Make a copy of the Complex Library and name the foldername to CComplex 
- Rename Complex.h to CComplex.h
- Rename Complex.cpp to CComplex.cpp
- change one line in CComplex.cpp to include CComplex.h"

RobTillaart, 2018-01-25:
Autocompletion error :(