RobTillaart/Arduino #91: Complex Library (Trigonometric Functions error)

alcazarjuanc, 2018-01-28:
[Complex Sine error.docx](https://github.com/RobTillaart/Arduino/files/1671663/Complex.Sine.error.docx)

I programmed an Extended Kalman Filter on Arduino DUE. I am using this complex library to compute the Jacobian of this algorithm. 

I noticed the sine function was doing wrong calculations. So that, I attach a file where you can see the problem and the solution that fixes the problem. 

Maybe it will be necessary to apply this fix in all library, so check it whenever you have time.

RobTillaart, 2018-01-29:
Now the code says
s = sin(re);
c = sqrt(1 - s*s);
and you propose to make
c = cos(re);

You're right, the sqrt() never returns negative values, so this optimization is not allowed.

I'll check the code today and fix it where needed.

Thanks for 'nailing' this bug

RobTillaart, 2018-01-29:
found four places where this "optimization" was used. 
Fix is under test

RobTillaart, 2018-01-29:
Fix delivered, please verify

RobTillaart, 2018-02-01:
Please reopen if problem exists

closed 2018-02-01 by RobTillaart