RobTillaart/Arduino #94: Inconsistent readings using DHT22 + UNO [DHTstable]

anphetamina, 2018-02-20:
I get wrong readings from time to time using the latest DHTstable (0.2.2) library, this is what my Serial monitor looks like:

**> 4755. OK,    Temperature: -11.10 C ~ Humidity: 3302.80 %**
> 4758. OK,    Temperature: 22.10 C ~ Humidity: 52.10 %
> 4761. OK,    Temperature: 22.10 C ~ Humidity: 52.00 %
> 4764. OK,    Temperature: 22.10 C ~ Humidity: 52.00 %
> 4767. OK,    Temperature: 22.10 C ~ Humidity: 52.00 %
> 4770. OK,    Temperature: 22.10 C ~ Humidity: 52.10 %
> 4773. OK,    Temperature: 22.10 C ~ Humidity: 52.10 %
> 4776. OK,    Temperature: 22.10 C ~ Humidity: 52.10 %
> 4779. Checksum error,    Temperature: 22.10 C ~ Humidity: 52.10 %
> 4783. OK,    Temperature: 22.10 C ~ Humidity: 52.00 %
> 4787. OK,    Temperature: 22.10 C ~ Humidity: 52.00 %
> 4790. OK,    Temperature: 22.10 C ~ Humidity: 52.00 %
> 4793. OK,    Temperature: 22.10 C ~ Humidity: 52.00 %
> 4796. OK,    Temperature: 22.10 C ~ Humidity: 52.00 %
> 4799. OK,    Temperature: 22.10 C ~ Humidity: 52.00 %
> 4802. OK,    Temperature: 22.10 C ~ Humidity: 52.10 %
**> 4805. OK,    Temperature: 11.00 C ~ Humidity: 3302.90 %**
> 4808. OK,    Temperature: 22.10 C ~ Humidity: 52.20 %
> 4811. OK,    Temperature: 22.10 C ~ Humidity: 52.20 %
> 4814. OK,    Temperature: 22.10 C ~ Humidity: 52.30 %
> 4817. OK,    Temperature: 22.10 C ~ Humidity: 52.30 %
> 4820. OK,    Temperature: 22.10 C ~ Humidity: 52.30 %
> 4823. OK,    Temperature: 22.10 C ~ Humidity: 52.30 %
> 4826. OK,    Temperature: 22.10 C ~ Humidity: 52.30 %
> 4829. Checksum error,    Temperature: 22.10 C ~ Humidity: 52.30 %
> 4833. OK,    Temperature: 22.10 C ~ Humidity: 52.30 %
> 4836. OK,    Temperature: 22.10 C ~ Humidity: 52.40 %
> 4839. OK,    Temperature: 22.10 C ~ Humidity: 52.40 %
> 4842. OK,    Temperature: 22.10 C ~ Humidity: 52.40 %
> 4845. OK,    Temperature: 22.10 C ~ Humidity: 52.40 %
> 4848. OK,    Temperature: 22.10 C ~ Humidity: 52.40 %
> 4851. OK,    Temperature: 22.10 C ~ Humidity: 52.40 %
**> 4854. OK,    Temperature: 11.00 C ~ Humidity: 3303.00 %**

This won't happen using the old DHTlib (0.1.26). I'm handling readings just like [this example](https://github.com/RobTillaart/Arduino/blob/master/libraries/DHTstable/examples/dht22_test/dht22_test.ino), my DHT22 sensor is working at 5V via the digital pin 2.

RobTillaart, 2018-02-20:
Note: The 0.1.26 is an AVR optimized version where DHTStable is a direct decendant from 0.1.13

difference is an extra  line HIGH

RobTillaart, 2018-02-20:
Can you verify if 0.1.13 is working correctly for you?

anphetamina, 2018-02-20:
Where is the 0.1.13?

RobTillaart, 2018-02-21:
If you mail me at rob.tillaart @ gmail I'll send you a zip file (thats easier than fetching it from github as I did not split up the libs.)

RobTillaart, 2018-02-21:
I have started a testrun with the DHT22_test.ino (now at 200 reads with no error)

few questions:
Which board are you using? 
Can you post the sketch you are using?
Does your sketch have a 2 second delay between readings?
Do you have a pull up resistor between DATA and 5V?

RobTillaart, 2018-02-21:
What I did notice that the time needed for a read slowly increases - never noticed before

at the start
DHT22, 	OK,	44.8,	19.8,	4916
DHT22, 	OK,	44.2,	19.8,	4912

after 2100 reads
DHT22, 	OK,	41.4,	18.8,	5112
DHT22, 	OK,	41.4,	18.8,	5120

so some 200 micros more  (4% slower)

RobTillaart, 2018-02-21:
Oops my testrun was not the right version. I will run a new test

anphetamina, 2018-02-21:
Readings are stable using 0.1.13, there is my Serial monitor:

> 0. OK,	Temperature: 23.50 C ~ Humidity: 48.00 %
> 3. OK,	Temperature: 23.40 C ~ Humidity: 48.10 %
> 6. OK,	Temperature: 23.50 C ~ Humidity: 48.10 %
> 9. OK,	Temperature: 23.50 C ~ Humidity: 48.10 %
> 12. OK,	Temperature: 23.50 C ~ Humidity: 48.10 %
> 15. OK,	Temperature: 23.50 C ~ Humidity: 48.10 %
> 18. OK,	Temperature: 23.50 C ~ Humidity: 48.20 %
> 21. OK,	Temperature: 23.50 C ~ Humidity: 48.20 %
> 24. OK,	Temperature: 23.50 C ~ Humidity: 48.20 %
> 27. OK,	Temperature: 23.50 C ~ Humidity: 48.20 %
> 30. OK,	Temperature: 23.50 C ~ Humidity: 48.20 %
> 33. OK,	Temperature: 23.50 C ~ Humidity: 48.20 %
> 36. OK,	Temperature: 23.50 C ~ Humidity: 48.10 %
> 39. OK,	Temperature: 23.50 C ~ Humidity: 48.10 %
> 42. OK,	Temperature: 23.50 C ~ Humidity: 48.10 %
> 45. OK,	Temperature: 23.50 C ~ Humidity: 48.10 %
> 48. OK,	Temperature: 23.50 C ~ Humidity: 48.10 %
> 51. OK,	Temperature: 23.50 C ~ Humidity: 48.10 %
> 54. OK,	Temperature: 23.50 C ~ Humidity: 48.10 %
> 57. OK,	Temperature: 23.50 C ~ Humidity: 48.10 %
> 60. OK,	Temperature: 23.50 C ~ Humidity: 48.10 %
> 63. OK,	Temperature: 23.50 C ~ Humidity: 48.10 %
> 66. OK,	Temperature: 23.50 C ~ Humidity: 48.10 %
> 69. OK,	Temperature: 23.50 C ~ Humidity: 48.10 %
> 72. OK,	Temperature: 23.50 C ~ Humidity: 48.10 %
> 75. OK,	Temperature: 23.50 C ~ Humidity: 48.10 %
> 78. OK,	Temperature: 23.50 C ~ Humidity: 48.20 %
> 81. OK,	Temperature: 23.50 C ~ Humidity: 48.50 %
> 84. OK,	Temperature: 23.50 C ~ Humidity: 48.80 %
> 87. OK,	Temperature: 23.50 C ~ Humidity: 49.00 %
> 90. OK,	Temperature: 23.50 C ~ Humidity: 49.00 %
> 93. OK,	Temperature: 23.50 C ~ Humidity: 48.90 %
> 96. OK,	Temperature: 23.50 C ~ Humidity: 48.80 %
> 99. OK,	Temperature: 23.50 C ~ Humidity: 48.70 %
> 102. OK,	Temperature: 23.50 C ~ Humidity: 48.70 %
> 105. OK,	Temperature: 23.50 C ~ Humidity: 48.60 %
> 108. OK,	Temperature: 23.50 C ~ Humidity: 48.50 %
> 111. OK,	Temperature: 23.50 C ~ Humidity: 48.40 %
> 114. Checksum error,	Temperature: 23.50 C ~ Humidity: 48.40 %
> 118. OK,	Temperature: 23.50 C ~ Humidity: 48.20 %
> 121. OK,	Temperature: 23.50 C ~ Humidity: 48.10 %
> 124. OK,	Temperature: 23.50 C ~ Humidity: 48.10 %
> 127. OK,	Temperature: 23.50 C ~ Humidity: 48.10 %
> 130. OK,	Temperature: 23.50 C ~ Humidity: 48.20 %
> 133. OK,	Temperature: 23.50 C ~ Humidity: 48.20 %
> 136. OK,	Temperature: 23.50 C ~ Humidity: 48.20 %
> 139. Checksum error,	Temperature: 23.50 C ~ Humidity: 48.20 %
> 143. OK,	Temperature: 23.50 C ~ Humidity: 48.20 %
> 146. OK,	Temperature: 23.50 C ~ Humidity: 48.20 %
> 149. OK,	Temperature: 23.50 C ~ Humidity: 48.10 %
> 152. OK,	Temperature: 23.50 C ~ Humidity: 48.10 %
> 155. OK,	Temperature: 23.40 C ~ Humidity: 48.20 %
> 158. OK,	Temperature: 23.50 C ~ Humidity: 48.20 %
> 161. OK,	Temperature: 23.50 C ~ Humidity: 48.20 %
> 164. Checksum error,	Temperature: 23.50 C ~ Humidity: 48.20 %
> 168. OK,	Temperature: 23.50 C ~ Humidity: 48.20 %
> 171. OK,	Temperature: 23.50 C ~ Humidity: 48.20 %
> 174. OK,	Temperature: 23.50 C ~ Humidity: 48.20 %
> 177. OK,	Temperature: 23.50 C ~ Humidity: 48.30 %
> 180. OK,	Temperature: 23.50 C ~ Humidity: 48.30 %
> 183. OK,	Temperature: 23.50 C ~ Humidity: 48.20 %
> 186. OK,	Temperature: 23.50 C ~ Humidity: 48.30 %
> 189. OK,	Temperature: 23.50 C ~ Humidity: 48.20 %
> 192. OK,	Temperature: 23.50 C ~ Humidity: 48.20 %
> 195. OK,	Temperature: 23.50 C ~ Humidity: 48.30 %
> 198. OK,	Temperature: 23.50 C ~ Humidity: 48.20 %
> 201. OK,	Temperature: 23.50 C ~ Humidity: 48.20 %
> 204. OK,	Temperature: 23.50 C ~ Humidity: 48.20 %
[... 519 lines cut ...]
> 1796. OK,	Temperature: 23.80 C ~ Humidity: 47.80 %
> 1799. Checksum error,	Temperature: 23.80 C ~ Humidity: 47.80 %
> 1803. OK,	Temperature: 23.90 C ~ Humidity: 47.80 %
> 1806. OK,	Temperature: 23.80 C ~ Humidity: 47.90 %
> 1809. OK,	Temperature: 23.80 C ~ Humidity: 47.90 %
> 1812. OK,	Temperature: 23.90 C ~ Humidity: 47.90 %
> 1815. OK,	Temperature: 23.80 C ~ Humidity: 47.90 %
> 1818. OK,	Temperature: 23.80 C ~ Humidity: 47.90 %
> 1821. OK,	Temperature: 23.90 C ~ Humidity: 47.80 %
> 1824. OK,	Temperature: 23.80 C ~ Humidity: 47.80 %
> 1827. OK,	Temperature: 23.90 C ~ Humidity: 47.80 %
> 1830. OK,	Temperature: 23.80 C ~ Humidity: 47.70 %
> 1833. OK,	Temperature: 23.80 C ~ Humidity: 47.60 %
> 1836. OK,	Temperature: 23.90 C ~ Humidity: 47.60 %
> 1839. OK,	Temperature: 23.90 C ~ Humidity: 47.50 %
> 1842. OK,	Temperature: 23.80 C ~ Humidity: 47.60 %
> 1845. OK,	Temperature: 23.90 C ~ Humidity: 47.50 %
> 1848. OK,	Temperature: 23.80 C ~ Humidity: 47.50 %
> 1851. OK,	Temperature: 23.90 C ~ Humidity: 47.50 %
> 1854. OK,	Temperature: 23.90 C ~ Humidity: 47.60 %
> 1857. OK,	Temperature: 23.80 C ~ Humidity: 47.60 %
> 1860. OK,	Temperature: 23.90 C ~ Humidity: 47.70 %
> 1863. OK,	Temperature: 23.90 C ~ Humidity: 47.70 %

Are all those checksum errors expected?

RobTillaart, 2018-02-21:
>    2. OK, Temperature: 23.40 C ~ Humidity: 48.10 %
>    12. OK, Temperature: 23.50 C ~ Humidity: 48.20 %
>    13. OK, Temperature: 23.50 C ~ Humidity: 48.10 %
>    14. OK, Temperature: 23.50 C ~ Humidity: 48.10 %
>    16. OK, Temperature: 23.50 C ~ Humidity: 48.10 %
>    17. OK, Temperature: 23.50 C ~ Humidity: 48.10 %
>    19. OK, Temperature: 23.50 C ~ Humidity: 48.10 %
>    20. OK, Temperature: 23.50 C ~ Humidity: 48.10 %
>    22. OK, Temperature: 23.50 C ~ Humidity: 48.10 %
>    23. OK, Temperature: 23.50 C ~ Humidity: 48.10 %
>    25. OK, Temperature: 23.50 C ~ Humidity: 48.10 %
>    26. OK, Temperature: 23.50 C ~ Humidity: 48.10 %
>    28. OK, Temperature: 23.50 C ~ Humidity: 48.50 %
>    29. OK, Temperature: 23.50 C ~ Humidity: 48.80 %
>    30. OK, Temperature: 23.50 C ~ Humidity: 49.00 %
>    31. OK, Temperature: 23.50 C ~ Humidity: 49.00 %
>    32. OK, Temperature: 23.50 C ~ Humidity: 48.90 %
>    33. OK, Temperature: 23.50 C ~ Humidity: 48.80 %
>    34. OK, Temperature: 23.50 C ~ Humidity: 48.70 %
>    35. OK, Temperature: 23.50 C ~ Humidity: 48.70 %
>    36. OK, Temperature: 23.50 C ~ Humidity: 48.60 %
>    37. OK, Temperature: 23.50 C ~ Humidity: 48.50 %
>    38. OK, Temperature: 23.50 C ~ Humidity: 48.40 %
>    39. Checksum error, Temperature: 23.50 C ~ Humidity: 48.40 %
>    43. OK, Temperature: 23.50 C ~ Humidity: 48.10 %
> [... 530 lines cut ...]
>    606. OK, Temperature: 23.90 C ~ Humidity: 47.50 %
>    607. OK, Temperature: 23.90 C ~ Humidity: 47.60 %
>    608. OK, Temperature: 23.80 C ~ Humidity: 47.60 %
>    609. OK, Temperature: 23.90 C ~ Humidity: 47.70 %
>    610. OK, Temperature: 23.90 C ~ Humidity: 47.70 %

Found one bug in the DHT stable version, is allready solved on github in
0.2.3

checksum errors do sometimes occur; depends on length lines, voltage used
etc. The T&H are ok

Do you have a pull up resistor between data and Vcc?
Do you have 2 seconds between reads?
What board do you use?

regards,
Rob

anphetamina, 2018-02-21:
> Do you have a pull up resistor between data and Vcc?

No

> Do you have 2 seconds between reads?

3 seconds between them

> What board do you use?
Arduino Genuino / Uno

PS: how do you hide the text? I should hide that wall of text

RobTillaart, 2018-02-21:
Text hiding is done as I replied by mail. You could edit your post and add code tags around the "wall"

RobTillaart, 2018-02-21:
Please add a pull up resistor.

anphetamina, 2018-02-21:
Which library do I need to use from now? Is DHTstable 0.2.3 working now for my case?

RobTillaart, 2018-02-21:
0.2.3 is stable here for several hundreds reads, so please give it a try.
I use an UNO too, 3 seconds is ok so we can eliminate those as cause 

so please try a pull up resistor.

anphetamina, 2018-02-21:
How about 0.1.26? I tested it all night long and it gives me consistent readings but it won't recognize any cheksum error for some reason

RobTillaart, 2018-02-21:
If it works for you, feel free to use it, the only point is the 0.1.26 is AVR specific optimized.

RobTillaart, 2018-02-21:
The timing of the DHT's is critical and I a pull up resistor gives better signals when cables are longer.  If the timing of a bit is misread due to RC effects you allready have a checksum error.

anphetamina, 2018-02-22:
To sum up:

- 0.2.3: still getting readings errors like post #1
- 0.1.26: stable readings but no Checksum errors
- 0.1.13: stable readings and Checksum errors

I've put a pull up resistor as you suggested, it will help getting less checksum errors but won't fix the inconsistent readings for the 0.2.3.

RobTillaart, 2018-02-22:
Thanks for this summary, but unfortunately I cannot reproduce. 
Can you mail me your sketch?

anphetamina, 2018-02-22:
I used the given `dht22_test.ino` in the library folder to test out. For some reason I get more checksum errors in mine sketch than the example ones. I'm simply using the DHT class to get the readings and then printing them.

anphetamina, 2018-02-22:
I will provide a sample of my sketch to let you run the test as I did asap

closed 2018-03-01 by RobTillaart