RobTillaart/Arduino #96: Protection from interrupts

CKEA123, 2018-03-24:
I suspect that interrupts triggered from another serial devices connected to the Arduino can cause the read of a DHT22 to fail with a timeout error.  

(I am using an Arduino UNO with version 0.1.14  of this library. The other serial device is an ESP-01 ESP8266 running as a TCP server. When I disabled the ESP-01 the reads by this DHT library never failed.
I see nothing in the history to suggest taking the latest version of the DHT will fix my problem.)

Other libraries (e.g. the Adafruit one) disables interrupts during the main part of the read process. I tried that library and had not problems. However, I'd prefer to stick with your smaller library.

Can you consider adding code to temporarily disable interrupts during the read?

RobTillaart, 2018-03-24:
Known issue, will have a look into it coming week, OK?

I'm thinking to give the class a bool flag to conditional disabling interrupts, how about that.

A read is approx 5 millisec so that is 80K clocktics, quite a time....

CKEA123, 2018-03-25:
That's a very prompt reply. Thank you.
Your proposal to make it an optional feature sounds very sensible.
Thank you.

RobTillaart, 2018-03-26:
Posted a new version of  https://github.com/RobTillaart/Arduino/tree/master/libraries/DHTlib that includes a setDisableIRQ(bool) and a getDisableIRQ() call . There is one demo sketch added that shows how it should work. As I have no sensor nearby I could not test it.

Please give it a try. (if it works it will be implemented in the other DHT libs too)

RobTillaart, 2018-04-03:
setDisableIRQ(bool)  added to the libraries, if you encounter problems, please open a new issue.

closed 2018-04-03 by RobTillaart