RobTillaart/Arduino #98: MAX44009 port settings suggestion

Bzzz, 2018-04-01:
Hi,

just tried your MAX44009 library with a Wemos D1 / ESP8266. My GY-49 board has I2C address 0x4A (which can be set in the sketch), but the Wemos can use various pins for data / clock lines. So for my SCL=D1 and SDA=D6 configuration, I need to change the max44009.cpp the following way:

```
Max44009::Max44009(const uint8_t address)
{
    _address = address;
    Wire.begin(D6, D1);
    // TWBR = 12; // 400KHz
    _data = 0;
    _error = 0;
}
```

Would you mind making that a configuration option in the Max44009 myLux(0x4A); init line, maybe?

Thanks! :)

RobTillaart, 2018-04-01:
Hi Bzzz

I will have a look at it this week. Assume you can test it?

Proposal to parameterize data & clock pin, like this --> backwards compatible
``` C++
#ifdef ESP8266
Max44009::Max44009(const uint8_t address, uint8_t dataPin = D6, uint8_t clockPin = D1)
#else
Max44009::Max44009(const uint8_t address)
#endif
{
    _address = address;
#ifdef ESP8266
    Wire.begin(dataPin, clockPin);
#else  // other platforms
    Wire.begin();
#endif
    // TWBR = 12; // 400KHz   --> should be Wire.setClock(400000);
    _data = 0;
    _error = 0;
}
```
and similar construct in max44009.h file

Opinion?

tip: highlight code by adding language after the backquotes e.g.  ``` C++

Bzzz, 2018-04-01:
Alright, this seems to work:

Header:
```C++
class Max44009
{
public:
    #ifdef ESP8266
    Max44009(const uint8_t address, uint8_t dataPin, uint8_t clockPin);
    #else
    Max44009(const uint8_t address);
    #endif
```
and so on,

Sources:
```C++
#ifdef ESP8266
Max44009::Max44009(const uint8_t address, uint8_t dataPin = 5, uint8_t clockPin = 4)
#else
Max44009::Max44009(const uint8_t address)
#endif
{
    _address = address;
    #ifdef ESP8266
        Wire.begin(dataPin, clockPin);
    #else  // other platforms
        Wire.begin();
    #endif
    // TWBR = 12; // 400KHz   --> should be Wire.setClock(400000);
    _data = 0;
    _error = 0;
}
```

Arduino code will then use either `Max44009 myLux(0x4A);` (for non-ESP stuff) or ` Max44009 myLux(0x4A, D6, D1);` with the ESP8266

Compiles and runs fine with said Wemos device, compiles fine with Mega2560 board, however, I do not have a test unit. Smaller Arduino boards most likely lack memory to store and run this little GFX monster of mine.

I've changed defaults to GPIO 5/4, as this seems to be the Wemos default for SCL/SDA. Also, not sure if uint8_t likes the "Dx" notation of pins. Demo code is available at https://github.com/Bzzz/epaper_max44009_esp8266_plot

Is it possible to use the ESP variant everywhere, but make the second and third variables optional? Couldn't get that to work, but looks more elegant to me. The if-tree for non-ESP devices always runs Wire.begin(), so there's no damage in having two variables with default values that are not used at all.

Also, thanks a lot for the quick reply!

RobTillaart, 2018-04-02:
Thanks for testing so far

One constructor would mean
```C++
Max44009::Max44009(const uint8_t address, const uint8_t dataPin = 5, const uint8_t clockPin = 4)
{
    _address = address;
    #ifdef ESP8266
        Wire.begin(dataPin, clockPin);
    #else  // other platforms
        Wire.begin();
    #endif
    // TWBR = 12; // 400KHz   --> should be Wire.setClock(400000);
    _data = 0;
    _error = 0;
}
```
I could give it a try (some where after Easter )

RobTillaart, 2018-04-02:
ESP version published  (ESP = Easter Surprise Package ;)

Please give it a try, It compiles for ESP8266 and UNO so you should not have much problems.

Bzzz, 2018-04-03:
Tested this today at work on a (Windows) machine that never had seen any of the involved hardware before - I had to update my demo sketch a bit, but the library is doing just fine on both ESP8266 hardware and classic Arduinos. It's not recognized as a proper library when cloning as a whole, so I just moved the MAX44009 folder into the library subdir...but otherwise - perfect, well done! :)

closed 2018-04-03 by Bzzz