RobTillaart/Arduino #100: Always DHTLIB_ERROR_TIMEOUT reading from an DHT22 with the DHTStable library

RoqueCampos, 2018-04-03:
Hi,

In the past I was reading from an DHT11 sensor using an ESP32 microcontroller, with the DHTStable library, and it was working very well. Because the values of the DHT11 are not very accurate, I "upgraded" to the DHT22 sensor (I buyed [this ](https://www.amazon.de/DHT22-AM2302-Digital-Temperatur-Feuchtesensor/dp/B01DB8JH4M?SubscriptionId=AKIAJGRXUIZQXTNADDGQ&tag=smart-thermostat-de-21&linkCode=al1&camp=2025&creative=165953&creativeASIN=B01DB8JH4M) one).

I did the same connections like with the DHT11, but I always receive an DHTLIB_ERROR_TIMEOUT when trying to read from the sensor. I don't know much about electronics, but maybe I'm using the DHT22 wrong. Do I have to use "of necessity" a resistor for the DHT22 (for the DHT11 I'm not using a resistor and it works very well).

And in the case the resistor is the problem, do you know if there is a DHT22 with incorporated resistor?

Thank you very much for your library, it works much better than the others!

Regards!

RobTillaart, 2018-04-03:
Do you have a timer / interrupt in the background?
That is seen with the ESP series more often than Arduino Uno and can disrupt the communication.

- see also this issue https://github.com/RobTillaart/Arduino/issues/96

I am working on an additional flag to disable interrupts but it is not in all versions yet.

RoqueCampos, 2018-04-03:
Hi, I am not sure if I have an interrupt in the background but I don't think so. You can see the complete sketch in [this page](https://www.smart-thermostat.eu/wireless-sensor/).

Thank you very much for your help, and please, forgive my ignorance.

RobTillaart, 2018-04-03:
Wifi server in the background, can be a problem, but should popup with DHT11 too.
I will update the DHTstable this evening to have a disable interrupts flag so you can test.

Can you run one of the test sketches of the lib for the DHT22? 
If that fails it can also be a sensor problem.

RoqueCampos, 2018-04-03:
Ok, I will run the sketchs (I should have done it before asking, apologizes).

Please give me a couple of days, and I will tell you the results of my tests.

Thank you very much for your help!

RobTillaart, 2018-04-03:
DHT libraries are updated to fix #96 , Now you can disable interrupts during acquisition.  Then timing should be more stable. If this doesn''t fix your problem let me know.

closed 2018-04-03 by RobTillaart