RobTillaart/Arduino #101: Errors Compiling in Arduino IDE 1.8.6

Salty-Doug, 2018-04-16:
Hi Rob,
I am seeing errors when I compile your dht22_test example. I just downloaded your library and am trying it out with my ESP32 Dev board. I have been working in in Arduino for only a month so I don't see anything I could check, so let me know if there is something I've overlooked. This is what I see:

Arduino: 1.8.6 Hourly Build 2018/04/03 01:12 (Mac OS X), Board: "ESP32 Dev Module, QIO, 80MHz, 4MB (32Mb), 460800, None"

In file included from /Users/SaltyDoug/Documents/Arduino/Utilities/DHT-Stable-master/Examples/dht22_test/dht22_test.ino:16:0:
/Users/SaltyDoug/Documents/Arduino/libraries/DHT-Stable-master/dht.h:30:3: error: stray '\302' in program
   <title>Arduino/dht.h at master · RobTillaart/Arduino</title>
   ^
/Users/SaltyDoug/Documents/Arduino/libraries/DHT-Stable-master/dht.h:30:3: error: stray '\267' in program
In file included from /Users/SaltyDoug/Documents/Arduino/Utilities/DHT-Stable-master/Examples/dht22_test/dht22_test.ino:16:0:
/Users/SaltyDoug/Documents/Arduino/libraries/DHT-Stable-master/dht.h:136:3: error: missing terminating ' character
   </option></form><form class="js-site-search-form" data-scoped-search-url="/RobTillaart/Arduino/search" data-unscoped-search-url="/search" action="/RobTillaart/Arduino/search" accept-charset="UTF-8" method="get"><input name="utf8" type="hidden" value="&#x2713;" />
   ^
In file included from /Users/SaltyDoug/Documents/Arduino/Utilities/DHT-Stable-master/Examples/dht22_test/dht22_test.ino:16:0:
/Users/SaltyDoug/Documents/Arduino/libraries/DHT-Stable-master/dht.h:265:9: error: missing terminating ' character
         <li></option></form><form class="logout-form" action="/logout" accept-charset="UTF-8" method="post"><input name="utf8" type="hidden" value="&#x2713;" /><input type="hidden" name="authenticity_token" value="DTna65SSmyehaec4LiDKH2E0el9Onc26XS+/gD7Yvrb559WW1DMYMUnJFkwzaska9k7Ze5mpv4SYMq129CFhuQ==" />
         ^
/Users/SaltyDoug/Documents/Arduino/libraries/DHT-Stable-master/dht.h:277:9: error: missing terminating ' character
         </option></form><form class="sr-only right-0" action="/logout" accept-charset="UTF-8" method="post"><input name="utf8" type="hidden" value="&#x2713;" /><input type="hidden" name="authenticity_token" value="soI9V5BvTMFPjFFWXhlqxaYVw2jQUrKx8tXAWbLqNUZGXDIq0M7P16csoCJDU2nAMW9gTAdmwI83yNKveBPqSQ==" />
         ^
/Users/SaltyDoug/Documents/Arduino/libraries/DHT-Stable-master/dht.h:313:9: error: missing terminating ' character
         </option></form><form data-autosubmit="true" data-remote="true" class="js-social-container" action="/notifications/subscribe" accept-charset="UTF-8" method="post"><input name="utf8" type="hidden" value="&#x2713;" /><input type="hidden" name="authenticity_token" value="oS6yy/ASg/ozxS0l+SE4KANRo6e9ConbNbQQS1JLUCruqB8TUalCQ7Y5KU2FBD7N6z3B9wq1qjkOCm1PxcFK3g==" />      <input type="hidden" name="repository_id" id="repository_id" value="2544179" class="form-control" />
         ^
In file included from /Users/SaltyDoug/Documents/Arduino/Utilities/DHT-Stable-master/Examples/dht22_test/dht22_test.ino:16:0:
/Users/SaltyDoug/Documents/Arduino/libraries/DHT-Stable-master/dht.h:348:81: error: stray '@' in program
                     <span class="description">Be notified when participating or @mentioned.</span>
                                                                                 ^
/Users/SaltyDoug/Documents/Arduino/libraries/DHT-Stable-master/dht.h:393:5: error: missing terminating ' character
     </option></form><form class="starred js-social-form" action="/RobTillaart/Arduino/unstar" accept-charset="UTF-8" method="post"><input name="utf8" type="hidden" value="&#x2713;" /><input type="hidden" name="authenticity_token" value="e++HYuT8PSdsnMa+budujpTUk221e5VFDUWkdhf7D4jREVxGvkNJwRNa5q5qW4Ml2o7VTNt1UoYt7HCTwz6Lzg==" />
     ^
/Users/SaltyDoug/Documents/Arduino/libraries/DHT-Stable-master/dht.h:408:5: error: missing terminating ' character
     </option></form><form class="unstarred js-social-form" action="/RobTillaart/Arduino/star" accept-charset="UTF-8" method="post"><input name="utf8" type="hidden" value="&#x2713;" /><input type="hidden" name="authenticity_token" value="WRDcAyZ3ff+cbrOFPg3Wmo8y/fdPv7YlepfbclXxYeW14qYKBPOLP/FN75pEz/osg1E62Y3cUiDQ/FQE1gae0g==" />
     ^
/Users/SaltyDoug/Documents/Arduino/libraries/DHT-Stable-master/dht.h:427:11: error: missing terminating ' character
           </option></form><form class="btn-with-count" action="/RobTillaart/Arduino/fork" accept-charset="UTF-8" method="post"><input name="utf8" type="hidden" value="&#x2713;" /><input type="hidden" name="authenticity_token" value="Uutr7H1ec84nyKUBR83bCKVgAZd1XI05X1J5Ic0k5R7Yu7UPjicPO6r14HQiRz8oH2AKQ4N3VzzUczgb1cjHLg==" />
           ^
In file included from /Users/SaltyDoug/Documents/Arduino/Utilities/DHT-Stable-master/Examples/dht22_test/dht22_test.ino:16:0:
/Users/SaltyDoug/Documents/Arduino/libraries/DHT-Stable-master/dht.h:629:11: error: missing terminating ' character
           </option></form><form class="inline-form js-update-url-with-hash" action="/RobTillaart/Arduino/edit/master/libraries/DHTstable/dht.h" accept-charset="UTF-8" method="post"><input name="utf8" type="hidden" value="&#x2713;" /><input type="hidden" name="authenticity_token" value="wXO1l1TE+8FIoQHzCLYJXXRiwMJjdAe37OMEdvWoG5ikaAzCta09KOi00QzZmuCiizZ1GDXm23ayxfJEr9prvg==" />
           ^
/Users/SaltyDoug/Documents/Arduino/libraries/DHT-Stable-master/dht.h:635:9: error: missing terminating ' character
         </option></form><form class="inline-form" action="/RobTillaart/Arduino/delete/master/libraries/DHTstable/dht.h" accept-charset="UTF-8" method="post"><input name="utf8" type="hidden" value="&#x2713;" /><input type="hidden" name="authenticity_token" value="ElXq8BPQttTHjPmS7lFGpw/HTj36uTpbE+dJUVG3ALkFlI3x+vT2qaybheANkWvt5Q8uctckQUlTKQHbM3SDXA==" />
         ^
/Users/SaltyDoug/Documents/Arduino/libraries/DHT-Stable-master/dht.h:700:70: error: stray '#' in program
         <td id="LC12" class="blob-code blob-code-inner js-file-line">#<span class="pl-k">ifndef</span> dht_h</td>
                                                                      ^
/Users/SaltyDoug/Documents/Arduino/libraries/DHT-Stable-master/dht.h:704:70: error: stray '#' in program
         <td id="LC13" class="blob-code blob-code-inner js-file-line">#<span class="pl-k">define</span> <span class="pl-en">dht_h</span></td>
                                                                      ^
/Users/SaltyDoug/Documents/Arduino/libraries/DHT-Stable-master/dht.h:713:70: error: stray '#' in program
         <td id="LC15" class="blob-code blob-code-inner js-file-line">#<span class="pl-k">if</span> ARDUINO &lt; 100</td>
                                                                      ^
/Users/SaltyDoug/Documents/Arduino/libraries/DHT-Stable-master/dht.h:717:70: error: stray '#' in program
         <td id="LC16" class="blob-code blob-code-inner js-file-line">#<span class="pl-k">include</span> <span class="pl-s"><span class="pl-pds">&lt;</span>WProgram.h<span class="pl-pds">&gt;</span></span></td>
                                                                      ^
/Users/SaltyDoug/Documents/Arduino/libraries/DHT-Stable-master/dht.h:721:70: error: stray '#' in program
         <td id="LC17" class="blob-code blob-code-inner js-file-line">#<span class="pl-k">else</span></td>
                                                                      ^
/Users/SaltyDoug/Documents/Arduino/libraries/DHT-Stable-master/dht.h:725:70: error: stray '#' in program
         <td id="LC18" class="blob-code blob-code-inner js-file-line">#<span class="pl-k">include</span> <span class="pl-s"><span class="pl-pds">&lt;</span>Arduino.h<span class="pl-pds">&gt;</span></span></td>
                                                                      ^
/Users/SaltyDoug/Documents/Arduino/libraries/DHT-Stable-master/dht.h:729:70: error: stray '#' in program
         <td id="LC19" class="blob-code blob-code-inner js-file-line">#<span class="pl-k">endif</span></td>
                                                                      ^
/Users/SaltyDoug/Documents/Arduino/libraries/DHT-Stable-master/dht.h:738:70: error: stray '#' in program
         <td id="LC21" class="blob-code blob-code-inner js-file-line">#<span class="pl-k">define</span> <span class="pl-en">DHT_LIB_VERSION</span> <span class="pl-s"><span class="pl-pds">&quot;</span>0.2.3 - dhtstable<span class="pl-pds">&quot;</span></span></td>
[... cut ...]
         <td id="LC36" class="blob-code blob-code-inner js-file-line"><span class="pl-k">const</span> <span class="pl-k">int</span> DHTLIB_TIMEOUT = (F_CPU/<span class="pl-c1">40000</span>);</td>
                                                                                                                                                                                              ^
exit status 1
Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

RobTillaart, 2018-04-17:
You apparently have UTF-8 or UTF16 characters in the file.  Copy the files of the library in a plain text editor  (a la notepad) and save them again, that should remove them.

Salty-Doug, 2018-04-18:
Thanks Rob.
I'll process the files again and make sure they are saved as "Plane Text".
Doug.

Salty-Doug, 2018-04-18:
Thanks Rob, that fixed it.

closed 2018-04-18 by Salty-Doug