RobTillaart/Arduino #102: DHT22 with ESP8266 seeing eventual timeout errors

tobymurray, 2018-04-29:
Pre-question: is ESP8266 supported for the DHT22 libraries? I've read through a couple issues and comments that seem like you've at least accepted fixes to make it more likely to work. If it's not supported, I welcome any insight you can provide but feel free to close.

Using an ESP8266 (specifically on the LiLon NodeMCU V3) with two DHT22s, a ~2 second sampling frequency, and the DHTStable library version 0.2.3 and the 0.2.4 on the head of `master` I'm seeing eventual (after a couple days) persistent timeout errors. Each sensor will at some point start timing out and then not recover.

Things I've tried:
- setting `DHT.setDisableIRQ(true);`, no discernible impact (although I haven't kept track of time-to-failure)
- I've separated the power supply for the DHT22s from the ESP8266, and if I power cycle the DHT22s the problem temporarily resolves itself
- I've swapped out the DHT22s for different ones from different batches, they seem to all behave similarly

Any insight into what may be causing it? Unfortunately I don't have much information to offer in terms of reproduction case, I haven't seen it start happening in less that a day which makes experiments difficult. It's something I'm in the middle of so I'd be happy to provide any additional information that would be useful.

tobymurray, 2018-04-29:
Awesome! I can answer some of your questions now, I'll have to get back to you on others.

> Q2: Do the DHT's have pull up resistors in place?

The most recent failure was with DHT22s on breakout boards that have resistors wired in, in this case 10kΩ. The sensors are roughly 2 feet away from the board (in terms of wire length). The NodeMCU also has built in pull up resistors, but as nothing is explicitly setting them they default to do nothing. I have some unattached if varying the resistance is something we want to play with.

> Q3: Do both sensors fail at the same time?

No, the most recent failures were ~12 hours apart.

> Q4: Is it after exactly the same time that the code fails?

My intuition is that it's not, but I haven't keep track so far - I can confirm this in the next few days.

> Q5: What are there other "tasks" that run on the ESP? (e.g. webserver)

Right now there's a WiFi connection that is fetching NTP packets and sending temperature and humidity out. The temperature and humidity readings are used to turn other GPIOs on and off. The code is [here](https://github.com/tobymurray/esp8266-test/blob/master/attempt2/attempt2.ino) (note that I've made no effort to make it consumable by other people as it's very much a WIP).

> Q6: Is it possible to add an extra "delay" after a failing read?

I can absolutely do that.

> Q7: Is it possible that your sketch runs out of RAM?

I'm new to C++, but I don't believe I'm intentionally allocating memory dynamically. For both NTP and MQTT messages I'm using statically allocated memory. I haven't really looked too closely at the libraries I'm using - could be something in there. I'll see if I can start logging memory usage. 

> Q8: how is the system powered?

At the time of the most recent failure, the ESP8266 was powered via my desktop computer's USB port and the DHT22s were powered with the 5v line of a separate computer PSU. 

> Q9: is there other hardware connected to the project?

Yes - 1 GPIO driving a SSR (keeping heat stable), and one GPIO driving a transistor that itself drives an electromechanical relay (keeping humidity stable).

RobTillaart, 2018-04-29:
> >Q2: Do the DHT's have pull up resistors in place?
> 
> The most recent failure was with DHT22s on breakout boards that have resistors wired in, in this case 10kΩ. The sensors are roughly 2 feet away from the board (in terms of wire length). The NodeMCU also has built in pull up resistors, but as nothing is explicitly setting them they default to do nothing. I have some unattached AM2302s if varying the resistance is something we want to play with.

for two feet 10K might be at the high end, I would go for 4K7 or 3K3 
see - http://www.gammon.com.au/forum/?id=10896&reply=5#reply5 - for the effect off pull ups
(it is not a DHT but the pictures show the effect) Adding a 10K in parallel might do the trick .
**Action: change pull up**

> > Q3: Do both sensors fail at the same time?
> No, the most recent failures were ~12 hours apart.
For me this is a strong indication that it is not a library code problem. 

> > Q4: Is it after exactly the same time that the code fails?
> My intuition is that it's not, but I haven't keep track so far - I can confirm this in the next few days.
The answer on Q3 does expect it is not a code related, so I expect it will happen not at same time, but lets await your test. 

> Right now there's a WiFi connection that is fetching NTP packets and sending temperature and humidity out. The temperature and humidity readings are used to turn other GPIOs on and off. The code is [...]

Quick glance (< 1 min) at the code ==> line 131 - char* timeString = asctime(timeinfo);
allocates memory every time the function is called. It is ever freed?? ==> might be a mem leak.
NTP *ntp is allocated only once in setup, so that looks clean. No other dynamic mem seen.
If this is a memory leak, it could add up in 2 days, so 
**Action: add code that prints free memory every loop()**  

> > Q6: Is it possible to add an extra "delay" after a failing read?
> I can absolutely do that.
Probably not needed as we have 2 possible causes that could explain the effects seen

> I'm new to C++, but I don't believe I'm intentionally allocating memory dynamically. For both NTP and MQTT messages I'm using statically allocated memory. I haven't really looked too closely at the [...]
See Q5

> > Q8: how is the system powered?
> At the time of the most recent failure, the ESP8266 was powered via my desktop computer's USB port and the DHT22s were powered with the 5v line of a separate computer PSU.
Sounds OK

> > Q9: is there other hardware connected to the project?
> Yes - 1 GPIO driving a SSR (keeping heat stable), and one GPIO driving a transistor that itself drives an electromechanical relay (keeping humidity stable).
The GPIO driving a SSR might need a transistor too - or is it low current? 
If not low current I expect it would fail much faster.

OK, after this first round I see 1 serious candidate - dynamic memory line131.
The 10K pull ups should be lower imho but given that they work properly for 2 days I do not expect them to be the root cause.  4K7 should be better.

So hunt for memory leaks, 
found this link, might be helpfull  - https://github.com/esp8266/Arduino/issues/81

RobTillaart, 2018-04-29:
Had a quick look at asctime() memory behavior
- https://stackoverflow.com/questions/11257774/ctime-return-a-string-why-we-dont-need-to-free-this-string-s-memory
seems to be memory safe. 

Still checking for mem leak is prio.

tobymurray, 2018-04-29:
> Action: add code that prints free memory every loop()

My conclusion was the same as yours regarding `asctime()` - it shouldn't be necessary (or desirable) to free the associated memory. I just added a log now to take a look (ended up on the page you linked) and after 10 minutes (~300 samples) the value of `getFreeHeap()` has not changed. Worth leaving for longer to see if there's some event that causes memory allocation, but first impression is that it's not hemorrhaging memory. 

> The GPIO driving a SSR might need a transistor too - or is it low current?
I got the SSRs specifically because I wanted to be able to drive them directly. From my tests they seem to work reliably at 3.3v with only 5.8 mA - pretty awesome!

I've noticed intermittent checksum errors as well, although they're usually one-off and quite rare (e.g. one read in a few thousand will have a checksum error. Hasn't been concerning to me as I just ignore them, but figured it was worth mentioning.

> Action: change pull up

I'm going to try and limit the changes as much as possible to try and reproduce the issue more confidently. Once I've gotten some data points around time to failure I'll start changing things and see if it has any impact.

tobymurray, 2018-04-29:
Seems like you were right off the bat, it looks like there's a memory leak in `udp.write(NTPBuffer, NTP_PACKET_SIZE);`. Every time I get the NTP packet (once every 10 minutes) it leaks 240 bytes. With ~41000 bytes free that means every ~1.2 days it would OOM. So either I'm doing something dumb or the library itself is leaking memory (or both).

tobymurray, 2018-04-29:
@ogiewon are you referring to https://github.com/esp8266/Arduino/issues/4497? That seems to be related to client connection, what I've experienced appears to be memory leaking when writing packets but not reading them within a single client connection. Thanks for mentioning it though, if what I've experienced is a bug I'll make sure to try it again with 2.4.2/2.5 and if it persists I'll try and reproduce and log a ticket over there.

tobymurray, 2018-04-30:
With the memory issue resolved (completely stable so far as I can tell), I've connected two additional DHT22s. I've used 4.7k pull up resistors on both. This means I have two with 10k and two with 4.7k pull ups. At present, I read all sequentially, wait 2 seconds, then repeat. On boot up one sensor times out consistently - every read is a time out until I power cycle it (same sensor each time). I'm seeing between 0.2% and 3% checksum errors (on both 4.7k and 10k sensors), but I'll leave it for longer to get a better picture, and to hopefully (or not) elicit the dreaded timeout errors.

tobymurray, 2018-05-01:
Sensor 1 (on a 10k pull up resistor) just failed. It was the most reliable previously but now it is timing out on 100% of reads. It failed on read #48966 ( after 1 day 3 hours 12 minutes 20 seconds) with 40352 bytes of free heap.

RobTillaart, 2018-05-02:
So it works well almost 50K times, and then the sensor is blocked. Apparently the processor inside the sensor is in a lock/unknown state as the normal wake-up signal does not work. And a power reset does.

**3.3V ?**
So I have been googling and saw on a RPI forum similar problems were mentioned, but no solutions that solved the problem. The ESP8266 is a 3.3 Volt device, the RPI is too. So it might be a power problem. Can you measure the actual voltage? 
Looked again in the datasheet of the DHT22 and it states the DHT22 needs 3.3V-6V specified so 3.3V is definitely on the edge.
Note a minimal but disruptive voltage drop might be caused by switching on some of the other devices.

Two tests come in my mind 
1) add a level converter e.g. https://www.sparkfun.com/products/12009  
2) Test the sensor with a 5V device, e.g. an UNO ?

If one of these tests go beyond lets say 75K or 100K  reads, it strongly points towards the 3.3V.

**Option?**
Power the sensor from an ESP pin (or indirectly with an transistor so you are able to reset it from program control.

tobymurray, 2018-05-02:
> Can you measure the actual voltage?

The voltage the sensor is providing when pulling high? The sensors are currently supplied from a separate power supply (at 3.3V with a common ground).

> it states the DHT22 needs 3.3V-6V specified so 3.3V is definitely on the edge

I've tried independently providing 5v to the sensors, but didn't observe a meaningful difference. I can up the voltage and let it soak and see how the stats compare.

> add a level converter e.g. https://www.sparkfun.com/products/12009

I can do that. With shipping and all that it'll be a while before I can try it out (few weeks likely).

> Test the sensor with a 5V device, e.g. an UNO ?

I have a few devices around - I think I can try on an Arduino Mega.

RobTillaart, 2018-05-02:
> > it states the DHT22 needs 3.3V-6V specified so 3.3V is definitely on the edge

> I've tried independently providing 5v to the sensors, but didn't observe a meaningful difference. I can up the voltage and let it soak and see how the stats compare.

Both the sensor itself and its data line should get 5V during the test. 

> I have a few devices around - I think I can try on an Arduino Mega.
Test with Mega would be great!

RobTillaart, 2018-05-13:
Any progress?

tobymurray, 2018-05-13:
Nope - logic converters are shipping, most pessimistic date is early July. Power cycling the sensors on timeout seems to work well enough, although more a workaround instead of a fix. I haven't gotten around to pulling out my Mega. Interestingly, while previously I was looping like: `read, read, delay(2000)` I've changed to looping continuously with `if (> 2000 millis since last read, read)` (i.e. not using `delay`) and it seems like the number of checksum errors has dropped. See similar post [here](https://github.com/esp8266/Arduino/issues/1667). I'm at ~0.1% on 100k reads. Not sure if it's related, but yet another moving piece.

Antori91, 2018-06-06:
I use DHT22 with ESP8266 for about one year and a half. I've mainly used the Adafruit library. I've moved to Rob library (DHTstable version one) a few days ago. I use two DHT22/ESP8266 sensors, one used outdoor and one used indoor. 

With the Adafruit lib, I get a lot of NAN value returned for both sensors : around 5 times per day with a request every 5 mn, so around 2% failed. For ONLY the outdoor one, I've noticed from time to time (about one time every 2/3 months) the DHT22 in a lock state. I've to power down/up the sensor. Then,  it's OK again.

With your library, Rob, I still notice NAN readings, I mean timeout or checksum errors for both the sensors (indoor and outdoor).  I've also noticed (happen only one time) a weird value for humidity about 260% with a sudden drop in temp value.

RobTillaart, 2018-06-06:
@Antori91 
some questions:
- Does my library work better or worse than Adafruits? 
  You only state the NaN qualitatively not quantitatively
- did you check voltages?
- what is the length of the wires?
- pull up resistor? value?

The humidity of 260% should also have an error (CRC) as it is unlikely that it could come through without error.

Please also check if this issue can help - https://github.com/RobTillaart/Arduino/issues/104

Antori91, 2018-06-06:
Hello Rob,

Does my library work better or worse than Adafruits?
Difficult to answer to your first question because I've used your library only for a very few days. Here is some daily stats given by a self-monitoring  feature. Before June 04,  it is figures regarding  Adafuit lib. Don't take care about June 04 and 05 because during these days, I've stopped and/or uploaded new software in the ESP8266. Regarding your library only figures of june 06 are really relevant (by the way June 06 means stats of the day before). Today stats are :
Today stats >>>>>>>>>>>>>
2018-06-06 04:37:12.731 Error - OUTDOOR_ESP8266_DHT22 has returned a wrong value (NAN)
2018-06-06 13:12:23.203 Error - INDOOR_ESP8266_DHT22 has returned a wrong value (NAN)
2018-06-06 16:22:26.975 Error - OUTDOOR_ESP8266_DHT22 has returned a wrong value (NAN)
2018-06-06 17:47:28.707 Error - INDOOR_ESP8266_DHT22 has returned a wrong value (NAN)
2018-06-06 19:27:30.727 Error - OUTDOOR_ESP8266_DHT22 has returned a wrong value (NAN)
2018-06-06 20:42:32.237 Error - INDOOR_ESP8266_DHT22 has returned a wrong value (NAN)
2018-06-06 21:57:33.767 Error - INDOOR_ESP8266_DHT22 has returned a wrong value (NAN)

If you want, I can post new figures in the next days.
Daily stats >>>>>>>>>>>>>>>
*** Thu May 24 2018 07:20:56 GMT+0200 (CEST) - Domoticz iot_ESP8266/DHT22 started ***

*** Fri May 25 2018 00:01:17 GMT+0200 (CEST) Indoor/Outdoor DHT22 NAN Reading Error= 2 / 5

*** Sat May 26 2018 00:01:46 GMT+0200 (CEST) Indoor/Outdoor DHT22 NAN Reading Error= 0 / 7

*** Sun May 27 2018 00:02:15 GMT+0200 (CEST) Indoor/Outdoor DHT22 NAN Reading Error= 0 / 5

*** Mon May 28 2018 00:02:44 GMT+0200 (CEST) Indoor/Outdoor DHT22 NAN Reading Error= 0 / 16

*** Tue May 29 2018 00:03:14 GMT+0200 (CEST) Indoor/Outdoor DHT22 NAN Reading Error= 1 / 7

*** Wed May 30 2018 00:03:43 GMT+0200 (CEST) Indoor/Outdoor DHT22 NAN Reading Error= 1 / 6

*** Thu May 31 2018 00:04:12 GMT+0200 (CEST) Indoor/Outdoor DHT22 NAN Reading Error= 0 / 3

*** Fri Jun 01 2018 00:04:41 GMT+0200 (CEST) Indoor/Outdoor DHT22 NAN Reading Error= 1 / 5

*** Sat Jun 02 2018 00:00:10 GMT+0200 (CEST) Indoor/Outdoor DHT22 NAN Reading Error= 0 / 5

*** Sun Jun 03 2018 00:00:39 GMT+0200 (CEST) Indoor/Outdoor DHT22 NAN Reading Error= 1 / 9

*** Mon Jun 04 2018 00:01:08 GMT+0200 (CEST) Indoor/Outdoor DHT22 NAN Reading Error= 16 / 41

*** Tue Jun 05 2018 00:01:37 GMT+0200 (CEST) Indoor/Outdoor DHT22 NAN Reading Error= 14 / 4

*** Wed Jun 06 2018 00:02:07 GMT+0200 (CEST) Indoor/Outdoor DHT22 NAN Reading Error= 2 / 7 

did you check voltages?
NO
what is the length of the wires?
About 10 cm
pull up resistor? value?
Yes, a 10 K one

About the humidity of 260%, it seems it happened *without* a checksum error.

RobTillaart, 2018-06-07:
What does "Error= 0 / 16" mean?

Pull up sounds OK with mentioned wire length.

A humidity of 260% can mathematically only come up if 2 bits are shifted     65 shift 2 => 260
but that would imply all bits of the message are shifted, which would normally cause a CRC error although there is a chance that the CRC would be correct. 

Voltage must be at least 3 Volt - check datasheet for exact minimum - for the DHT22 and the ESP series can operate just at the edge of working, which might cause incidental error reading. 3 Volt operating means that the rising/falling edges are more critical.

As the NaN readings happen intermittent the timing in the libraries is most probably the cause.
As these happen more often in my lib I expect Adafruits is a bit more relaxed in this. 

Does the NaN also pop up when you use the test sketches?
As it happen once per 2 hrs approx, this should not take long to test.

Antori91, 2018-06-07:
Rob,
Here is daily summary of yesterday 
*** Thu Jun 07 2018 00:02:36 GMT+0200 (CEST) Indoor/Outdoor DHT22 NAN Reading Error= 5 / 4
with 5/4 meaning 5 NAN reading errors for Indoor sensor and 4 (errors) for Outdoor one.

To run your test, I need another DHT22 I currently don't have (unless I stop my current two sensors I use for home automation). You can see the code I use here : https://github.com/Antori91/Home_Automation/blob/master/iot_Esp8266_Dht22.ino
I think for the DHT part, it is the same as the test sample code? I use your test code as a template to write my own code. But of course I add code for wifi, ota, a webserver... management

RobTillaart, 2018-06-07:
Can you change your code a tiny bit
```
    case DHTLIB_ERROR_CHECKSUM:
        strcpy( str_t, "DHT22 ERROR"); strcpy( str_h, "DHT22 ERR01"); strcpy( str_hic, "DHT22 ERR01");
        // Serial.print("Checksum error, ");
        break;
    case DHTLIB_ERROR_TIMEOUT:
        strcpy( str_t, "DHT22 ERROR"); strcpy( str_h, "DHT22 ERR02"); strcpy( str_hic, "DHT22 ERR02");
        // Serial.print("Time out error, ");
        break;
    default:
        strcpy( str_t, "DHT22 ERROR"); strcpy( str_h, "DHT22 ERR00"); strcpy( str_hic, "DHT22 ERR00");
        // Serial.print("Unknown error, ");
        break;
```
to see which error you really got.

Antori91, 2018-06-07:
Yes, I'm going to do it. But then  I have to change also the node.js server side script which polls the sensors and log the errors (if not we will not see which error I get). Once I've do it, I come back to you.

Antori91, 2018-06-07:
Done ! (not to complicated).
I already got some errors logged. All were TIMEOUT ERROR

2018-06-07 10:36:21.309 Error - INDOOR_ESP8266_DHT22 has returned a wrong value (NAN). Error code was DHT22 TIMEOUT ERROR

Antori91, 2018-06-07:
A new error logged a few minutes ago, this time CHECKSUM ERROR
2018-06-07 13:41:25.216 Error - OUTDOOR_ESP8266_DHT22 has returned a wrong value (NAN). Error code was DHT22 CHECKSUM ERROR

RobTillaart, 2018-06-07:
TIMEOUT errors are caused by not being able to clock in the bits properly.
This means that the timing and/or the electric signal (form of the pulse) do not match.

Did you measure operating voltage?

Please lower the 10K to a 4K7 resistor to get a signal with sharper edges. 
10K are normally recommended for 5V

At the moment I do not have time to build a test infra with ESP, Sorry

Antori91, 2018-06-07:
OK Rob,
I can understand you don't have time to investigate further. I've lived for more than one year with such NAN errors. I can modify my ESP8266 code to do a new reading when such issue happen. But...I got again a few minutes ago a weird error about humidity with sudden temp drop. This time I'm sure no checksum error came with this weird reading. This is more serious...Such issue never occurs before I moved to your library.

RobTillaart, 2018-06-07:
It seems like H was 260 again. if H is received bit shifted than big chance that T bits are shifted too.

What are the nummeric values of the misread T & H?
what were expected values approx.?

can you add these lines in your setup()
```
#ifdef ROBTILLAART_LIB  
DHTsetDisableIRQ(true);
#endif
```
interrupts in the background (wifi ?)  are known to corrupt the handshake timing

Antori91, 2018-06-08:
About the misread values, I'm sorry I already removed them from the database. It was about 240 for humidity and 14 for temperature. The expected values were about 76 for humidity and 24.2 for temperature.

RobTillaart, 2018-06-08:
**shift error?**
240 / 76 = ~3.15 does not sound like a bitshift error 
14 / 24.2 = ~0.57 ditto  ( = ~1.7 inverted)
a bit shift error would be near a power of 2   (+- 0.1 )
With a bit shift error you would expect both T&H to go up or both down.
So very unlikely

bit misreads (flips) go in both directions, and reads go good most of the time. If misreads all were in the same direction (LOW interpreted as HIGH or vv) timing would definitely be suspect nr one.

open causes to test
- voltage level => to be measured.

- signal form => 4K7 resistor improves this.

- interrupt => set flag as proposed.

- external => Could an external device switching on/off interfere with the readout? 
   probably too few disrupts for that.

Antori91, 2018-06-12:
In 4 consecutive days, I didn't even have a single NAN. It never happened that I remember. It seems that I have solved my problem. What did I do? I did not change anything at all to the hardware (power supply or pull-up resistance). I just tried the Adafruit library again with the code posted (version 0.2) on Github. This code is actually a new version of my code that I had never tried with this library. What may be the key to the solution in my environment, perhaps being in STA mode only at WiFi (not the case in my previous code version) (and also this legacy lib uses noInterrupts()).

RobTillaart, 2018-06-12:
Good to hear,

Do you mean a new Adafruit DHT library or a new Wifi Library?
Can you provide a link so I can study it?

Antori91, 2018-06-12:
Sorry Rob, my last comment was not so clear. To try to clarify, I don't use a new version of Adafruit library or a new version of WiFi library. By code  version 0.2 on Github, I was referring to the last version of my own code which uses the ESP8266 Wifi in station mode only (STA). By default, the ESP8266 runs WiFi in AP+STA mode. My previous code version was using WiFi in default mode (so AP+STA). My current code version (0.2) uses WiFi in STA mode only. I assume this is the reason I don't get no more NAN. My try with your own library was also with the version 0.2 of my own code.

Antori91, 2018-06-17:
It's not perfect but almost. The situation has improved significantly using the Adafruit lib. Here is the stats for the last days (almost no reading errors compared to before):

*** Sun Jun 10 2018 00:02:36 GMT+0200 (CEST) Indoor/Outdoor DHT22 NAN Reading Error= 0 / 0

*** Mon Jun 11 2018 00:03:05 GMT+0200 (CEST) Indoor/Outdoor DHT22 NAN Reading Error= 0 / 0

*** Tue Jun 12 2018 00:03:34 GMT+0200 (CEST) Indoor/Outdoor DHT22 NAN Reading Error= 0 / 0

*** Wed Jun 13 2018 00:04:04 GMT+0200 (CEST) Indoor/Outdoor DHT22 NAN Reading Error= 0 / 0

*** Thu Jun 14 2018 00:04:33 GMT+0200 (CEST) Indoor/Outdoor DHT22 NAN Reading Error= 0 / 1

*** Fri Jun 15 2018 00:00:02 GMT+0200 (CEST) Indoor/Outdoor DHT22 NAN Reading Error= 0 / 0

*** Sat Jun 16 2018 00:00:31 GMT+0200 (CEST) Indoor/Outdoor DHT22 NAN Reading Error= 1 / 2

*** Sun Jun 17 2018 00:01:00 GMT+0200 (CEST) Indoor/Outdoor DHT22 NAN Reading Error= 0 / 0

Antori91, 2018-06-17:
Hello Rob, 
The time is not the time of the errors but the time at which the daily stats of the day before are printed, in theory at midnight. 
My ESP CPU Freq is 80 MHz

kpr1, 2018-11-14:
Hello guys, maybe I meet similar problem. I didn't read whole forum because you speak a lot here :-).
People using DHT22 can meet various problems, or combinations of them. Its painful to make DHT22 working with ESP8266

**Known issues**
- pullup resistor can do issue (remove 10k from chinese, place here 2k - 3k3)
cannot fix in SW. At least better results on oscilloscope.

- powering voltage can do issue with DHT22 (3.3V is end of spec, rather use 3.6 - 5.0V by connection change). Most of ESP shields have voltage drops, so usually you have below 3.3V at DHT22.
cannot fix directly in SW, on ESP you can monitor internal voltage (maybe) and you can run ESP8266 on higher voltage, i.e. 3.6V as higher spec.

- Timming base of DHT22 can vary by few %. Maybe because not every DHT purchased is from same manufacturer.
adaptive time alg. is fine solution. Please consider that oscillator inside DHT may vary during temperature range. 

- mostly Arduinos with 16MHz works well, mostly ESP8266 (80MHz high frequencies) have problems.
I dont know if is possible, during compilation if DHT library can recognize F_CPU or which target microcontroller is used. then timming strategy will be applied.

And unhappy end, I was happy that I found this DHT stable library ... but for me still didnt work.
_"DHT22, 	Time out error,	-999.0,	-999.0,	2369"_

Antori91, 2018-11-18:
Hello Kristian,
I've definitely solved my DHT22 issues three months ago. The root cause was...my +3.3 V power supply (second item of your list) ...with voltage drops from time to time to 3.1 V. I've changed the power supply. And now it works perfectly.

teverth, 2018-12-20:
My DHT22 on an ESP32 fail whenever the temp goes below about 20 Deg C... 
I am using the current Adafruit DHT library downloaded from Adafruit. 
Wired.... I have two DHT22 on my system. The sensors stop working (read nan) when the temp goes below 22.20 Deg C reported by the software.
Could it be a software encoding issue that at the bit pattern below 22.20 things go wrong? It is not only the temp reading that then turns to nan but also the humidity reading. So probably a sensor internal issue?

tobymurray, 2018-12-28:
Little update on this (been a while): I hooked up the ESP 32 (specifically ESP-WROOM-32) to DHT22s with this library. I have 2 running on 3.3v and 2 running on 5v via a logic level converter (one of the earlier suggestions). With absolutely nothing else running but a loop running `DHT.read22` on each pin I've had > 10 000 successful reads (no time outs, no anything).

This is a totally different board, and lots of things have changed in the last 8 months, so I'll try and narrow it down, but so far as I can tell the ESP-32 is rock solid with version 0.2.4 of DHTStable. I still have some ESP8266s lying around, I'll swap one in with exactly the same setup and see if the issues show up again.

closed 2019-09-03 by RobTillaart