RobTillaart/Arduino #103: esp8266 Nodemcu DHTlib error

Dancopy, 2018-04-30:
Greetings Ben-Noah Engelhaupt!
I have an Arduino code "SmartMirror" made available on the web by Ben-Noah Engelhaupt, using the "DHT Library;" I am loading the code in esp8266 -Nodemcu and am getting the following error:

WARNING: The DHTlib library claims to run on architecture (s) (atmelavr) and may be incompatible with its current board, which runs on architecture (s) (esp8266).
C: \ Users \ Daniel \ Documents \ Arduino \ libraries \ DHTlib \ dht.h: 19: 0,

from C: \ Users \ Daniel \ Documents \ Arduino \ libraries \ DHTlib \ dht.cpp: 48:

C: \ Users \ Daniel \ Documents \ Arduino \ libraries \ DHTlib \ dht.cpp: In member function 'int8_t dht :: _ readSensor (uint8_t, uint8_t, uint8_t)':

C: \ Users \ Daniel \ AppData \ Local \ Arduino15 \ packages \ esp8266 \ hardware \ esp8266 \ 2.4.1 \ cores \ esp8266 / Arduino.h: 231: 63: error: can not convert 'volatile uint32_t * {aka volatile unsigned int *} 'to' volatile uint8_t * {aka volatile unsigned char *} 'in initialization

 #define portInputRegister (port) ((volatile uint32_t *) & GPI)

                                                               ^

C: \ Users \ Daniel \ Documents \ Arduino \ libraries \ DHTlib \ dht.cpp: 155: 29: note: in expansion of macro 'portInputRegister'

     volatile uint8_t * PIR = portInputRegister (port);

                             ^

exit status 1
Compiling error for NodeMCU 1.0 (ESP-12E Module)

Thank you,

Daniel Fernandes

tobymurray, 2018-04-30:
The warning is just a warning, but the error looks legitimate:
```
can not convert 'volatile uint32_t * {aka volatile unsigned int *} 'to' volatile uint8_t * {aka volatile unsigned char *} 'in initialization
```
One of your values is a `volatile uint32_t` and one is a `volatile uint8_t`. You'll have to change the type of one of them or convert between the two types. It's pointing right to the place in your code where the mismatch occurs.

RobTillaart, 2018-04-30:
The DHTLIb contains AVR optimized register calls for the Atmel-328 (UNO) so it won't work on non-AVR.

Please give the dhtnew library a try as that one should be portable for other platforms.

This dhtnew is my primary DHT library for the future. It adds:
- an offset to temp and hum to have a 1st order calibration.
- autodetection of type dht22/dht11 to be easier to use.
- a flag to disable interrupts to prevent communication errors

Dancopy, 2018-04-30:
Thank you; How should I do in the code? Help me please!

Thanks ++

Dancopy, 2018-04-30:
Hi Rob! Please clarify to me what I should do in the code using DHTLib or dhtnew ;

thank you

Dancopy, 2018-05-03:
here: https://github.com/bensoftde/smartmirror

RobTillaart, 2018-05-03:
4 lines need changes, this is the new content to use the dhtnew library

line15 
#include "dhtnew.h"

line 32-33
#define DHT11_PIN 3
DHTNEW DHT(DHT11_PIN);

line316
  if (DHT.read() == DHTLIB_OK) {

Give it a try

Dancopy, 2018-05-03:
Thanks Rob; I'll test, then let you know.

RobTillaart, 2018-05-13:
Any progress?

Dancopy, 2018-05-14:
Hi Rob! I'm sorry I did not let you know but it worked perfectly with your 'dhtnew' library;
Thank you very much,
Daniel

closed 2018-05-14 by RobTillaart