RobTillaart/Arduino #105: getLux() values not accurate

morxGrillmeister, 2018-05-09:
Hi,
The values I get from `float Max44009::getLux(void)` are not at a resolution of approx. 0.045 lux but approx. 0.76.
Tests showed that when I increase brightness slowly from complete darkness, `getLux()` jumps from 0.00 to 0.76.
Expected was a jump from 0.00 to 0.045.

I'm using Visual Code 2017 and a NodeMCU board with ESP8266 and your (much appreciated !)
library Max44009_0.1.7_1349.

Analysis found that always both bytes in uint16_t data were identical, e.g. 0101 or 4D4D.

I could fix this by changing the 1st line of code in getLux():
        `uint16_t data = read(MAX44009_LUX_READING, 2);`
to this:
`uint8_t mhi = read(MAX44009_LUX_READING, 1);`
`uint8_t mlo = read(MAX44009_LUX_READING + 1, 1);`
`uint16_t data = (mhi << 8) + mlo;`

I guess the real problem is in 
       ` uint16_t Max44009::read(uint8_t reg, uint8_t bytes)`
which seems not to read both registers 3 and 4 of the sensor.

RobTillaart, 2018-05-09:
Thanks for reporting, I will investigate the read() when time permits. Can take a few weeks, sorry

morxGrillmeister, 2018-05-09:
No problem, it works for me with the getLux() patch. 

Viele Grüße,
Markus Blöchlinger

RobTillaart, 2018-05-10:
Markus, as I have no Max44009 around, can you check if the library 0.1.7 works with an UNO?
(assumiption you have one). If that is the case the bug is in the I2C/Wire library.

morxGrillmeister, 2018-05-13:
Because the Max chip operates at 3.3V levels, I didn't test with an UNO. But I found a Arduino Pro Mini 328 - 3.3V/8Mhz and tested with this. It produces the same errors as the ESP8266 code.
In the attached zip file there is my test program and modified library code in order run the tests.
Please read mbl-readme.txt for details and some sample test results:
[MAX44009-Library.zip](https://github.com/RobTillaart/Arduino/files/1998434/MAX44009-Library.zip)

morxGrillmeister, 2018-05-14:
Hi Rob
tested 0.1.8 and it works very well! Thanks !

closed 2018-05-15 by RobTillaart