RobTillaart/Arduino #111: DHT12: Temperature conversion.

Ouss4, 2018-09-01:
Hi,

As per the DHT12 datasheet (http://www.robototehnika.ru/file/DHT12.pdf) the temperature sign is stored in the lower byte of the data.

So unlike the other DHTxx sensors the line: `if (bits[2] & 0x80)` should be `if (bits[3] & 0x80)` in both https://github.com/RobTillaart/Arduino/blob/master/libraries/DHT12/DHT12.cpp#L46 and 
https://github.com/RobTillaart/Arduino/blob/master/libraries/DHTlib/dht.cpp#L89

and `temperature = (bits[2] & 0x7F) + bits[3] * 0.1; ` to `temperature = bits[2] + (bits[3] & 0x7F) * 0.1;` in https://github.com/RobTillaart/Arduino/blob/master/libraries/DHTlib/dht.cpp#L88 and https://github.com/RobTillaart/Arduino/blob/master/libraries/DHT12/DHT12.cpp#L45

RobTillaart, 2018-09-01:
I will have a look at it coming days.

RobTillaart, 2018-09-02:
Fixed in both classes, please verify

closed 2018-09-02 by RobTillaart

Ouss4, 2018-09-02:
Looks good.

RobTillaart, 2018-09-03:
Thanks  for noticing!