RobTillaart/Arduino #113: _WriteBlock issue

wothke, 2018-09-10:
Hi, 

When using the lib on a 160MHz ESP8266 at 400kHz I2C I sporadically got errors when repeatedly updating/writing the same 2-byte data.

Adding the below change solved the problem for me..

int I2C_eeprom::_WriteBlock(const uint16_t memoryAddress, const uint8_t* buffer, const uint8_t length)
{
    waitEEReady();

    this->_beginTransmission(memoryAddress);
    WIRE_WRITE(buffer, length);
    int rv = Wire.endTransmission();
  
    // XXX wait for "write" to properly complete (bug in original implementation -
    // without this, repeated "write" fails on 400kHz/ESP8266)
    for (uint8_t i=0; i<50; i++) {
        delay(1);		// implicit yield should also make ESP8266 happy
        this->_beginTransmission(0);
        rv = Wire.endTransmission();
        if (rv == 0) break;
    }

    _lastWrite = micros();
    return rv;
}

RobTillaart, 2018-09-10:
Thanks for your issue.  I'll look into it asap. 

A quick look made me write the variation below. It skips the delay if it succeeds the first time. Furthermore countdown and compare to zero is faster (on most processors). Can you verify?

```for (uint8_t i = 50; i > 0; i--) 
{
    this->_beginTransmission(0);
    rv = Wire.endTransmission();
    if (rv == 0) break;
    delay(1);		// implicit yield should also make ESP8266 happy
}
```

RobTillaart, 2018-09-10:
2nd thought
Could the delay(1) be replaced by delayMicroseconds(1000) and tune it a bit more?
a millisecond is quite a long period :)

wothke, 2018-09-10:
The delay(1) could be replaced by delayMicroseconds(xxx) + yield() and the loop changed accordingly. From what I've read delayMicroseconds() does not include yield() on the ESP8266 - whereas delay() does. 

I only found sketchy information about how long an ESP8266 can go without yield(), before the WiFi stack blows up/it reboots (information ranging from "some millis" to "few hundred millis"). I just did not want to take any risks of the ESP8266 rebooting due to some silly "not enough yield" scenario (I've had enough of that shit already..) and since I don't care too much about updating "large" EEPROM blocks in my application I just kept the fix simple..

Based on reliable information (if anybody finds it) regarding how long a yield() call actually takes when there is nothing to do, and on how long the program code can safely run on an ESP8266 without using yield, the loop could certainly be optimized.

PS: I just tried your above suggestion and the fix seems to work just as well with the delay() at the end of the loop. (I just did one quick test so some more testing may be needed..)

RobTillaart, 2018-09-10:
do not want to introduce "exploding stacks" so delay(1) will it be

RobTillaart, 2019-09-03:
Took a year to get back to this , sorry,
The advantage is a fresh look :)

The implementation of waitEEReady(); does not use yield() directly, or indirectly through delay().
So the waitEEready() in _WriteBlock() et al can block for as much as 5 milliseconds. On a 160MHz processor this is ~800.000 clocks which is a lot. 

As a robust solution I propose to add yield() explicitly to the waitEEReady() to make the waiting for the EEProm to be ready works for the ESPxxxx processors. Will update this asap.

RobTillaart, 2019-09-03:
Please reopen issue if issue still exists

closed 2019-09-03 by RobTillaart