RobTillaart/Arduino #117: MAX44009 wrong pins for ESP8266

Bolukan, 2018-12-07:
Using the library with an ESP8266 I got error 2 and looked for the cause:
I read issue #98, but for the ESP8266 the typical pins are SCL as 5 (D1) and SDA as 4 (D2), see documentation for [D1 mini](https://github.com/esp8266/Arduino/blob/master/variants/d1_mini/pins_arduino.h) or [NodeMCU](https://github.com/esp8266/Arduino/blob/master/variants/nodemcu/pins_arduino.h). However, these pin numbers were switched in the code:
https://github.com/RobTillaart/Arduino/blob/a9eb3aecf3ef6802f2840b0ebd84d109f921597b/libraries/Max44009/Max44009.h#L45

The ESP8266 doesn't need the pins to be defined, just wire it the right way. 

So it should not be or only be implemented as an override for people who want to use their custom pins.

Do you appreciate a pull request? (my preference would be to remove the custom pins and as alternative to apply it on request only.

RobTillaart, 2018-12-07:
A pull request would be appreciated. Thanks 

(I have too little time to dive into the ESP and align all classes to it)

Bolukan, 2018-12-07:
See https://github.com/RobTillaart/Arduino/pull/118
Tested succesfully using default and custom pins.

Default behaviour:
`Max44009 myLux(0x4A); // Datapin D2 (4) and Clock D1 (5)`

The "wrong" switched pin configuration can be used with
`Max44009 myLux(0x4A, 5, 4);`

RobTillaart, 2018-12-08:
Merged and fixed version nr.
Thanks!

closed 2018-12-08 by RobTillaart