RobTillaart/Arduino #120: Negative Temperatures do not work

baderas, 2019-01-19:
I can't give any technical details, but with your libarary temperatures below 0°C do not work. They report as e.g. -3277.70 not -1.0. 
Changed back to adafruits library -> everything works fine with my DHT22.
Therefore I think you library does not convert the negative temperatures correctly.

yesyesuk, 2019-01-19:
Are you sure you are using the latest version? This is something that has been fixed years ago.

baderas, 2019-01-19:
It happened on DHTNEW with Commit 0c4ee966da8e0f5281148e65432ca275eae96448

baderas, 2019-01-19:
You fixed it a few month ago on DHTlib
Which one of your DHT libraries should I use on ESP8266 and DHT22?

RobTillaart, 2019-01-20:
DHTlib should work, do not forget to use a 3.3 volts to 5.0 volts convertor as the DHT sensor functions more reliable at 5.0 volts. 3.3 volts is at the edge of its working range.

I indeed need to add this same fix to DHTNEW. Will look at it a.s.a.p.

RobTillaart, 2019-01-20:
The problem is that the negative bit is not masked, causing an offset of (128*256)/10 = 3276
This matches your observation.
Fix is underway.

RobTillaart, 2019-01-20:
Fix delivered, if the problem still occurs please reopen,
Thanks for posting the issue!

closed 2019-01-20 by RobTillaart