RobTillaart/Arduino #125: max31855 status codes fix

tablatronix, 2019-07-23:
had to change to
`#define STATUS_SHORT_TO_GND     0x03`

```
tc status: GND Short
1111111111111010001100100000011
```

RobTillaart, 2019-07-30:
Sorry for the late response,

Max31855.h (version 0.1.9) does not define status short as 0x03 but as 0x02 in line 24

#define STATUS_SHORT_TO_GND     0x02

What version are you using?
What platform are you using?

Can you post a minimalistic sketch in which the error occurs?

Thinking it over, the user of the library should use masking to check detailed status
from the return value of read();

The library does not provide functions for that. Maybe it should something like 

bool shortToGND()
{
  return (_status & STATUS_SHORT_TO_GND) != 0;;
}
bool shortToVCC()
{
  return (_status & STATUS_SHORT_TO_VCC) != 0;;
}

And some examples to show how to use those.

tablatronix, 2019-07-30:
Yeah it was wrong, I had to change it to `0x03`

RobTillaart, 2019-07-31:
OK I assume it is solved for now but I will add the functions to the library when time permits.

closed 2019-07-31 by RobTillaart

tablatronix, 2019-07-31:
Do you want me to pr it?

RobTillaart, 2019-07-31:
I updated the lib with 4 status functions and how to use them.
Please note that the STATUS values are bit patterns in which 1 bit indicates an error (or not) as described in the datasheet of the MAX38155

The lib has now 4 functions added to see if there is an error, and to test them individually
- bool statusError() 
- bool shortToGND()  
- bool shortToVCC()  
- bool openCircuit()

There is a new example added that shows how it can be used.

Note the 0x03 error you got is in fact a combination of shortToGND and openCircuit, apparently these can occur simultaneously.

tablatronix, 2019-08-01:
yeah I was not thinking and testing only without a tc, doh bit flags, my bad.