RobTillaart/Arduino #128: error: no matching function for call to 'min(uint16_t&, int)'

aeakin1, 2019-09-02:
There is a continuing bug in the ESP8266 Arduino Core libraries that causes this error for all uses of mim() and max() when loading common libraries like <ESP8266WiFi.h>

See "closed" issue:
https://github.com/esp8266/Arduino/issues/263

My compile error:

<redacted>\.platformio\lib\I2C_EEPROM_ID1346\I2C_eeprom.cpp: In member function 'uint16_t I2C_eeprom::readBlock(uint16_t, uint8_t*, uint16_t)':
<redacted>\.platformio\lib\I2C_EEPROM_ID1346\I2C_eeprom.cpp:124:49: error: no matching function for call to 'min(uint16_t&, int)'

         uint8_t cnt = min(len, I2C_TWIBUFFERSIZE);

Change calls on following line from min/max to _min/_max to avoid issue

	Line 124:         uint8_t cnt = min(len, I2C_TWIBUFFERSIZE);
	Line 192:         uint8_t cnt = min(len, bytesUntilPageBoundary);
	Line 193:         cnt = min(cnt, I2C_TWIBUFFERSIZE);

Change to 

	Line 124:         uint8_t cnt = _min(len, I2C_TWIBUFFERSIZE);
	Line 192:         uint8_t cnt = _min(len, bytesUntilPageBoundary);
	Line 193:         cnt = _min(cnt, I2C_TWIBUFFERSIZE);

RobTillaart, 2019-09-03:
Thanks for pointing out this bug
To keep the lib working for AVR could you please change line 124 

```
#if defined(ESP8266) || defined(ESP32) //  || defined(...)
        uint8_t cnt = _min(len, I2C_TWIBUFFERSIZE);
#else
        uint8_t cnt = min(len, I2C_TWIBUFFERSIZE);
#endif
```

and line 192 / 193

```
#if defined(ESP8266) || defined(ESP32) //  || defined(...)
        uint8_t cnt = _min(len, bytesUntilPageBoundary);
        cnt = _min(cnt, I2C_TWIBUFFERSIZE);
#else
        uint8_t cnt = min(len, bytesUntilPageBoundary);
        cnt = min(cnt, I2C_TWIBUFFERSIZE);
#endif
```
and give it a try? ( I have no ESP nearby to verify)

RobTillaart, 2019-09-03:
Should work properly now, I tested compilation for ESP32 and for Atmel 328

closed 2019-09-03 by RobTillaart

Liferme, 2021-06-08:
Thanks a lot, it works!!!