RobTillaart/Arduino #131: Use of MCP4725 lib for ESP8266

TomsCircuits, 2019-10-13:
In [MCP4725.cpp](https://github.com/RobTillaart/Arduino/blob/cd673f01c908965358f0f3b6f051a500ac704bbb/libraries/MCP4725/MCP4725.cpp#L43) the TWBR register is accessed. This only works for Atmel controllers. Using the library for e.g. ESP8266 fails, because the register doesn't exist.

RobTillaart, 2019-10-14:
Please verify 0.1.9 now works for ESP?

TomsCircuits, 2019-10-20:
Yes, that works! Thank you very much.

I have another - minor - remark that probably doesn't warrant a new issue.

When I tested the updated library, I based my test code on your examples. In both, there is a line saying

https://github.com/RobTillaart/Arduino/blob/cd673f01c908965358f0f3b6f051a500ac704bbb/libraries/MCP4725/examples/mcp4725_test/mcp4725_test.ino#L15

and

https://github.com/RobTillaart/Arduino/blob/cd673f01c908965358f0f3b6f051a500ac704bbb/libraries/MCP4725/examples/smooth2Value/smooth2Value.ino#L15

I found that my MCP4725 can only be addressed with 0x60. Both 0x62 and 0x63 resulted in a NACK on the I2C bus. I tested this with a logic analyser.

The data sheet says that the device is addressed via '1100' + A2,A1,A0, where A2, A1 is factory programmed. Also: "The A2 and A1 are programmed to ‘00’ (default), if not requested by customer"

So I think that the default addresses should be 0x60 or 0x61, not 0x62 or 0x63.

Thank you for your hard work and providing these libraries. I hope I can give a little back with my feedback.

TomsCircuits, 2019-10-20:
...and that was rubbish, of course. The LSB of the device adress is of course bit1 and not bit0, so the addresses should be 0x60 or 0x62.

RobTillaart, 2019-10-21:
You have a point, address can be different than in the demo sketches, I must make a not of it.

You can use an I2C scanner like - https://github.com/RobTillaart/Arduino/tree/master/sketches/MultiSpeedI2CScanner - to determine the I2C address used. The ones I have use address 0x62 or 0x63 so I based my demos on it but others are possible. 

Thanks,

RobTillaart, 2019-10-21:
From datasheet
```
ordernr              Bitpattern    Hex addresses
MCP4725A0T-E/CH:     0110 000U     0x60 - 0x61
MCP4725A1T-E/CH:     0110 001U     0x62 - 0x63
MCP4725A2T-E/CH:     0110 010U     0x64 - 0x65
MCP4725A3T-E/CH:     0110 011U     0x66 - 0x67

```