RobTillaart/Arduino #132: MAX44009 Lux Incorrect

louislewis2, 2019-12-19:
Hi, saw that your library for the MAX44009 sensor, is giving the same issue, that has been reported on other libraries, where the max lux value that is received is 2937.6

Here is a sample of an issue, where this was reported and fixed, perhaps it can save some time.

https://github.com/dantudose/MAX44009/issues/1

RobTillaart, 2019-12-19:
Thanks, I will look into it asap

RobTillaart, 2019-12-19:
Note: The value 0x40 is not literally in the datasheet, it is bit 6 MANUAL of the control register (0x02).

1. As far as I understand the issue, the reading of the second data byte (register 0x04) gives ' trouble'. 

2. Also found by reading the datasheet that my code does not handle overflow [see page 9]:
_Exponent (E[3:0]): Exponent bits of the lux reading (0000 to 1110). Note: A reading of 1111 represents an overrange condition._
This is especially important in manual mode. 

3. I have currently no MAX44009 nearby so I cannot verify the fix (to be here soon). 
@louislewis2 Can you?

louislewis2, 2019-12-19:
@RobTillaart I am happy you found a solution on your side, that is fantastic news.
I have quite a few units, so will gladly verify a fix.

RobTillaart, 2019-12-19:
@louislewis2 Thanks,
- can you first verify that the current version does not work correctly?
  That it "stops/ overflows" at ~2937.60 LUX

louislewis2, 2019-12-19:
I am certain the current version does not work as expected and does not
return values over 2937.6.

RobTillaart, 2019-12-19:
Strange as the manual mode is not set so the chip should automatically adjust the TIM and CDR .
From datasheet page 8, just above table 5:

_In automatic mode (MANUAL = 0), reading the contents of TIM[2:0] and CDR bits reflects the automatically generated values from an internal timing register and are read-only. In manual mode (MANUAL = 1), the contents of TIM[2:0] and CDR bits can be modified by the users through the I2C bus._

mmm, I do not set any configuration explicitly in the constructor. I assume the chip has its default start-up value of automatic mode. Need to make that explicit.

RobTillaart, 2020-01-15:
Test sensors have finally arrived. Now I need some time ....

closed 2020-01-27 by RobTillaart

RobTillaart, 2020-01-30:
Verified today - UNO + lib version 0.3.3 works far above 2967 lux (seen 55k lux in test)