RobTillaart/Arduino #134: Multiple DAC855x's

jwktje, 2020-01-13:
Would it be possible to use multiple DAC's with this library? 
For example, when I want 5 channels using a DAC8554 and a DAC8551.
How would I implement that in my sketch? Or is this impossible?

RobTillaart, 2020-01-13:
Not tried it but should at least be possible this way.
- use the software SPI constructor for both 8554 and 8551
- use the same data and clock pin for both (you might even use the HW SPI for these
- use different slave select pin for 8554 and 8551

code would have something like 
```
// SW SPI pins 4,5,6
DAC8554 DAC1(4, 5, 6);
DAC8551 DAC2(4, 5, 7);
```

This way the two DACs will only use 4 pins. This will work as they will not be accessed simultaneously.

As said, not tried so let me know if it works, ideally with a minimal code example  that could be added to te examples. Such a demonstrator could easily be created on merging the sw_spi examples of the two libraries.

Give it a try

jwktje, 2020-01-13:
That approach sounds like it should work. 
I will for sure attempt this somewhere in the coming month or two. 
(Sadly I'm a bit busy at the moment) 

I will update with a PR once I've created a minimal example and I'll update this issue then. Thanks for clarifying!

RobTillaart, 2020-01-13:
Success! 
(I'm to busy too to give it a try :)

jwktje, 2020-01-13:
Komt goed. Dankjewel Rob. Ik ga m’n best doen. Moet vast lukken 👌🏼

RobTillaart, 2020-01-15:
I will close the issue for now, if needed you can reopen it,
groeten,

closed 2020-01-15 by RobTillaart