RobTillaart/Arduino #135: Libraries not getting picked up by Arduino IDE

divsys1, 2020-01-13:
Hi, I noticed your (excellent) libraries are not getting picked up by the Arduino IDE auto-scan library manager (I'm using ver 1.18.10 of the IDE).

In particular, I was hoping to use the MAX44009 library, but a much more rudimentary version by " dantudose" is the only one that shows under the IDE Library Manager.
I certainly can do a manual install (and have), but it would be nice if the Library Manager could keep things up to date.

Just a guess on my part (I'm no expert in this area) but it looks like your base directory structures are missing the "src" folder to contain the relevant .h, .cpp, etc. files. The only other thing I noticed (for the MAX44009 library) is an empty "paragraph" section in the library.properties file.

Once again your library work has been a great help, it would be nice if the rest of the Arduino community could use it a little more readily (especially the MAX44009 stuff).

Thanks

RobTillaart, 2020-01-15:
Did a short investigation why they are not seen by the lib manager.
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

The problem is that I have all my libraries in one big archive, so it does not follow the rules to be taken into account.  see under " Is my Git repo OK?"
Most important is to check if the name already is in use, so I might need to rename all the libs to e.g. 
_rob_max44009 like the _adafruit_libraries.

When I have spare time I could recreate my 60+ libs into separate repo's, but that would automatically give trouble for 3K++ followers of the current repo. Food for thought.

SeGerber, 2020-03-20:
Is there a work around or a way to manually add one of your library into Arduino IDE with this issue of one big git folder? Neither to zip a library (e.g. FastMap) or searching for I can include it into Arduino. Thanks a lot it would help greatly to use your fantastic libraries.

RobTillaart, 2020-03-20:
It is on my todo list to redo the libs to be able to be pickef up by Arduino infra. Some preparations have started.
If there is some particular lib or issue let me know. I could start preparing with that lib.

Konzen8, 2020-03-20:
Thanks for quick response, the library FastMap would be fantastic, as it
greatly improves sensor processing speed.

RobTillaart, 2020-03-21:
In case you want it asap.

Download the whole package as zip file.

Copy the libs / folders you need from the zip file to your work/library folder where all 3rd party folders are located.

RobTillaart, 2020-03-21:
@Konzen8 
Please check out the FastMap repo in https://github.com/RobTillaart/FastMap 
If there are problems please post an issue in that repo.

RobTillaart, 2020-03-21:
@divsys1 
Today I published FastMap as a separate library. If that works well I will go for the MAX44009 asap.
Sorry it took so long but I finally get some time now to dive into all details that need to be right to publish a lib.

RobTillaart, 2020-03-23:
@Konzen8
Fastmap is now available through library manager

closed 2020-03-23 by RobTillaart