RobTillaart/Arduino #136: When the row is full, the last value is placed on front, order loses

tonbor, 2020-01-14:
When the row is full, the last value is placed on front, order loses. There is not chosen for a array shift in memory. The function of getElement() is not much of a use. If we know the place of the first added value we can reorder the row. So please add the function uint8_t getFirstID() const in the library. I did it works. 
// returns the first added
uint8_t RunningAverage::getFirstID() const
{ if (_cnt < _size) return 0;
 return _idx;}
See: https://forum.arduino.cc/index.php?topic=548667.0

RobTillaart, 2020-01-14:
Thanks for your feedback, I will investigate asap

RobTillaart, 2020-01-15:
in the topic I see the following issues mentions:
- library only holds 256 values/floats as it has 8 bit counters.
==> making these 16 bit introduces a problem with malloc() for the 328 processor IIRC.  
      need to check the details of malloc() again but I have been working around it in my bitArray class.

- it was proposed to use running average as a buffer but it never was meant for that. For the problem described one can better allocate  a static array (e.g. in FRAM which is quite fast)

- getElement() can be used to retrieve the data in the internal array without knowing the order. Your proposal to add functions to walk through the values in order have definitely an added value so I will think how to implement this. I see how your proposal works but I think users would like to have the internal value immediately, so they can write a loop like:
```
for (uint8_t i = 0; i < ra.getCount(); i++)
{
   Serial.println(ra.getValue(i));
}
```
would that make sense to you?

RobTillaart, 2020-01-15:
To comment on myself, getValue(n) will eventually return an other value every time when new elements are added (when _cnt == _size).

RobTillaart, 2020-01-15:
propose
```
float RunningAverage::getValue(const uint8_t idx)
{
  if (_cnt == 0) return NAN;
  if (idx >= _cnt) return NAN;  // cannot ask more than is added
  uint8_t i = (idx + _idx + _cnt) % _cnt;
  return _ar[i];
}
```

so I can iterate easily
```
//
//    FILE: ra_getValue.ino
//  AUTHOR: Rob Tillaart
// VERSION: 0.1.0
//    DATE: 2020-01-15
//
// PUPROSE: demonstrate access in order of the values added
//

#include "RunningAverage.h"

RunningAverage myRA(10);
int samples = 0;

void setup(void)
{
  Serial.begin(115200);
  Serial.println("Demo RunningAverage lib");
  Serial.print("Version: ");
  Serial.println(RUNNINGAVERAGE_LIB_VERSION);
  myRA.clear(); // explicitly start clean
}

void loop(void)
{
  Serial.println();

  myRA.clear();
  for (int i = 0; i < 7; i++)
  {
    myRA.addValue(i);
  }

  Serial.print("\t getValue 0..getCount(): ");
  for (int i = 0; i < myRA.getCount(); i++)
  {
    Serial.print("\t");
    Serial.print(myRA.getValue(i), 0);
  }
  Serial.println();

  // note first values (0..2) will be overwritten by 10..12
  myRA.clear();
  for (int i = 0; i < 13; i++)
  {
    myRA.addValue(i);
  }

  Serial.print("\t getValue 0..getCount(): ");
  for (int i = 0; i < myRA.getCount(); i++)
  {
    Serial.print("\t");
    Serial.print(myRA.getValue(i), 0);
  }
  Serial.println();

  Serial.print("\t get last 5 elements added: ");
  int last = myRA.getCount() - 1;     // -1 as first idx == 0
  for (int i = last; i > last - 5 && i >= 0; i--)
  {
    Serial.print("\t");
    Serial.print(myRA.getValue(i), 0);
  }
  Serial.println();

  delay(1000);
}

// END OF FILE
```

tonbor, 2020-01-15:
Great, googling the Internet it seems that  people are aperently not aware
of the limitation of the 265 population for the Runningaverage library.

Your right, me ordering the population is good, the library solving the
problem is mutch better.

Thanks,

Rob

RobTillaart, 2020-01-15:
I will push the above changes tomorrow to the repo

RobTillaart, 2020-01-16:
version 0.2.14 pushed including getValue() to get the internal values in order of addition. See example code.

closed 2020-01-16 by RobTillaart