RobTillaart/Arduino #140: MAX44009 - Enable setting Wire interface

moritz89, 2020-01-17:
In chips that have multiple I2C (TwoWire) interfaces, it would be useful to be able to pass a pointer to the used TwoWire object.

The question is how this should be implemented, as a dedicated function? Other sensor interfaces usually have a begin() function where this can be added as an optional parameter. Both implementations can be implemented in a backward compatible manner.

RobTillaart, 2020-01-17:
The method to get this working is something like the following 

- create a reference to a Wire object 
- add it as an optional parameter to the constructor

``` 
[max44009.h]

Max44009(const uint8_t address, TwoWire *wir = &Wire);

private:
TwoWire *_wi;
...
in the implementation of the ctor, assign the parameter to the internal TwoWire reference 

```
[max44009.cpp]

Max44009::Max44009(const uint8_t address, TwoWire *wir)
{
    _address = address;
    _wi = wir;
    _data = 0;
    _error = 0;

    _wi->begin();
    _wi->setClock(400000);
}
```

replace  Wire.  by  _wi->  in remaining of the  .cpp file

I have no time to implement/test the above on short term, so you might give it a try.

RobTillaart, 2020-01-21:
@moritz89

Should the call to configure() not include
```
      _wire->begin();
```
as the user explicitly selects an instance of _wire

(i'm fixing some other issues now so I can include the fix)

moritz89, 2020-01-21:
It would make sense to make this optional. This would allow both scenarios to be covered. As the default is to allows call begin() it would make sense to follow this convention something along the lines of:

    void configure(const uint8_t address, TwoWire *wire, const Boolean begin = Boolean::True);

RobTillaart, 2020-01-21:
If I understand correctly, at some time the user should call somewire.begin()  himself if he selects False.

Otherwise reading and setting registers is not possible (although this might be implementation dependant). 

I will implement your proposal

RobTillaart, 2020-01-21:
@moritz89 
Think the configure function should be core of constructors as this enforces "identical" behavior.
your opinion?

```
Max44009::Max44009(const uint8_t address, const Boolean begin)
{
	Max44009::configure(address, &Wire, begin);
}

Max44009::Max44009(const Boolean begin)
{
	Max44009::configure(MAX44009_DEFAULT_ADDRESS, &Wire, begin);
}

void Max44009::configure(const uint8_t address, TwoWire *wire, const Boolean begin)
{
    _address = address;
    _data = 0;
    _error = MAX44009_OK;
    _wire = wire;
    if (begin == Boolean::True)
    {
      _wire->begin();
    }
}
```

moritz89, 2020-01-22:
Looks great and improves code reuse. Would it make sense to include default values for `const Boolean begin` to ensure backward API compatibility?

RobTillaart, 2020-01-27:
Yes, should be TRUE by default for all

RobTillaart, 2020-01-27:
@moritz89 
Updated the constructors & configure.

If you have time please confirm it still works as intended. IF not please reopen the issue.

closed 2020-01-27 by RobTillaart