RobTillaart/Arduino #143: Store large numbers

EliasA97, 2020-02-29:
Hello, is it possible to store large numbers like 1.000.000? If yes how can I do it?

RobTillaart, 2020-02-29:
Yes, but there are several ways to do this and depending on your detailed needs and usage one way or the other will be a better way.

Format: as text as integer as float ....
Precision: all digits correct or order of magnitude
Storage medium: FRAM EEPROM SDCARD ...

Is it related to one of my libraries? If so which one? If not the question is better asked on e.g. the Arduino forum.

EliasA97, 2020-03-06:
Hi Rob, sorry for not being clear about what I asked you. The number 1.000.000 (one million) must be a long integer, so it would be like this: 1000000. The thing is that each fram's or eeprom's byte supports numbers up to 255. What I want to do is store each of my number's byte to 7 different fram bytes.

RobTillaart, 2020-03-06:
For a long you only need 4 bytes in storage.
In pseudo code

Long x = 1000000;
Long t = x;. //work copy
For (i = 1 to 4)
    Byte b = t & 255;
    Store b
    T = t / 255;

Reading it back
Long t = 0
For i = 1 to 4
    Byte b
    Recall b
    T = t x 256 + b
Long x = t

Thats all

RobTillaart, 2020-03-07:
The FRAM library has two functions for storing / retrieving a long

```
  void  write32(uint16_t memaddr, uint32_t value);
  uint32_t read32(uint16_t memaddr);
```

so 
``` 
long x = 1000000;
uint16_t address = 12345; // wherever
FRAM.write32(address, x);

long y = FRAM.read32(address);
Serial.println(y);
```

RobTillaart, 2020-03-08:
Think this solves the issue otherwise reopen it

closed 2020-03-08 by RobTillaart