RobTillaart/Arduino #145: DS18b20 class

arslankhawar, 2020-03-08:
Hello can any one help me in resolving the following issue in arduino
"'class DS18B20' has no member named 'begin'"

by the way i have already installed the library and still the error is there.
kindly help me .
Thanks

arslankhawar, 2020-03-08:
This is the code that i'm using

/* This code is to use with DS18b20 Temperature sensor
 * It shows the temperature in °C or °F on the serial monitor
 */

#include <OneWire.h>
#include <DS18B20.h>

OneWire oneWire(2);
DS18B20 sensor(&oneWire);

float f;

void setup(void)
{
  Serial.begin(9600);
  sensor.begin();
}

void loop(void)
{ 
  sensor.requestTemperatures();
  while (!sensor.isConversionComplete());
  Serial.print("Temperature:");
  Serial.print("\t");
//  Serial.print(sensor.getTempC());       //Uncomment these lines to print in °C
//  Serial.println(" °C");

  f=(sensor.getTempC() * 1.8) + 32;       //°F
  Serial.print(f);
  Serial.println(" °F");
  delay(1000);
}

RobTillaart, 2020-03-08:
Code seems ok
What platform do you use?  Uno esp...
Which version of IDE? Or other environment.

arslankhawar, 2020-03-09:
I'm using arduino uno board and arduino ide.

RobTillaart, 2020-03-09:
Tested your sketch and it does compile with the IDE 1.8.12
You might need to reinstall the latest version of the library.

RobTillaart, 2020-03-09:
Have you compiled the example sketches?
Same error?

arslankhawar, 2020-03-13:
First of all sorry for late reply.
I have already installed the latest version of the library and I did run the sample sketch same error.
Thanks

RobTillaart, 2020-03-16:
I have compiled your code and it works for me. So the sketch itself is OK. 
It might be in the DS18B20 lib or in the OneWire lib?

Which version of the OneWire library do you have?  [ I use 2.3.5]

IDE menu ==> tools ==> manage libraries ==> type == installed  just type onewire in search box

RobTillaart, 2020-03-16:
Another thing you might do is open the sketch in notepad (or any ascii only editor) and check if there are non-ascii characters. Happens sometimes...

RobTillaart, 2020-05-03:
@arslankhawar  
I close this issue, as a new version of the DS18B20 lb has moved to https://github.com/RobTillaart/DS18B20_RT
If the problem remains, please (re)open an issue there.

Thanks

closed 2020-05-03 by RobTillaart