RobTillaart/Arduino #146: Dewpoint calculation reference

samweisgamdschie, 2020-04-04:
Hi!

I need a source or reference where you have the formula of `float dewPoint(float celsius, float humidity)` from. According to the link in the comments the only function I consider to be relevant is `FUNCTION DWPT(T,RH)` but it looks way different to your approach. And the website of NOAA is like an information grave.I would be happy for any hint!

RobTillaart, 2020-04-04:
Which library are you refering to?

RobTillaart, 2020-04-04:
OK found all references I used:

[1] https://wahiduddin.net/calc/density_algorithms.htm
[2] https://web.archive.org/web/20100528030817/https://www.colorado.edu/geography/weather_station/Geog_site/about.htm
** Found the 2nd reference in the web archive, not online anymore

Reference [2] explains the last lines of the formula  ( added some comments)
```
    // VP = saturation vapor pressure (kPa) mwntioned in [2]
    float VP = pow(10, SUM-3) * humidity;      // the -3 is to go from Pa to kPa
    float T = log(VP/0.61078);                 // temp var
    return (241.88*T) / (17.558-T);
```

The code to calculate the saturation vapor pressure SVP in [2] uses the following approximating polynomial (see Lowe, P.R. **1930**. J. Appl. Meteor., 16:100-103):
```
SVp = 6.107799961 +T*(4.436518521*10^-1 +T*(1.428945805*10^-2 +T*(2.650648471*10^-4 
         +T*(3.031240396*10^-6 +T*(2.034080948*10^-8 +6.136820929*10^-11*T)))))
```

The NOAA uses a formula from ESGG(temp)  based on work from **1963** so newer, see [1].
(search for FUNCTION ESGG  )  
As the algoritme ESGG [2] is newer and not a polynomial formula, so I may assume it fits real data better. 

Merging these two pieces of math resulted in my final code.

Does this answer your question?

I'll update these comments in the code / readme.md

samweisgamdschie, 2020-04-04:
Yes, perfectly! Many Thanks for your explanations and research. As my bac work uses your function I needed to know where it comes from. I will add a link to your project my work then.
If you are interested in newer calculations you will find great starting point here https://doi.org/10.1175/BAMS-86-2-225 . Many thanks again!

samweisgamdschie, 2020-04-04:
Maybe one further thing I was wondering: if this function is your own creation, in all copies I found, https://wahiduddin.net/calc/density_algorithms.htm added as a reference but I can't figure out why? The only function which calculated the dewpoint with temperature and RH as input is FUNCTION DWPT(T,RH) but it's a completely different approach..

closed 2020-04-04 by samweisgamdschie

RobTillaart, 2020-04-04:
That reference has the saturation vapor pressure SVP code in the form of the ESGG() function which is embedded in dewPoint() as explained above. The 2nd reference was removed apparently when the site had gone.  

Soon this lib will get its own repo as I am in a process of splitting this large repo in individual libs. That way the PR, issues etc are more clear and the Arduino IDE can include them in the lib manager.

When time permits I hope to port more of the NOAA fortran code to the Arduino.

RobTillaart, 2020-04-06:
Please note I have cleaned up the temperature code in a new repo, added tests and removed / added code.
 
https://github.com/RobTillaart/Temperature.

dark2strike, 2020-04-07:
Why in separate repository?

RobTillaart, 2020-04-07:
Then the libraries can be included in the Arduino library manager. With one big repo that is not possible.

Furthermore it keeps the issues focussed for on soecfic lib and it will ve easier to get a certain version of the library. 

Goal is to have all repos split by the end of the year.